Koenkk/zigbee2mqtt #21779: aqara d1 paired and supported but exposes are null

jaychen1230, 2024-03-13:
### What happened?

can only control state and operation mode, other exposes(power, voltage, action)are null.
[image]
[image]

### What did you expect to happen?

any fix

### Zigbee2MQTT version

 1.36.0 commit: 86ed71c

### Adapter firmware version

 20210319

### Adapter

cc2652p

### Setup

docker container

mrskycriper, 2024-03-13:
This is odd. I've looked into it's converter and saw strange endpoint configuration:

```js
endpoint: (device) => {
    return {'left': 1, 'right': 2, 'system': 1};
},
```

They shouldn't be duplicated like that. Maybe it's causing this problem somehow.

I'll start converting D1 to modern extends soon, that might be enough to get it working.

Also, can you please provide entry from the `zigbee2mqtt/datebase.db` on this device? That might be helpful for later.

jaychen1230, 2024-03-14:
I paired another aqara triple switch with neutral, here is database.db info and debug log. Meanwhile, other aqara double and triple switch got the same problem
```
"id":1,"type":"Coordinator","ieeeAddr":"0x00124b002291d36c","nwkAddr":0,"manufId":0,"epList":[242,47,13,12,110,11,10,8,6,5,4,3,2,1],"endpoints":{"1":{"profId":260,"epId":1
["id":2,"type":"Router","ieeeAddr":"0x54ef4410007cff33","nwkAddr":49032,
"manufId":4447,"manufName""LUMI","powerSource":"Mains (single phase)","modelId":"lumi.switch.n3acn3
```

mrskycriper, 2024-03-14:
> I paired another aqara triple switch with neutral, here is database.db

It cuts off abruptly, can you please copy it once again? Make sure to include the full string length. If you have different D1 switches, please add their entries too.

adynis, 2024-03-14:
could be similar to: https://github.com/Koenkk/zigbee2mqtt/issues/21679

jaychen1230, 2024-03-24:
after update the sonoff dongle plus firmware to the latest version. some exposes work but action is still null.
[image]
```
Info 2024-03-25 02:25:39Successfully changed options
Info 2024-03-25 02:25:39MQTT publish: topic 'zigbee2mqtt/bridge/response/options', payload '{"data":{"restart_required":false},"status":"ok","transaction":"jyfke-2"}'
Debug 2024-03-25 02:25:47Received MQTT message on 'zigbee2mqtt/bridge/request/permit_join' with data '{"device":null,"time":254,"transaction":"jyfke-3","value":true}'
Info 2024-03-25 02:25:47Zigbee: allowing new devices to join.
Info 2024-03-25 02:25:47MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":true},"status":"ok","transaction":"jyfke-3"}'
Info 2024-03-25 02:25:48Device '0x54ef4410007cfe4b' joined
Info 2024-03-25 02:25:49MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x54ef4410007cfe4b","ieee_address":"0x54ef4410007cfe4b"},"type":"device_joined"}'
Info 2024-03-25 02:25:49Starting interview of '0x54ef4410007cfe4b'
Info 2024-03-25 02:25:49MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x54ef4410007cfe4b","ieee_address":"0x54ef4410007cfe4b","status":"started"},"type":"device_interview"}'
Debug 2024-03-25 02:25:49Device '0x54ef4410007cfe4b' announced itself
Info 2024-03-25 02:25:49MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x54ef4410007cfe4b","ieee_address":"0x54ef4410007cfe4b"},"type":"device_announce"}'
Debug 2024-03-25 02:25:49Received Zigbee message from '0x54ef4410007cfe4b', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
Debug 2024-03-25 02:25:49Skipping message, still interviewing
Debug 2024-03-25 02:25:49Received Zigbee message from '0x54ef4410007cfe4b', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 2 with groupID 0
Debug 2024-03-25 02:25:49Skipping message, still interviewing
Debug 2024-03-25 02:25:49Received Zigbee message from '0x54ef4410007cfe4b', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 3 with groupID 0
Debug 2024-03-25 02:25:49Skipping message, still interviewing
Info 2024-03-25 02:25:49Successfully interviewed '0x54ef4410007cfe4b', device has successfully been paired
Info 2024-03-25 02:25:49Device '0x54ef4410007cfe4b' is supported, identified as: Aqara Smart wall switch D1 (with neutral, triple rocker) (QBKG26LM)
Info 2024-03-25 02:25:49MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Smart wall switch D1 (with neutral, triple rocker)","exposes":[{"endpoint":"left","features":[{"access":7,"description":"On/off state of the switch","endpoint":"left","label":"State","name":"state","property":"state_left","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"type":"switch"},{"endpoint":"center","features":[{"access":7,"description":"On/off state of the switch","endpoint":"center","label":"State","name":"state","property":"state_center","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"type":"switch"},{"endpoint":"right","features":[{"access":7,"description":"On/off state of the switch","endpoint":"right","label":"State","name":"state","property":"state_right","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"type":"switch"},{"access":7,"description":"Decoupled mode for left button","endpoint":"left","label":"Operation mode","name":"operation_mode","property":"operation_mode_left","type":"enum","values":["control_relay","decoupled"]},{"access":7,"description":"Decoupled mode for center button","endpoint":"center","label":"Operation mode","name":"operation_mode","property":"operation_mode_center","type":"enum","values":["control_relay","decoupled"]},{"access":7,"description":"Decoupled mode for right button","endpoint":"right","label":"Operation mode","name":"operation_mode","property":"operation_mode_right","type":"enum","values":["control_relay","decoupled"]},{"access":1,"category":"diagnostic","description":"Instantaneous measured power","label":"Power","name":"power","property":"power","type":"numeric","unit":"W"},{"access":7,"category":"config","description":"Enable/disable the power outage memory, this recovers the on/off mode after power failure","label":"Power outage memory","name":"power_outage_memory","property":"power_outage_memory","type":"binary","value_off":false,"value_on":true},{"access":7,"category":"config","description":"Enable/disable the LED at night","label":"LED disabled night","name":"led_disabled_night","property":"led_disabled_night","type":"binary","value_off":false,"value_on":true},{"access":1,"category":"diagnostic","description":"Measured electrical potential value","label":"Voltage","name":"voltage","property":"voltage","type":"numeric","unit":"V"},{"access":1,"description":"Sum of consumed energy","label":"Energy","name":"energy","property":"energy","type":"numeric","unit":"kWh"},{"access":1,"category":"diagnostic","description":"Temperature of the device","label":"Device temperature","name":"device_temperature","property":"device_temperature","type":"numeric","unit":"°C"},{"access":7,"category":"config","description":"After turn on, the indicator light turns on while switch is off, and vice versa","label":"Flip indicator light","name":"flip_indicator_light","property":"flip_indicator_light","type":"binary","value_off":"OFF","value_on":"ON"},{"access":1,"descri [... cut ...]
Info 2024-03-25 02:26:15MQTT publish: topic 'zigbee2mqtt/0x54ef4410007cfe4b', payload '{"action":null,"consumption":4.8809717156927945e-8,"current":0,"device_temperature":26,"energy":0,"flip_indicator_light":null,"led_disabled_night":false,"linkquality":127,"operation_mode_center":null,"operation_mode_left":null,"operation_mode_right":null,"power":0,"power_outage_count":0,"power_outage_memory":false,"state":"OFF","state_center":"OFF","state_left":"OFF","state_right":"OFF","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":228.93}'
```

closed 2024-07-09 by jaychen1230