Koenkk/zigbee2mqtt #21780: [Wrong device]: Moes ERS-10TZBVB-AA identified as Loginovo ZG-101ZL

klu16, 2024-03-13:
### Link

https://pl.aliexpress.com/item/1005005805357128.html

### Model

ERS-10TZBVB-AA

### Description

Smart button

### Vendor

Moes

### Picture (link)

https://www.zigbee2mqtt.io/devices/ERS-10TZBVB-AA.html

### Database entry

{"id":21,"type":"EndDevice","ieeeAddr":"0x187a3efffefadc35","nwkAddr":29431,"manufId":4098,"manufName":"_TZ3000_ja5osu5g","powerSource":"Battery","modelId":"TS004F","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":260,"inClusterList":[0,1,3,4,6,4096,57345],"outClusterList":[25,10,3,4,6,8,4096],"clusters":{"genBasic":{"attributes":{"65503":"E\u00174-\u0011�a|-\u0011�\u0015�-\u0011j͂-\u0012","65506":31,"65508":0,"65534":0,"modelId":"TS004F","manufacturerName":"_TZ3000_ja5osu5g","powerSource":3,"zclVersion":3,"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genOnOff":{"attributes":{"tuyaOperationMode":1}},"manuSpecificTuya_3":{"attributes":{"53265":"�\u0000\u0003�\u0003�\u0000Z\u0003�\u0003�\u0000\u0003�\u0000d\u0000�\u0003�\u0003�\u0000\u0001�\u0003 \u0001\u000e\u0003�\u0003�\u0001\u0001�\u0000\u0000\u0000�\u0000�\u0000�"}},"genPowerCfg":{"attributes":{"batteryVoltage":29,"batteryPercentageRemaining":184}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b002a2f276b","endpointID":1},{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b002a2f276b","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":65000,"repChange":0,"manufacturerCode":null}],"meta":{}}},"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-429758583},"lastSeen":1710321214644}

### Notes

[image]

but it should be this device: https://www.zigbee2mqtt.io/devices/ERS-10TZBVB-AA.html

Z2M details (stick: Sonoff Dongle-P)

[image]

fliespl, 2024-04-07:
Same here and actions stopped working.

Identifies as brightness or color change instead of single.

a1exs, 2024-05-12:
It stopped working for me too.

kkirov00, 2024-11-07:
Hey, have you found a solution for this? I just transfered my automations from Tuya to HA and I noticed the same. This button has been giving me troubles even when it was in Tuya and it continues in HA (sometimes not reacting at all), so was thinking that maybe if HA identifies it properly, then it would work properly.

lucianoberger, 2024-11-24:
I found the same issue here; one of the updates replaced the device.

romainpiel, 2024-12-07:
I think it was caused by this PR https://github.com/Koenkk/zigbee-herdsman-converters/pull/6897 which fixed https://github.com/Koenkk/zigbee2mqtt/issues/20757. The Moes button is white labelled with Loginovo ZG-101ZL but these are actually different devices (the Moes button doesn't have a led for example). cc @danieledwardgeorgehitchcock

closed 2025-02-14 by github-actions[bot]

s0ftice, 2025-04-05:
this should be corrected and not closed as not planned...

leah-potato, 2025-06-19:
Same issue here.