Koenkk/zigbee2mqtt #21782: [Wrong device]: Water leak sensor Onenuo 898GZ identified as Aubess IH-K665

klu16, 2024-03-13:
### Link

https://pl.aliexpress.com/item/1005006579032290.html

### Model

898GZ

### Description

Water leak sensor

### Vendor

Onenuo

### Picture (link)

https://ibb.co/Xpwq0Rn

### Database entry

{"id":20,"type":"EndDevice","ieeeAddr":"0xa4c138326b87162a","nwkAddr":58547,"manufId":4417,"manufName":"_TZ3000_upgcbody","powerSource":"Battery","modelId":"TS0207","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[1,3,1280,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0207","manufacturerName":"_TZ3000_upgcbody","powerSource":3,"zclVersion":3,"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":""}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b002a2f276b","zoneState":1}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b002a2f276b","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":65000,"repChange":0,"manufacturerCode":null}],"meta":{}}},"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":491743244},"lastSeen":1710314783189}

### Notes

[image]

but it should be this one: https://www.zigbee2mqtt.io/devices/TS0207_water_leak_detector.html

Z2M details (stick: Sonoff Dongle-P):

[image]

TheNutHouse84, 2024-03-13:
I have the same water leak sensor ([Onenuo](https://www.aliexpress.us/item/3256804880152642.html)) and have the same mis-identification.

In z2m ver 1.35.3 it was "more" correctly (physical appearance wise) identified as "TS0207_water_leak_detector_2" and showed correct picture.

After upgrade to 1.36.0 it was mis-id'd

I also have [this version](https://www.aliexpress.us/item/3256806097907081.html) of nearly identical water leak sensor and it is listed in z2m as "TS0207_water_leak_detector" and was correctly identified in bot 1.35.3 and 1.36.0

As a side note to anyone contemplating purchase of these devices. The Onenuo style/brand (imo) is constructed better, the case is made of better quality plastic and the fit and finish is better.  Both work equally well though.

Bsector, 2024-03-21:
I bought the same 898EZ sensor from ONENUO and the same identification error occurs.
Furthermore, the value of Water lake is not being exported.
Does the same happen to you? How could I fix this?
[image]

klu16, 2024-03-21:
> I bought the same 898EZ sensor from ONENUO and the same identification error occurs. Furthermore, the value of Water lake is not being exported. Does the same happen to you? How could I fix this? ![Captura de tela 2024-03-21 172548](https://private-user-images.githubusercontent.com/53239310/315615860-4efdfc2d-b57a-4cc9-95f0-87a9354cbfbe.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTEwNTM0MzEsIm5iZiI6MTcxMTA1MzEzMSwicGF0aCI6Ii81MzIzOTMxMC8zMTU2MTU4NjAtNGVmZGZjMmQtYjU3YS00Y2M5LTk1ZjAtODdhOTM1NGNiZmJlLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMjElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzIxVDIwMzIxMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWIwNDFlNDMxYjZjODM0N2MxMTcxOTA2ZGRjZTkwZmMxNWNlODVmMDMxZTk1OTNhOTgxNTcxN2I5MjA0YTJkNTkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.vgvCkzVnUx6NRV9G-ODr7b2GTAbniarm2IVGnlTTp1o)

No, entities are identified fine. Just the model is wrong.

[image]

Bsector, 2024-03-31:
Hello, has anyone managed to solve this problem?

lordlightman, 2024-04-09:
Same problem occurs with [ONENUO model 898WZ](https://www.aliexpress.com/item/1005006412524793.html). Device was interviewed for a long, long time and then identified as Aubess IH-K665.

klu16, 2024-10-15:
It's still wrong detecting.

shawnhaywood, 2024-10-19:
I purchased a device recognized by Zigbee2mqtt as Aubess IH-K665, but doesn't match the picture.  This is what the device looks like.  The device registers fine and will detect a leak, but doesn't report anything else, or update.  
[image]

cypherbits, 2024-12-28:
Same issue as https://github.com/Koenkk/zigbee2mqtt/issues/24759, but wrong identification since I have another completely different and have the same _TZ3000_upgcbody id.

closed 2025-05-07 by github-actions[bot]