Koenkk/zigbee2mqtt #21784: No converter available for 'TS0042' with cluster 'genOnOff' and type 'commandTuyaAction' and data '{"value":0}'

nikitah4x, 2024-03-13:
### What happened?

My Wireless Switch stopped working (_TZ3000_dfgbtub0), and even when Im clicking on the button, the action is still N/A.

### What did you expect to happen?

Restore functionality to the wireless switch

### Zigbee2MQTT version

1.36.0-dev commit: 56feb77

### Adapter firmware version

20210708

### Adapter

Sonoff Dongle P

### Setup

Add-on on Home Assistant running on Raspberry Pi 4B

### Debug log

Debug 2024-03-13 15:13:12No converter available for 'TS0042' with cluster 'genOnOff' and type 'commandTuyaAction' and data '{"value":0}'
Debug 2024-03-13 15:13:12Received Zigbee message from 'Wireless_Switch', type 'commandTuyaAction', cluster 'genOnOff', data '{"value":0}' from endpoint 2 with groupID 0
Debug 2024-03-13 15:13:13No converter available for 'TS0042' with cluster 'genOnOff' and type 'commandTuyaAction' and data '{"value":0}'
Debug 2024-03-13 15:13:14Received Zigbee message from 'Wireless_Switch', type 'commandTuyaAction', cluster 'genOnOff', data '{"value":0}' from endpoint 1 with groupID 0
Debug 2024-03-13 15:13:14No converter available for 'TS0042' with cluster 'genOnOff' and type 'commandTuyaAction' and data '{"value":0}'

nikitah4x, 2024-03-13:
[image]

nikitah4x, 2024-03-13:
[image]

nikitah4x, 2024-03-13:
To fix this issue, I removed the Edge version and migrated to the normal one.

Koenkk, 2024-03-14:
Pushed a fix, can you check if it works?

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

1igeza, 2024-03-14:
It's fixed for me.
    model: 'TS0041',
    vendor: '_TZ3000_yj6k7vfo',
[tuya.txt](https://github.com/Koenkk/zigbee2mqtt/files/14608217/tuya.txt)

adynis, 2024-03-14:
Fixed also for me (HA yellow)
Z2M Edge version: 1.36.0-dev commit: [a4f9c49](https://github.com/Koenkk/zigbee2mqtt/commit/a4f9c49)
```
2024-03-14T22:01:43.683Z zigbee-herdsman:controller:log Received 'zcl' data '{"address":44008,"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":76,"manufacturerCode":null,"commandIdentifier":10},"Payload":[{"attrId":0,"dataType":37,"attrData":[0,4596]}],"Command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}},"endpoint":1,"linkquality":176,"groupID":0,"wasBroadcast":false,"destinationEndpoint":1}'
Zigbee2MQTT:debug 2024-03-15 00:01:43: Received Zigbee message from 'releu_aqara_vizor', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,4596]}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2024-03-15 00:01:43: MQTT publish: topic 'zigbee2mqtt/releu_aqara_vizor', payload '{"auto_off":false,"button_lock":"OFF","consumer_connected":null,"current":null,"device_temperature":32,"energy":4.6,"led_disabled_night":false,"linkquality":176,"overload_protection":null,"power":0.9,"power_outage_memory":false,"state":"ON","update":{"installed_version":41,"latest_version":32,"state":"idle"},"update_available":null,"voltage":null}'
2024-03-14T22:01:44.363Z zigbee-herdsman:adapter:ember:uart:ash:parser <<<< [FRAME raw=103db1a97d312a15b658924a24ab5593499c609f6124309874f9c7e574fc9c127e]
2024-03-14T22:01:44.363Z zigbee-herdsman:adapter:ember:uart:ash <--- [FRAME type=DATA]
2024-03-14T22:01:44.363Z zigbee-herdsman:adapter:ember:uart:ash <--- [FRAME type=DATA ackNum=0]
2024-03-14T22:01:44.363Z zigbee-herdsman:adapter:ember:uart:ash <--- [FRAME type=DATA ackNum=0 frmNum=1] Added to rxQueue
2024-03-14T22:01:44.363Z zigbee-herdsman:adapter:ember:uart:ash ---> [FRAME type=ACK frmRx=2]
2024-03-14T22:01:44.364Z zigbee-herdsman:adapter:ember:uart:ash:writer >>>> [FRAME raw=82503a7e]
2024-03-14T22:01:44.364Z zigbee-herdsman:adapter:ember:ezsp <=== [FRAME: ID=69:"INCOMING_MESSAGE_HANDLER" Seq=127 Len=28]
2024-03-14T22:01:44.364Z zigbee-herdsman:adapter:ember:ezsp ezspIncomingMessageHandler(): callback called with: [type=UNICAST], [apsFrame={"profileId":260,"clusterId":6,"sourceEndpoint":1,"destinationEndpoint":1,"options":256,"groupId":0,"sequence":46}], [lastHopLqi=184], [lastHopRssi=202], [sender=65225], [bindingIndex=255], [addressIndex=255], [messageContents=0186fd00]
2024-03-14T22:01:44.367Z zigbee-herdsman:controller:log Received 'zcl' data '{"address":65225,"frame":{"Header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":134,"manufacturerCode":null,"commandIdentifier":253},"Payload":{"value":0},"Command":{"ID":253,"parameters":[{"name":"value","type":32}],"name":"tuyaAction"}},"endpoint":1,"linkquality":184,"groupID":0,"wasBroadcast":false,"destinationEndpoint":1}'
2024-03-14T22:01:44.368Z zigbee-herdsman:controller:endpoint DefaultResponse 0xa4c13814dd7d8de7/1 6(253, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
2024-03-14T22:01:44.369Z zigbee-herdsman:adapter:ember:queue Status queue=0 priorityQueue=0.
Zigbee2MQTT:debug 2024-03-15 00:01:44: Received Zigbee message from 'bucatarie_loratap_4', type 'commandTuyaAction', cluster 'genOnOff', data '{"value":0}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2024-03-15 00:01:44: MQTT publish: topic 'zigbee2mqtt/bucatarie_loratap_4', payload '{"action":"1_single","battery":100,"linkquality":184,"voltage":3000}'
Zigbee2MQTT:info  2024-03-15 00:01:44: MQTT publish: topic 'zigbee2mqtt/bucatarie_loratap_4', payload '{"action":"","battery":100,"linkquality":184,"voltage":3000}'
Zigbee2MQTT:info  2024-03-15 00:01:44: MQTT publish: topic 'zigbee2mqtt/bucatarie_loratap_4/action', payload '1_single'
2024-03-14T22:01:44.378Z zigbee-herdsman:adapter:ember:adapter ~~~> [ZCL to=65225 apsFrame={"profileId":260,"clusterId":6,"sourceEndpoint":1,"destinationEndpoint":1,"options":4352,"groupId":0,"sequence":0} header={"frameControl":{"reservedBits":0,"frameType":0,"direction":1,"disableDefaultResponse":true,"manufacturerSpecific":false},"transactionSequenceNumber":134,"manufacturerCode":null,"commandIdentifier":11}]
2024-03-14T22:01:44.378Z zigbee-herdsman:adapter:ember:ezsp ===> [FRAME: ID=52:"SEND_UNICAST" Seq=128 Len=26]
2024-03-14T22:01:44.379Z zigbee-herdsman:adapter:ember:uart:ash ---> [FRAME type=DATA frmTx=0 frmRx=2]
2024-03-14T22:01:44.379Z zigbee-herdsman:adapter:ember:uart:ash:writer >>>> [FRAME raw=02c221a9602a157ba7904b23aa5493498d4e27aba2cb7f0df63b6306b47e]
2024-03-14T22:01:44.388Z zigbee-herdsman:adapter:ember:uart:ash:parser <<<< [FRAME raw=21c2a1a9602a15b61ead7e]
2024-03-14T22:01:44.388Z zigbee-herdsman:adapter:ember:uart:ash <--- [FRAME type=DATA]
2024-03-14T22:01:44.388Z zigbee-herdsman:adapter:ember:uart:ash <--- [FRAME type=DATA ackNum=1]
2024-03-14T22:01:44.388Z zigbee-herdsman:adapter:ember:uart:ash <--- [FRAME type=DATA ackNum=1 frmNum=2] Added to rxQueue
2024-03-14T22:01:44.389Z zigbee-herdsman:adapter:ember:uart:ash ---> [FRAME type=ACK frmRx=3]
2024-03-14T22:01:44.389Z zigbee-herdsman:adapter:ember:uart:ash:writer >>>> [FRAME raw=83401b7e]
2024-03-14T22:01:44.389Z zigbee-herdsman:adapter:ember:ezsp <=== [FRAME: ID=52:"SEND_UNICAST" Seq=128 Len=7]
2024-03-14T22:01:44.390Z zigbee-herdsman:adapter:ember:ezsp ~~~> [SENT type=DIRECT apsSequence=4 messageTag=79 status=SUCCESS]
```

Koenkk, 2024-03-15:
perfect!

closed 2024-03-15 by Koenkk