Koenkk/zigbee2mqtt #21786: SNZB-02D ingnoring reporting interval

rmtucker, 2024-03-13:
### What happened?

Trying to change reporting on change from 20 to 10.
Woke up the device first and presses apply.
No errors but device is ignoring the change it still only reports on every 0.2deg of change.
[image]

### What did you expect to happen?

The device would report on every 0.1deg change

### Zigbee2MQTT version

1.36.0

### Adapter firmware version

20230507

### Adapter

Sonoff zigbee 3.0 dongle plus

### Setup

Add on homeassistant os

rmtucker, 2024-03-14:
Just wondering if the standard non display type SNZB-02 has the same problem?

Ziron, 2024-03-25:
The SNZB-02D only have a resolution of 0.2 °C listed on its product page and data sheet. https://sonoff.tech/product/gateway-and-sensors/snzb-2d/
So it would be expected that the firmware of the sensor limits the minimum value of the parameter to 20 in that case.

a1ex-ak, 2024-07-28:
I faced the same problem but I see the error:

z2m: Request 'zigbee2mqtt_1/bridge/request/device/configure_reporting' failed with error: 'Bind 0x90ab96fffe77bfd0/1 msTemperatureMeasurement from '0x00124b0024cbee34/1' failed (AREQ - ZDO - bindRsp after 10000ms)'

optimistic-optician, 2024-08-22:
> The SNZB-02D only have a resolution of 0.2 °C listed on its product page and data sheet. https://sonoff.tech/product/gateway-and-sensors/snzb-2d/ So it would be expected that the firmware of the sensor limits the minimum value of the parameter to 20 in that case.

Hi, you are right, if you look closely at the images on their website, it even shows the graph has 0.2°C steps. Unfortunately they quote "Accuracy +-0.2°C", which I think is just fine for a consumer product, but that this also implies "Resolution 0.2°C is a bit misleading. The new SNZB-02P which claims to have the same sensor but 4 years battery life instead of 2, resolves 0.1°C.

rmtucker, 2024-08-22:
I also found out that the humidity required a calibration of 8%,which is fine,but the display does not reflect the calibration.
But Homeassistant show it correctly,So i opted to only buy the SNZB-02P which works perfectly,still had to calibrate the humidity,but without the display it was a better product.

closed 2025-01-07 by github-actions[bot]