Koenkk/zigbee2mqtt #21787: [New device support]: _TZE200_lawxy9e2

dotosouza, 2024-03-13:
### Link

https://es.aliexpress.com/item/1005005835095616.html?spm=a2g0o.productlist.main.1.44527fCg7fCgi5&algo_pvid=7fb3c305-7bff-4840-a2e4-372bb6f11641&algo_exp_id=7fb3c305-7bff-4840-a2e4-372bb6f11641-0&pdp_npi=4%40dis%21BRL%2188.15%2188.15%21%21%21121.17%21121.17%21%402103242517103533450598063e4746%2112000034518044886%21sea%21BR%213365536725%21&curPageLogUid=uNDEPdupkWB7&utparam-url=scene%3Asearch%7Cquery_from%3A

### Database entry

{"id":16,"type":"Router","ieeeAddr":"0xa4c138ed34c8ed28","nwkAddr":3533,"manufId":4417,"manufName":"_TZE200_lawxy9e2","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"P�}-iϳ}-iϳ}-iг}-iг}-iѳ}-iҳ}-i","65506":54,"65508":0,"stackVersion":0,"dateCode":"","appVersion":70}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1710349544758}

### Comments

I'm not a programmer, but did some copying and pasting. It just works for me. Don't know if it can be done in a better way

### External definition

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const {} = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: 'Tuya',
    description: 'Interruptor com luz e ventilador',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    configure: tuya.configureMagicPacket,
    exposes: [
        e.binary('status_indication', ea.STATE_SET, 'ON', 'OFF').withDescription('Light switch'),
        tuya.exposes.switch(),
        e.power_on_behavior(['OFF', 'ON']).withAccess(ea.STATE_SET),
        e.numeric('fan_speed', ea.STATE_SET).withValueMin(1).withValueMax(5).withValueStep(1)
            .withDescription('Speed off the fan')
    ],
    meta: {
        tuyaDatapoints: [
            [1, 'state', tuya.valueConverter.onOff],
            [3, 'fan_speed', tuya.valueConverterBasic
                .lookup({'1': tuya.enum(0), '2': tuya.enum(1), '3': tuya.enum(2), '4': tuya.enum(3), '5': tuya.enum(4)})],
            [11, 'power_on_behavior', tuya.valueConverterBasic.lookup({'OFF': tuya.enum(0), 'ON': tuya.enum(1)})],
            [5, 'status_indication', tuya.valueConverter.onOff],
        ],
    },
    whiteLabel: [
        {vendor: 'Liwokit', model: 'Fan+Light-01'},
    ],
};

module.exports = definition;
```

Koenkk, 2024-03-14:
Added, thanks!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-03-14 by Koenkk

GGG-KILLER, 2024-04-13:
@Koenkk this one might be a bit different as it includes a light switch as well, just bought one of these, added the external definition according to your commit and there's only an on/off for the fan, but not the light.

GGG-KILLER, 2024-04-13:
The following worked for me in the end:
```js
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const exposes = require('zigbee-herdsman-converters/lib/exposes');

const e = exposes.presets;
const ea = exposes.access;

module.exports = {
    fingerprint: tuya.fingerprint('TS0601', ['_TZE200_lawxy9e2']),
    model: 'TS0601_fan_switch',
    vendor: 'TuYa',
    description: 'Fan switch',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    configure: tuya.configureMagicPacket,
    exposes: [
        e.binary('status_indication', ea.STATE_SET, 'ON', 'OFF').withDescription('Light switch'),
        tuya.exposes.switch(), e.power_on_behavior(['off', 'on']).withAccess(ea.STATE_SET).withDescription('Fan switch'),
        tuya.exposes.countdown().withValueMin(0).withValueMax(43200).withUnit('s').withDescription('Max ON time in seconds'),
        e.numeric('fan_speed', ea.STATE_SET).withValueMin(1).withValueMax(5).withValueStep(1)
            .withDescription('Speed off the fan'),
    ],
    meta: {
        tuyaDatapoints: [
            [1, 'state', tuya.valueConverter.onOff],
            [2, 'countdown', tuya.valueConverter.countdown],
            [3, 'fan_speed', tuya.valueConverterBasic
                .lookup({'1': tuya.enum(0), '2': tuya.enum(1), '3': tuya.enum(2), '4': tuya.enum(3), '5': tuya.enum(4)})],
            [11, 'power_on_behavior', tuya.valueConverterBasic.lookup({'off': tuya.enum(0), 'on': tuya.enum(1)})],
            [5, 'status_indication', tuya.valueConverter.onOff],
        ],
    },
    whiteLabel: [
        {vendor: 'Lerlink', model: 'T2-Z67/T2-W67'},
    ],
};
```

Koenkk, 2024-04-14:
Could you make a pull request?

GGG-KILLER, 2024-04-14:
I'm currently trying to make both the light switch and fan switch show up in home assistant and having issues, if I manage to fix that I'll make a PR, but in its current state it isn't very helpful in my opinion.

rmribeiro2024, 2024-04-25:
> @Koenkk this one might be a bit different as it includes a light switch as well, just bought one of these, added the external definition according to your commit and there's only an on/off for the fan, but not the light.
> 
> Update: My bad, there _is_ a light switch in the definition but it seems to turn the fan on and off at least.

Hello! same problem here. Z2m only shows the fan switch, but the light does not appear. However, he switch has both functions.

Try pairing with SmartLife and working normally.

Any solution for this problem?

rmribeiro2024, 2024-04-25:
> The following worked for me in the end:
>
> ```js
> const tuya = require('zigbee-herdsman-converters/lib/tuya');
> const exposes = require('zigbee-herdsman-converters/lib/exposes');
>
> const e = [...]

Yes, thank you! Now it works. Only an error occurs when we set the fan to speed 5.

Zigbee2MQTT:error 2024-04-25 16:29:39: Exception while calling fromZigbee converter: Value '5' is not allowed, expected one of [object Object],[object Object],[object Object],[object Object],[object Object]}

anyone can helpe please?

GGG-KILLER, 2024-04-26:
> Yes, thank you! Now it works. Only an error occurs when we set the fan to speed 5.
>
> Zigbee2MQTT:error 2024-04-25 16:29:39: Exception while calling fromZigbee converter: Value '5' is not allowed, [...]

That's what I've been trying to fix, along with making the light switch show up on Home Assistant, however I've been unable to.

rmribeiro2024, 2024-04-26:
> That's what I've been trying to fix, along with making the light switch show up on Home Assistant, however I've been unable to.

You still can't control the light through the home assistant? Strange, here I can control both the light and the fan normally with the code you shared. The only strange thing is that message when setting it to speed 5.

aroslanov, 2024-04-27:
Also, in addition to the error above, it reports wrong fan speed value (value+1) when I tap on fan power on/off button.

Koenkk, 2024-04-28:
It doesn't work with `lookup({'1': tuya.enum(0), '2': tuya.enum(1), '3': tuya.enum(2), '4': tuya.enum(3), '5': tuya.enum(4)})`?

GGG-KILLER, 2024-04-28:
> It doesn't work with `lookup({'1': tuya.enum(0), '2': tuya.enum(1), '3': tuya.enum(2), '4': tuya.enum(3), '5': tuya.enum(4)})`?

No, it produces the error @rmribeiro2024 mentioned above:
> Zigbee2MQTT:error 2024-04-25 16:29:39: Exception while calling fromZigbee converter: Value '5' is not allowed, expected one of [object Object],[object Object],[object Object],[object Object],[object Object]}

Koenkk, 2024-04-28:
And with `lookup({'1': tuya.enum(0), '2': tuya.enum(1), '3': tuya.enum(2), '4': tuya.enum(3), '5': tuya.enum(4)}, '5')` ?

aroslanov, 2024-04-28:
Exception seems to be gone on my side, thank you, @Koenkk. Still misreading on LED and MQTT report, but I am unsure if this is a hardware error.

Koenkk, 2024-04-29:
Integrated the fix!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

brkr1, 2024-12-15:
Need help with my zigbee device

I have this device: https://www.zigbee2mqtt.io/devices/TS0601_fan_5_levels_and_light_switch.html#tuya-ts0601_fan_5_levels_and_light_switch

When I bought it, it worked greatly. Starting a week ago it started behaving oddly. 
Now if I want to turn on the fan, for example, I gotta run it twice. 
Sometimes I turn of the fan and it will also turn on the light. Very weird!

I’ve tried so far, removing and configuring it from scratch, restoring for backup etc

What else can I do?