Koenkk/zigbee2mqtt #21788: [New device support]: Moes Zigbee Water Valve Timer

BroTo-Typ, 2024-03-13:
### Link

https://de.aliexpress.com/item/1005006336288215.html?spm=a2g0o.productlist.main.33.2363pzFfpzFfUT&algo_pvid=f3490478-2ad5-40af-81a9-b59ab7b3dbb4&algo_exp_id=f3490478-2ad5-40af-81a9-b59ab7b3dbb4-16&pdp_npi=4%40dis%21EUR%2173.24%2123.95%21%21%21561.66%21183.69%21%40213db7b317103563157986698e02f9%2112000036814287930%21sea%21DE%21846559442%21&curPageLogUid=xAj67NX2yJ9P&utparam-url=scene%3Asearch%7Cquery_from%3A

### Database entry

{"id":26,"type":"EndDevice","ieeeAddr":"0xa4c138ea5d9d7dae","nwkAddr":10691,"manufId":4417,"manufName":"_TZ3000_cjfmu5he","powerSource":"Battery","modelId":"TS0049","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":0,"inClusterList":[3,4,5,1,6,57345,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65534":0,"stackVersion":0,"dateCode":"","manufacturerName":"_TZ3000_cjfmu5he","zclVersion":3,"appVersion":72,"modelId":"TS0049","powerSource":3}},"genOnOff":{"attributes":{"onOff":0}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-708457359},"lastSeen":1710356452534}

### Comments

Well - it has successfully paired, shows a very similar gadget but I don´t know what to do, I cannot switch the valve on or off.
It shows On/Off but neither the device reacts when switching it via zigbee2mqtt nor does zigbee2mqtt recognize if the device is switched on/off directly.

DEBUG-Logs with the device filtered show this:

debug 2024-03-13 20:04:55Received Zigbee message from '0xa4c138ea5d9d7dae', type 'raw', cluster 'manuSpecificTuya_3', data '{"data":[8,175,10,64,208,72,3,12,2],"type":"Buffer"}' from endpoint 1 with groupID 0
debug 2024-03-13 20:05:01Received Zigbee message from '0xa4c138ea5d9d7dae', type 'raw', cluster 'manuSpecificTuya_3', data '{"data":[8,177,10,64,208,72,6,15,0,0,0,2],"type":"Buffer"}' from endpoint 1 with groupID 0
debug 2024-03-13 20:05:08Received Zigbee message from '0xa4c138ea5d9d7dae', type 'raw', cluster 'manuSpecificTuya_3', data '{"data":[8,178,10,64,208,72,6,11,0,0,2,88],"type":"Buffer"}' from endpoint 1 with groupID 0
debug 2024-03-13 20:05:14Received Zigbee message from '0xa4c138ea5d9d7dae', type 'raw', cluster 'manuSpecificTuya_3', data '{"data":[8,179,10,64,208,72,3,12,1],"type":"Buffer"}' from endpoint 1 with groupID 0
debug 2024-03-13 20:05:20Received Zigbee message from '0xa4c138ea5d9d7dae', type 'raw', cluster 'manuSpecificTuya_3', data '{"data":[8,181,10,64,208,72,3,12,2],"type":"Buffer"}' from endpoint 1 with groupID 0
debug 2024-03-13 20:05:27Received Zigbee message from '0xa4c138ea5d9d7dae', type 'raw', cluster 'manuSpecificTuya_3', data '{"data":[8,183,10,64,208,72,6,15,0,0,0,1],"type":"Buffer"}' from endpoint 1 with groupID 0
debug 2024-03-13 20:05:33Received Zigbee message from '0xa4c138ea5d9d7dae', type 'raw', cluster 'manuSpecificTuya_3', data '{"data":[8,184,10,64,208,72,6,11,0,0,2,88],"type":"Buffer"}' from endpoint 1 with groupID 0
debug 2024-03-13 20:05:40Received Zigbee message from '0xa4c138ea5d9d7dae', type 'raw', cluster 'manuSpecificTuya_3', data '{"data":[8,185,10,64,208,72,3,12,1],"type":"Buffer"}' from endpoint 1 with groupID 0
debug 2024-03-13 20:05:46Received Zigbee message from '0xa4c138ea5d9d7dae', type 'raw', cluster 'manuSpecificTuya_3', data '{"data":[8,187,10,64,208,72,3,12,2],"type":"Buffer"}' from endpoint 1 with groupID 0
debug 2024-03-13 20:05:53Received Zigbee message from '0xa4c138ea5d9d7dae', type 'raw', cluster 'manuSpecificTuya_3', data '{"data":[8,189,10,64,208,72,6,15,0,0,0,4],"type":"Buffer"}' from endpoint 1 with groupID 0
debug 2024-03-13 20:05:59Received Zigbee message from '0xa4c138ea5d9d7dae', type 'raw', cluster 'manuSpecificTuya_3', data '{"data":[8,190,10,64,208,72,6,11,0,0,2,88],"type":"Buffer"}' from endpoint 1 with groupID 0
debug 2024-03-13 20:06:05Received Zigbee message from '0xa4c138ea5d9d7dae', type 'raw', cluster 'manuSpecificTuya_3', data '{"data":[8,191,10,64,208,72,3,12,1],"type":"Buffer"}' from endpoint 1 with groupID 0
debug 2024-03-13 20:06:12Received Zigbee message from '0xa4c138ea5d9d7dae', type 'raw', cluster 'manuSpecificTuya_3', data '{"data":[8,193,10,64,208,72,3,12,2],"type":"Buffer"}' from endpoint 1 with groupID 0
debug 2024-03-13 20:06:18Received Zigbee message from '0xa4c138ea5d9d7dae', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
debug 2024-03-13 20:06:18Received Zigbee message from '0xa4c138ea5d9d7dae', type 'raw', cluster 'manuSpecificTuya_3', data '{"data":[8,195,10,64,208,72,6,15,0,0,0,2],"type":"Buffer"}' from endpoint 1 with groupID 0
debug 2024-03-13 20:06:25Received Zigbee message from '0xa4c138ea5d9d7dae', type 'raw', cluster 'manuSpecificTuya_3', data '{"data":[8,196,10,64,208,72,6,11,0,0,2,88],"type":"Buffer"}' from endpoint 1 with groupID 0
debug 2024-03-13 20:06:31Received Zigbee message from '0xa4c138ea5d9d7dae', type 'raw', cluster 'manuSpecificTuya_3', data '{"data":[8,197,10,64,208,72,3,12,1],"type":"Buffer"}' from endpoint 1 with groupID 0
debug 2024-03-13 20:06:37Received Zigbee message from '0xa4c138ea5d9d7dae', type 'raw', cluster 'manuSpecificTuya_3', data '{"data":[8,198,10,64,208,72,6,11,0,0,2,88],"type":"Buffer"}' from endpoint 1 with groupID 0
debug 2024-03-13 20:06:44Received Zigbee message from '0xa4c138ea5d9d7dae', type 'raw', cluster 'manuSpecificTuya_3', data '{"data":[8,199,10,64,208,72,3,12,2],"type":"Buffer"}' from endpoint 1 with groupID 0

Would be nice if I could get it working but I´m afraid I´ll need help.
(and hopefully I have given enough information)

Thank you very much in advance!

### External definition

```shell
const {battery, onOff} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['TS0049'],
    model: 'TS0049',
    vendor: '_TZ3000_cjfmu5he',
    description: 'Automatically generated definition',
    extend: [battery(), onOff({"powerOnBehavior":false})],
    meta: {},
};

module.exports = definition;
```

BroTo-Typ, 2024-03-14:
Update:
Changing the Endpoint to cluster "OnOff" and attribute "onOff" and pushig read leads to this (button pressed once, button pressed once more):
info 2024-03-14 06:33:28Read result of 'genOnOff': {"onOff":1}
info 2024-03-14 06:33:28Read result of 'genOnOff': {"onOff":0}

So there seems to be a kind of communication...

BroTo-Typ, 2024-03-17:
Any help how to get the valve working would be highly appreciated.
Thank you in advance!

BroTo-Typ, 2024-03-18:
Edit:
New Debug Message received (seems to bei once per day?):
 2024-03-18 08:24:12Received Zigbee message from '0xa4c138ea5d9d7dae', type 'read', cluster 'genTime', data '["localTime"]' from endpoint 1 with groupID 0

BroTo-Typ, 2024-03-18:
UPDATE:
I'm still encountering an issue with this Zigbee device. The device, identified as '0xa4c138ea5d9d7dae' and described as "Garden Irrigation," seems to have a configuration problem when used with an external converter in Zigbee2MQTT and I still cannot solve it.

Problem:
While the device is successfully receiving and transmitting data, it fails to respond to commands sent through Zigbee2MQTT. However, interestingly, it does respond appropriately when operated directly.

Configuration:
I've configured the device with an external converter named "MoesWaterValve.js" in Zigbee2MQTT - thanks to ChatGPT which did a good job. The configuration includes mappings for the device's capabilities, such as turning it on and off.
----> external converter at the end

Observations:
When I send commands like "ON" or "OFF" to the device via Zigbee2MQTT, it acknowledges the commands by updating its status and publishing the changes via MQTT. -> see LOG-File at the end
Despite receiving the commands, the device does not actually act on them. It remains unresponsive to the commands sent via Zigbee2MQTT.

Troubleshooting Attempts:
Restart: I've tried restarting both the Zigbee2MQTT service and the device itself, but the issue persists.

At this stage, I'm puzzled as to why the device isn't responding to commands sent through Zigbee2MQTT despite correctly receiving and acknowledging them. Any insights or suggestions on troubleshooting steps would be greatly appreciated. Thank you!
------------------------------------------------------------------------------
External converter:
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const e = exposes.presets;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    fingerprint: [
        {
            modelID: 'TS0049',
            manufacturerName: '_TZ3000_cjfmu5he',
        },
    ],
    model: 'TS0049_new',
    vendor: 'TuYa',
    description: 'Fill in a description of the device here',
    fromZigbee: [fz.onOff],
    toZigbee: [tz.onOff],
    exposes: [
        e.switch().withEndpoint('1'),
    ],
    meta: {
        // All datapoints go in here if needed
        tuyaDatapoints: [
            // You can add specific Tuya datapoints if required
        ],
    },
    configure: async (device, coordinatorEndpoint) => {
        const endpoint = device.getEndpoint(1);
        await endpoint.command('genOnOff', 'on', { onOff: 0 });
    },
};

module.exports = definition;

------------------------------------------------------------------------------
LOGS:
info 2024-03-18 13:29:51MQTT publish: topic 'zigbee2mqtt/0xa4c138ea5d9d7dae', payload '{"countdown":67,"linkquality":91,"state":"OFF"}'
debug 2024-03-18 13:29:51Received Zigbee message from '0xa4c138ea5d9d7dae', type 'commandActiveStatusReportAlt', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":1,"dp":101}],"seq":1}' from endpoint 1 with groupID 0
info 2024-03-18 13:29:51MQTT publish: topic 'zigbee2mqtt/0xa4c138ea5d9d7dae', payload '{"countdown":67,"linkquality":87,"state":"ON"}'
debug 2024-03-18 13:29:51Received Zigbee message from '0xa4c138ea5d9d7dae', type 'commandActiveStatusReportAlt', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":101}],"seq":1}' from endpoint 1 with groupID 0
info 2024-03-18 13:29:51MQTT publish: topic 'zigbee2mqtt/0xa4c138ea5d9d7dae', payload '{"countdown":67,"linkquality":69,"state":"OFF"}'
debug 2024-03-18 13:29:52Received Zigbee message from '0xa4c138ea5d9d7dae', type 'commandActiveStatusReportAlt', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":1,"dp":101}],"seq":1}' from endpoint 1 with groupID 0
info 2024-03-18 13:29:52MQTT publish: topic 'zigbee2mqtt/0xa4c138ea5d9d7dae', payload '{"countdown":67,"linkquality":69,"state":"ON"}'
debug 2024-03-18 13:29:52Received Zigbee message from '0xa4c138ea5d9d7dae', type 'commandActiveStatusReportAlt', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":101}],"seq":1}' from endpoint 1 with groupID 0
info 2024-03-18 13:29:52MQTT publish: topic 'zigbee2mqtt/0xa4c138ea5d9d7dae', payload '{"countdown":67,"linkquality":76,"state":"OFF"}'

BroTo-Typ, 2024-03-20:
@Koenkk I am really sorry for bothering you but is there a chance for getting this device working?
The device is communicating (and showing in the log when it is switched directly) but not reacting when switched via zigbee2mqtt.
When I push "read" (endpoint 1, cluster OnOff, attribute onOff) the result is: 
```
debug 2024-03-20 09:38:17Received MQTT message on 'zigbee2mqtt/0xa4c138ea5d9d7dae/1/set' with data '{"read":{"attributes":["onOff"],"cluster":"genOnOff","options":{}}}'
debug 2024-03-20 09:38:17Publishing 'set' 'read' to '0xa4c138ea5d9d7dae'
debug 2024-03-20 09:38:19Received Zigbee message from '0xa4c138ea5d9d7dae', type 'readResponse', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
debug 2024-03-20 09:38:21Received MQTT message on 'zigbee2mqtt/0xa4c138ea5d9d7dae/1/set' with data '{"read":{"attributes":["onOff"],"cluster":"genOnOff","options":{}}}'
debug 2024-03-20 09:38:21Publishing 'set' 'read' to '0xa4c138ea5d9d7dae'
debug 2024-03-20 09:38:23Received Zigbee message from '0xa4c138ea5d9d7dae', type 'readResponse', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
```

--------------------------------
When I push "write" (endpoint 1, cluster OnOff, attribute onOff) the result is:

```
debug 2024-03-20 09:36:45Received MQTT message on 'zigbee2mqtt/0xa4c138ea5d9d7dae/1/set' with data '{"write":{"cluster":"genOnOff","options":{},"payload":{"onOff":1}}}'
debug 2024-03-20 09:36:45Publishing 'set' 'write' to '0xa4c138ea5d9d7dae'
error 2024-03-20 09:36:46Publish 'set' 'write' to '0xa4c138ea5d9d7dae' failed: 'Error: Write 0xa4c138ea5d9d7dae/1 genOnOff({"onOff":1}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'READ_ONLY')'
debug 2024-03-20 09:36:46Error: Write 0xa4c138ea5d9d7dae/1 genOnOff({"onOff":1}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'READ_ONLY') at Endpoint.checkStatus (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:319:28) at Endpoint.write (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:392:22) at Object.convertSet (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/converters/toZigbee.ts:213:13) at Publish.onMQTTMessage (/opt/zigbee2mqtt/lib/extension/publish.ts:259:36) at EventEmitter.wrappedCallback (/opt/zigbee2mqtt/lib/eventBus.ts:174:17)
```

I would be very happy if you could give me a hint.
Unfortunately I haven't had the possibility to "sniff" any furter communication yet...
Best regards

Koenkk, 2024-03-20:
if you use:

```
fromZigbee: [fz.on_off],
toZigbee: [tz.on_off],
exposes: [
e.switch()
],
```

Can you switch it via the frontend -> exposes tab?

BroTo-Typ, 2024-03-23:
You are a genius! Now its switching on and off.
Well - I don´t have any automatic off-timer but at least its working!
I will do some more experiments but I can use the valve.
Thank you very much!

hyperlogic-dev, 2024-03-30:
Hi, could you provide the full configuration for this device? I tried the script and can turn it on and off, but I can't see the battery level.

BroTo-Typ, 2024-03-30:
I also can´t see the battery level but did not have the time the last weeks to try anything.
Will post here when I´ve found out more :-)

gawej, 2024-04-01:
did you managed to turn off and on that valve? how?

joehun, 2024-04-01:
When integrating the converter in my HA installation I get the following error message when starting Zigbee2MQTT:

`error 2024-04-01 22:50:02: Failed to load external converter file 'MoesWaterValve.js' (Cannot read properties of undefined (reading 'options'))`

Has anybody an idea whatI did wrong?

BroTo-Typ, 2024-04-03:
Well - my external configuration is as following:

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const e = exposes.presets;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
fingerprint: [
{
modelID: 'TS0049',
manufacturerName: '_TZ3000_cjfmu5he',
},
],
model: 'TS0049_new',
vendor: 'TuYa',
description: 'Irrigation Valve',
fromZigbee: [fz.on_off],
toZigbee: [tz.on_off],
exposes: [
e.switch()
],
meta: {
// All datapoints go in here if needed
tuyaDatapoints: [
// You can add specific Tuya datapoints if required
],
},
configure: async (device, coordinatorEndpoint) => {
const endpoint = device.getEndpoint(1);
await endpoint.command('genOnOff', 'on', { onOff: 0 });
},
};

module.exports = definition;

Maybe this will help?

joehun, 2024-04-03:
Perfekt, this works!

Koenkk, 2024-04-04:
Isn't this device working out-of-the-box (don't use any external converters) with z2m 1.36.1?

joehun, 2024-04-05:
No, it is not working out of the Box in Version 1.36.1. I have tested it just again.

gawej, 2024-04-05:
I never did external configuration. I understand that I have to put that code to the file and somehow indicate that this code is here?

Wysłano z programu Outlook dla systemu Android<https://aka.ms/AAb9ysg>

Koenkk, 2024-04-06:
What doesn't work in 1.36.1?

hyperlogic-dev, 2024-04-06:
> What doesn't work in 1.36.1?

In 1.13.1 the device can be paired successfully, but I am unable to control it or see any of its states. (Value is null)

Koenkk, 2024-04-07:
Does it work with the [generated external converter]?

gawej, 2024-04-09:
> No, it is not working out of the Box in Version 1.36.1. I have tested it just again.

could you share a solution step by step how to create and make external file and make it working?

gawej, 2024-04-09:
> Does it work with the [generated external converter](https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_2-creating-the-external-definition)?

i tried to generate external converter but for me it is not working. im new with HA and z2m and maybe i did something wrong.

Koenkk, 2024-04-10:
What's not working?

gawej, 2024-04-10:
> What's not working?

nothing. device is just recognized as tuya. i cant turn on or off. i cant see status. i cant see baterry status.

Koenkk, 2024-04-11:
Could you provide a screenshot of the device -> about and exposes tab?

gawej, 2024-04-11:
[image]
here is is. i also added how it really looks like.

Koenkk, 2024-04-13:
Sorry, I meant when not using an external definition.

gawej, 2024-04-15:
for me external is not working. but im quite new with HA and Z2M and maybe i dont know what tu use it properly. 

maybe @joehun should reply and could help all the guys waiting for working walve.

joehun, 2024-04-15:
@gawej:
All needed informations are within this discussion. You have to do the following:

- if you have already added the device, delete it from configuration 
- Create file "moes0049_ext.js" in the home directory of zigbee2mqtt.
- place the content of @BroTo-Typ post into this file.(take the latest content)
- goto zigbee2mqtt configuration and add "moes0049_ext.js" as external converter
- Restart Z2M
- Add your device

gawej, 2024-04-15:
@Koenkk 
for me external is not working so you can assume that my confirguration is not external. this is how it looks out of the box.

lornz, 2024-04-16:
> @gawej: All needed informations are within this discussion. You have to do the following:
>
> * if you have already added the device, delete it from configuration
> * Create file "moes0049_ext.js" in the [...]

I followed these steps and it is working for me. That means on/off is working. The timer function seems to be lost which isn't a problem for me because I'm using HomeAssistant.

gawej, 2024-04-16:
@lornz 
for me is not working. maybe i dont know how to add moes.js as converter. this is first line of my configuration of z2m:

homeassistant: true
advanced:
  external_converters:
    - moes.js

this is the way? :)

joehun, 2024-04-16:
It is very easy: open settings in Z2M, goto external converters, add the name of your script, submit

[image]

gawej, 2024-04-16:
@joehun thanks a lot. i really didnt know about that place to submit external definition... 

the switch now is working

@Koenkk 
my screen which i uploaded was without external definition. are you able to make an update to have all the features? if you need any screens let me know.

hyperlogic-dev, 2024-04-24:
Hi @Koenkk , I have tried your method as follows.
1. Go into the device  and go to the Dev Console tab.
2. Press the generate_external_definition button and copy the code to create a external converters.
3. Reboot HA
I found that it can be used correctly, including power on and off, battery status.

Koenkk, 2024-04-25:
@hyperlogic-dev great! Could you paste the generated definition here?

hyperlogic-dev, 2024-04-25:
Sure!, This code i'am use in external definition.

const {identify, battery, onOff} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['TS0049'],
    model: 'TS0049',
    vendor: '_TZ3000_cjfmu5he',
    description: 'Automatically generated definition',
    extend: [identify(), battery(), onOff({"powerOnBehavior":false})],
    meta: {},
};

module.exports = definition;

hyperlogic-dev, 2024-04-25:
> @hyperlogic-dev great! Could you paste the generated definition here?

Sure!, This code i'am use in external conveters file

const {identify, battery, onOff} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {

    zigbeeModel: ['TS0049'],

    model: 'TS0049',

    vendor: '_TZ3000_cjfmu5he',

    description: 'Automatically generated definition',

    extend: [identify(), battery(), onOff({"powerOnBehavior":false})],

    meta: {},

};

module.exports = definition;

Koenkk, 2024-04-27:
@hyperlogic-dev added, thanks!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-04-27 by Koenkk

gawej, 2024-04-29:
Thanks !
So to have updated valve working out of the box we should delete external definition? Or do something else?

Btw. With external definition this valve is working for 10min and turn off automatically. This will change with update?

Wysłano z programu Outlook dla systemu Android<https://aka.ms/AAb9ysg>

enricco1, 2024-05-01:
> * home directory

Where can I get the "moes0049_ext.js" file?
And where is the home directory?

enricco1, 2024-05-01:
> @gawej: All needed informations are within this discussion. You have to do the following:
>
> * if you have already added the device, delete it from configuration
> * Create file "moes0049_ext.js" in the [...]

Where can I get the "moes0049_ext.js" file?
And where is the home directory?

enricco1, 2024-05-01:
> ![Screenshot 2567-04-24 at 11 47 20](https://private-user-images.githubusercontent.com/33412551/325083207-626ec96c-643a-4f79-9616-4c25f88f126a.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTQ1MjM1MzgsIm5iZiI6MTcxNDUyMzIzOCwicGF0aCI6Ii8zMzQxMjU1MS8zMjUwODMyMDctNjI2ZWM5NmMtNjQzYS00Zjc5LTk2MTYtNGMyNWY4OGYxMjZhLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MDElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTAxVDAwMjcxOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWIwY2FjYjQ0OTlmMGQ0YzdhYmZmNDhjYWQ5YzlmZDkwNmRlZjY4OGU2M2I4NmVjNjBjMTNmOGI0YTZhYTExMzcmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.AmhqYuouHE1xzac1X8Mzpo2vhVybYWi-alPw_9Fgs40) Hi @Koenkk , I have tried your method as follows.
> 
> 1. Go into the device  and go to the Dev Console tab.
> 2. Press the generate_external_definition button and copy the code to create a external converters.
> 3. Reboot HA
>    I found that it can be used correctly, including power on and off, battery status.

Do I just copy and put the code in the generate_external?
I copied it, but I can't.
Do I have to do something else?

tomj44300, 2024-06-02:
hi i have bought a MOES copy , yellow valve version of gawej 
[image]
I have tried same configuration but  it didn't works , action on/off from valve is seen on z2m , from ha action on works but i can't set off 
how i can fix it ?

my external definition 

const {identify, battery, onOff} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['TS0049'],
    model: 'TS0049',
    vendor: '_TZ3000_kz1anoi8',
    description: 'Automatically generated definition',
    extend: [identify(), battery(), onOff({"powerOnBehavior":false})],
    meta: {},
};

module.exports = definition;

ariboehme, 2024-07-19:
@tomj44300 

I have the same device and the code that made it work (on and off) in my case was the following.
But I can´t set the countdown timer. Let me know if you get anything.

//---------------------------------------
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const e = exposes.presets;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
  fingerprint: [
    {
      modelID: 'TS0049',
      manufacturerName: '_TZ3000_kz1anoi8',
    },
  ],
  model: 'TS0049',
  vendor: '_TZ3000_kz1anoi8',
  description: 'Irrigation Valve',
  fromZigbee: [fz.on_off],
  toZigbee: [tz.on_off],
  exposes: [
  e.switch()
],
meta: {
  // All datapoints go in here if needed
  tuyaDatapoints: [
    // You can add specific Tuya datapoints if required
  ],
  },
  configure: async (device, coordinatorEndpoint) => {
  const endpoint = device.getEndpoint(1);
  await endpoint.command('genOnOff', 'on', { onOff: 0 });
  },
};

module.exports = definition;