Koenkk/zigbee2mqtt #21790: Woox R7060 - error tuya.firstDpValue

floow77, 2024-03-14:
### What happened?

When I change status (off/on), I had this error:

`Error 2024-03-14 08:14:49Exception while calling fromZigbee converter: tuya.firstDpValue is not a function}`

### What did you expect to happen?

No error

### How to reproduce it (minimal and precise)

Change status : off to on or off to on

### Zigbee2MQTT version

1.36.0-1

### Adapter firmware version

2024.03.0

### Adapter

Conbee2

### Setup

HA addon

### Debug log

`Error 2024-03-14 08:14:49Exception while calling fromZigbee converter: tuya.firstDpValue is not a function}`

Llenas09, 2024-04-07:
I also have problems with the device.

After syncing it, it works a couple of times then it doesn't respond anymore.

Matzebhv, 2024-04-13:
Same problem here.   The funny thing is, if you use the button in the device detail section it works.

Koenkk, 2024-04-14:
Could you provide the debug log of this?

See [this](https://www.zigbee2mqtt.io/guide/usage/debug.html) on how to enable debug logging.

Matzebhv, 2024-04-23:
Have tried to enable debug with no success.

validating /root/Documents/zigbee2mqtt/docker-compose.yml: (root) Additional property advanced is not allowed

Llenas09, 2024-11-03:
Hi, attached the log. The device is 0x003c84fffe0de849

[z2m-log.zip](https://github.com/user-attachments/files/17612394/z2m-log.zip)

closed 2025-01-13 by github-actions[bot]