Koenkk/zigbee2mqtt #21791: Mercator Ikuu SSWM-DIMZ warning messages in log.

scottyphillips, 2024-03-14:
### What happened?

Good evening, I paired a Mercator Ikuu SSWM-DIMZ dimmer mechnism to Z2M this evening. The device works fine but it does throw a warning message about 'Received unexpected Tuya DataPoint #2' when using the dimming function. 

`Zigbee2MQTT:warn  2024-03-14 19:45:52: zigbee-herdsman-converters:tuya_dimmer: Received unexpected Tuya DataPoint #2 from 0xcc86ecfffe9a118c with raw data '{"dp":2,"datatype":2,"data":{"type":"Buffer","data":[0,0,2,86]}}': type='commandDataResponse', datatype='value', value='598', known DP# usage: ["heatingSetpoint","coverPosition","eardaDimmerLevel","moesHold","moesSheatingSetpoint","silvercrestChangeMode","tuyaSabCO2","tuyaSahkMP25","tuyaSabCO","moes105DimmerLevel1","trsSensitivity","trsfSensitivity","tvMode","nousHumidity","tthHumidity","tIlluminanceLux","evanellMode","AM02PercentControl","x5hMode","connecteMode","tshpscSensitivity","alectoSmokeValue","zsMode"]`

Other then error message, the device appears to be working fine. 

### What did you expect to happen?

No warning messages in console when using dimmer. 

### How to reproduce it (minimal and precise)

Use dimming function on SSWM-DIMZ either physically or in app. Warning message appears. 

### Zigbee2MQTT version

1.36.0-1

### Adapter firmware version

0x26780700

### Adapter

Conbee II

### Setup

Home Assistant OS Add-on

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/14599910/log.txt)

scottyphillips, 2024-03-16:
I have completely resolved all the identified issues myself using the following external converter. The current 'TS0601_dimmer' needs to be updated from the legacy code. 

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const legacy = require('zigbee-herdsman-converters/lib/legacy')

const definition = {
    fingerprint: [
        {modelID: 'TS0601', manufacturerName: '_TZE200_whpb9yts'},
        {modelID: 'TS0601', manufacturerName: '_TZE200_ebwgzdqq'},
        {modelID: 'TS0601', manufacturerName: '_TZE200_ctq0k47x'},
        {modelID: 'TS0601', manufacturerName: '_TZE200_9i9dt8is'},
        {modelID: 'TS0601', manufacturerName: '_TZE200_dfxkcots'},
        {modelID: 'TS0601', manufacturerName: '_TZE200_w4cryh2i'},
        {modelID: 'TS0601', manufacturerName: '_TZE200_ojzhk75b'},
        {modelID: 'TS0601', manufacturerName: '_TZE200_swaamsoy'},
        {modelID: 'TS0601', manufacturerName: '_TZE200_3p5ydos3'},
        {modelID: 'TS0601', manufacturerName: '_TZE200_9cxuhakf'},
        {modelID: 'TS0601', manufacturerName: '_TZE200_a0syesf5'},
    ],
    model: 'TS0601_dimmer_test',
    vendor: 'TuYa',
    description: 'Zigbee smart dimmer',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    configure: tuya.configureMagicPacket,
    exposes: [tuya.exposes.lightBrightnessWithMinMax()],
    meta: {
            tuyaDatapoints: [
                [1, 'state', tuya.valueConverter.onOff, {skip: tuya.skip.stateOnAndBrightnessPresent}],
                [2, 'brightness', tuya.valueConverter.scale0_254to0_1000],
                [3, 'min_brightness', tuya.valueConverter.scale0_254to0_1000],
                [5, 'max_brightness', tuya.valueConverter.scale0_254to0_1000],
            ],
        },
        whiteLabel: [
        {vendor: 'Larkkey', model: 'ZSTY-SM-1DMZG-EU'},
        {vendor: 'Earda', model: 'EDM-1ZAA-EU'},
        {vendor: 'Earda', model: 'EDM-1ZAB-EU'},
        {vendor: 'Earda', model: 'EDM-1ZBA-EU'},
        {vendor: 'Mercator Ikuü', model: 'SSWD01'},
        {vendor: 'Moes', model: 'ZS-USD'},
        {vendor: 'Moes', model: 'EDM-1ZBB-EU'},
        tuya.whitelabel('Mercator Ikuü', 'SSWM-DIMZ', 'Switch Mechanism', ['_TZE200_9cxuhakf']),
        tuya.whitelabel('Mercator Ikuü', 'SSWRM-ZB', 'Rotary dimmer mechanism', ['_TZE200_a0syesf5']),
    ],
};

module.exports = definition;
```

Koenkk, 2024-03-17:
Could you make a pull request by pressing the edit icon [here](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/tuya.ts)?

scottyphillips, 2024-03-17:
Hi @Koenkk. Yeah no worries I’ll draft one tonight hopefully. 

I really appreciate your massive efforts, this is a great project, thank you!!! 🙏

bjammin, 2024-03-18:
I can confirm that this fixes this and other problems for the rotary version of the dimmer as well.

I support merging.

closed 2024-10-15 by github-actions[bot]