Koenkk/zigbee2mqtt #21795: Develco SMSZB-120 sensors show "Null" instead of value

Subito, 2024-03-14:
### What happened?

Configuring a new Smoke Sensor fails with 
```
error 2024-03-14 13:59:52Failed to configure 'Esszimmer Rauchmelder', attempt 2 (Error: Bind 0x0015bc0031021b1e/35 genBasic from '0x00212effff0db76b/1' failed (Error: status: 140) at DeconzAdapter.bind (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:776:19) at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:551:13) at Object.bind (/app/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:43:9) at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/devices/develco.ts:495:13) at Configure.configure (/app/lib/extension/configure.ts:122:13) at Configure.onMQTTMessage (/app/lib/extension/configure.ts:55:21) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17))
```

Afterwards some of the Sensors report "Null" instead of the value:

```
{
    "battery": 100,
    "battery_low": false,
    "linkquality": 186,
    "max_duration": 240,
    "restore_reports": true,
    "smoke": false,
    "supervision_reports": true,
    "test": false,
    "update": {
        "installed_version": 262152,
        "latest_version": 262152,
        "state": "idle"
    },
    "update_available": false,
    "voltage": 3000,
    "alarm": null,
    "fault": null,
    "reliability": null,
    "temperature": null
}
```

Might be related to #20850, #17803, #14997, #16244 and #14414

### What did you expect to happen?

Configuring works, the Device is listed as [supported](https://www.zigbee2mqtt.io/devices/SMSZB-120.html).

### Zigbee2MQTT version

1.36.0

### Adapter firmware version

0x26500900

### Adapter

ConBee3

### Setup

Homeassistant Add-on, Rasperry PI 4

derspe, 2024-10-05:
Same problem here

georgbachmann, 2025-01-01:
For me it's `battery` and `voltage` that's null. Is there anything to do about it?

mikke49, 2025-01-06:
> For me it's `battery` and `voltage` that's null. Is there anything to do about it?

I had the same issue, using Z2M. Went to the Exposes tab, then hit the Refresh button on the Battery item, and these two were populated. Don't know yet if they now will be updated like the other entities, time will show.

[image]

closed 2025-03-15 by github-actions[bot]