Koenkk/zigbee2mqtt #21797: TuYa WHD02 switch not reacting to "on_time"

musca76, 2024-03-14:
### What happened?

I am using a Sonoff 2652P coordinator with all Tuya switches via Zigbee2MQTT and Mosquito (installed on a Synology virtual machine). Everything works as expected except one thing: on tuya app I had the option to configure a switch to automatically turn off after a specific number of seconds. Every time I turn ON the switch it will automatically turn off after 5 seconds. This was somehow hardcoded in the switch. I have tried to do the same in HA with no success.
I have found this topic: [TuYa WHD02 control via MQTT | Zigbee2MQTT 2] (https://www.zigbee2mqtt.io/devices/WHD02.html)
but It does not work . 

When I use just simple  on/off payloads works perfectly. When I use {"state" : "ON", "on_time": 5, "off_wait_time": 2} I get  "2024-03-10 20:19:41Invalid message ‘null’, skipping…" error.
When I use both {"state" : "ON", "on_time": 5, "off_wait_time": 2} I do not get any error but is not working neither.

Anyone can help?

Wall switch module (WHD02), 
Firmware: 0122052017 
Zigbee Model TS0001
Zigbee Manufacturer _TZ3000_skueekg3

Connected via [Zigbee2MQTT Bridge]

Hub: SONOFF Zigbee 3.0 USB Dongle Plus, ZBDongle-P TI CC2652P + CP2102(N)
Sonoff firmware : 20230507

NAS: Synology DSM 7.1.1-42962 Update 6,

HA Core 2024.3.0; Supervisor 2024.02.1; Operating System 12.0; Frontend 0240306.0;

Please bear with me - I am new to this. 

### What did you expect to happen?

I expect every time I turn the switch ON (using from HA or from the wall switch) the switch shoud stay ON for 5 seconds and then it will turn OFF. 

### Zigbee2MQTT version

1.36.0 commit: unknown

### Adapter firmware version

20230507

### Adapter

Sonoff 2652P

### Setup

VM on a Synology DSM7.

### Debug log

024-03-14 15:50:25Invalid message 'null', skipping...

avbdr, 2024-03-26:
it looks like that this feature doesnt work on any tuya devices. I have Homeetec and tuya devices and this is doesnt work either

avbdr, 2024-04-01:
seems it is working only with a devices which can be controlled by 'state' sensor. if you have state_left, state_right,state_l1,etc it will fail
ive added some console.log()s:

```
{ state_right: 'ON', on_time: 500, off_wait_time: 50 }
Zigbee2MQTT:error 2024-04-01 19:04:53: Publish 'set' 'on_time' to 'livingroom-switch-main' failed: 'Error: 'null' not allowed, choose between: toggle,off,on'
```
converters/toZigbee.js: converters1.on_off is called twice somehow.

full log:
```
Zigbee2MQTT:debug 2024-04-01 19:29:09: Received MQTT message on 'zigbee2mqtt/livingroom-switch-main/set' with data '{"state_right" : "ON", "on_time": 500, "off_wait_time": 50}'
Zigbee2MQTT:debug 2024-04-01 19:29:09: Publishing 'set' 'state' to 'livingroom-switch-main'
converters/toZigbee.js: converters1.on_off
{ on_time: 500, off_wait_time: 50, state: 'ON' }
Zigbee2MQTT:debug 2024-04-01 19:29:09: Publishing 'set' 'on_time' to 'livingroom-switch-main'
converters/toZigbee.js: converters1.on_off
{ state_right: 'ON', on_time: 500, off_wait_time: 50 }
Zigbee2MQTT:error 2024-04-01 19:29:09: Publish 'set' 'on_time' to 'livingroom-switch-main' failed: 'Error: 'null' not allowed, choose between: toggle,off,on'
Zigbee2MQTT:debug 2024-04-01 19:29:10: Error: 'null' not allowed, choose between: toggle,off,on
    at Object.validateValue (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/lib/utils.ts:448:15)
    at Object.convertSet (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/converters/toZigbee.ts:38:18)
    at Publish.onMQTTMessage (/opt/zigbee2mqtt/lib/extension/publish.ts:259:52)
    at EventEmitter.wrappedCallback (/opt/zigbee2mqtt/lib/eventBus.ts:174:17)
Zigbee2MQTT:debug 2024-04-01 19:29:10: Received Zigbee message from 'livingroom-switch-main', type 'attributeReport', cluster 'genOnOff', data '{"offWaitTime":5000,"onTime":500}' from endpoint 3 with groupID 0
Zigbee2MQTT:debug 2024-04-01 19:29:10: Received Zigbee message from 'livingroom-switch-main', type 'attributeReport', cluster 'genOnOff', data '{"offWaitTime":5000,"onTime":500}' from endpoint 3 with groupID 0
Zigbee2MQTT:debug 2024-04-01 19:29:10: Received Zigbee message from 'livingroom-switch-main', type 'attributeReport', cluster 'genOnOff', data '{"offWaitTime":5000,"onTime":500}' from endpoint 3 with groupID 0
Zigbee2MQTT:debug 2024-04-01 19:29:10: Received Zigbee message from 'livingroom-switch-main', type 'attributeReport', cluster 'genOnOff', data '{"offWaitTime":5000,"onTime":500}' from endpoint 3 with groupID 0

```

avbdr, 2024-04-02:
hahaha, in fact after I posted that, I have reread last messages. on_time was indeed set. The problem is that "on_time" works as a delayed ON.
{"state_right" : "ON", "on_time": 1} -- will turn the light on *after* 10 seconds. I were expecting that It would turn it off in 10 seconds instead.  Is it an expected behavior?

andrzejpolis, 2024-04-13:
Have exactly the same issue with [TuYa TS0001_power](https://www.zigbee2mqtt.io/devices/TS0001_power.html). Documentation states that it should keep the state for on_time value in seconds. It only works for turning ON, impacts delay in state change instead of a state duration and is in tens of seconds instead of seconds.  For OFF it doesn't work or at least I did not found a way to get any additional behavior. 
I'm also very interested in information if this is an issue with these devices support or a problem with Zigbee2Mqtt implementation. 
On top of this I would like to comment that almost all devices have this "On with timed off" documentation section and I wonder if this is not just blindly copy pasted for each new devices or really indicates someone took an effort to check if this works.

closed 2024-11-10 by github-actions[bot]