Koenkk/zigbee2mqtt #21800: [New device support, Help needed multiple Endpoints]:

Nailik, 2024-03-14:
### Link

https://merten.de/produkt/meg5172-0000-universal-dimmer-einsatz-2fach.html

### Database entry

{"id":12,"type":"Router","ieeeAddr":"0x540f57fffe306b17","nwkAddr":6330,"manufId":4190,"manufName":"Schneider Electric","powerSource":"Mains (single phase)","modelId":"2GANG/DIMMER/2","epList":[3,4,21,22,242],"endpoints":{"3":{"profId":260,"epId":3,"devId":257,"inClusterList":[0,3,4,5,6,8,769,2821],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"2GANG/DIMMER/2","manufacturerName":"Schneider Electric","powerSource":1,"zclVersion":3,"appVersion":2,"stackVersion":6,"hwVersion":2,"dateCode":"","swBuildId":"002.004.020 R"}},"genOnOff":{"attributes":{"onOff":1}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffe1695e8","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":65000,"repChange":1,"manufacturerCode":null}],"meta":{}},"4":{"profId":260,"epId":4,"devId":257,"inClusterList":[0,3,4,5,6,8,769,2821],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"dateCode":""}},"genOnOff":{"attributes":{"onOff":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffe1695e8","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":65000,"repChange":1,"manufacturerCode":null}],"meta":{}},"21":{"profId":260,"epId":21,"devId":260,"inClusterList":[0,3,2821,65303],"outClusterList":[3,4,5,6,8,258],"clusters":{"genBasic":{"attributes":{"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"22":{"profId":260,"epId":22,"devId":260,"inClusterList":[0,3,2821,65303],"outClusterList":[3,4,5,6,8,258],"clusters":{"genBasic":{"attributes":{"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":2,"stackVersion":6,"hwVersion":2,"dateCode":"","swBuildId":"002.004.020 R","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1324213189},"lastSeen":1710440949781}

### Comments

Hi,

i want to support the Meg5172-0000 with 2 Channels but have issues using the Second Endpoint.
Unfortunately i can only modify the first endpoint and when i change something for the first Endpoint 21/right then the settings for 22/left change as well.
Also brightness is only shown once:

[image]
[image]

### External definition

```shell
const types = require('zigbee-herdsman-converters/lib/types');
const utils = require('zigbee-herdsman-converters/lib/utils');
const {} = require('zigbee-herdsman-converters/lib/modernExtend');
const { enumLookup } = require('zigbee-herdsman-converters/lib/modernExtend');

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');

const e = exposes.presets;
const ea = exposes.access;

function indicatorMode(endpoint) {
    let description = 'Set Indicator Mode.';
    if (endpoint) {
        description = 'Set Indicator Mode for ' + endpoint + ' Button.';
    }
    return enumLookup({
        name: 'indicator_mode',
        lookup: {
            'reverse_with_load': 2,
            'consistent_with_load': 0,
            'always_off': 3,
            'always_on': 1,
        },
        cluster: 'clipsalWiserSwitchConfigurationClusterServer',
        attribute: {ID: 0x0000, type: 0x30},
        description: description,
        endpoint: endpoint,
    });
}

function switchActions(endpoint) {
    let description = 'Set Switch Action.';
    if (endpoint) {
        description = 'Set Switch Action for ' + endpoint + ' Button.';
    }
    return enumLookup({
        name: 'switch_actions',
        lookup: {
            'light': 0,
            'light_opposite': 254,
            'dimmer': 1,
            'dimmer_opposite': 253,
            'standard_shutter': 2,
            'standard_shutter_opposite': 252,
            'schneider_shutter': 3,
            'schneider_shutter_opposite': 251,
            'scene': 4,
            'toggle_light': 5,
            'toggle_dimmer': 6,
            'alternate_light': 7,
            'alternate_dimmer': 8,
            'not_used': 127,
        },
        cluster: 'clipsalWiserSwitchConfigurationClusterServer',
        attribute: 'SwitchActions',
        description: description,
        endpoint: endpoint,
    });
}

const definition = {
    zigbeeModel: ['2GANG/DIMMER/2'],
    model: 'MEG5126-0300/MEG5172-0000',
    vendor: 'Schneider Electric',
    description: 'Merten MEG5172 PlusLink Dimmer insert with Merten Wiser System M Push Button (2fold)',
    fromZigbee: [fz.on_off, fz.brightness, fz.level_config, fz.wiser_lighting_ballast_configuration],
    toZigbee: [tz.light_onoff_brightness, tz.level_config, tz.ballast_config, tz.wiser_dimmer_mode],
    exposes: [e.light_brightness().withLevelConfig(),
        e.numeric('ballast_minimum_level', ea.ALL).withValueMin(1).withValueMax(254)
            .withDescription('Specifies the minimum light output of the ballast'),
        e.numeric('ballast_maximum_level', ea.ALL).withValueMin(1).withValueMax(254)
            .withDescription('Specifies the maximum light output of the ballast'),
        e.enum('dimmer_mode', ea.ALL, ['auto', 'rc', 'rl', 'rl_led'])
            .withDescription('Sets dimming mode to autodetect or fixed RC/RL/RL_LED mode (max load is reduced in RL_LED)')],
    extend: [indicatorMode('right'), indicatorMode('left'), switchActions('right'), switchActions('left')],
    meta: {multiEndpoint: true},
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint1 = device.getEndpoint(21);
        await reporting.bind(endpoint1, coordinatorEndpoint, ['genOnOff', 'genLevelCtrl', 'lightingBallastCfg']);
        await reporting.onOff(endpoint1);
        await reporting.brightness(endpoint1);
        const endpoint2 = device.getEndpoint(22);
        await reporting.bind(endpoint2, coordinatorEndpoint, ['genOnOff', 'genLevelCtrl', 'lightingBallastCfg']);
        await reporting.onOff(endpoint2);
        await reporting.brightness(endpoint2);
    },
    endpoint: (device) => {
        return {'3': 3, 'right': 21, 'left': 22};
    },
};

module.exports = definition;
```

Nailik, 2024-03-15:
Figured out allmost everything .. the only thing i cannot get to work is e.action() i get no reportings of any click on a button :/

Koenkk, 2024-03-15:
what is loggen in the debug log when triggering actions on the device?

Nailik, 2024-03-15:
@Koenkk do i need e.action to see it in the logs or should it always be shown ?

Koenkk, 2024-03-16:
What action do you expect? This is not a remote right? It reports a physical state change which is good.

Nailik, 2024-03-17:
i was hoping for it to also tell me when a button is pressed.
I recently added support for the 2 buttons / 1 dimmer model and i wanted to use the secondary button set for another device. However i only managed to setup the secondary buttons for a scene that's internally stored on the device (like store a brightness level and use the secondary buttons to go to this brightness level)

Koenkk, 2024-03-17:
You can use the [state action](https://www.zigbee2mqtt.io/devices/046677552343.html#options) option for this

Since your external converter looks good, could you make a pull request?

Nailik, 2024-03-19:
it was this pr https://github.com/Koenkk/zigbee-herdsman-converters/pull/7222

closed 2024-04-12 by Nailik