Koenkk/zigbee2mqtt #21807: IKEA Vindstyrka (E2112) pm25Measurement changes in 20 minute(?) steps

cbrunnkvist, 2024-03-15:
### What happened?

I've noticed something odd about the PM2.5 value by looking at the graphs I'm producing (subscribing to MQTT, dump into RRD file):

[image]

Ignoring the slightly enthusiastic graph colors (WIP 😅), we can clearly see that, while all values update correctly (=correspond to those shown on the monitors' own display) and with respect to their Min/Max reporting parameters, the value of the **PM2.5** only changes once roughly every 1200sec/20min!?

Now, having a window/hysteresis function applied to the `pm25measurement.measuredValue` might make a lot of sense in the IKEA's intended use case: to directly control a Starkvind Air Purifier. But for data logging purposes, it would be nice to see in the graph exactly what I'm seeing on the display.

I've noticed that there is a [`measuredValueIkea` attribute](https://github.com/Koenkk/zigbee-herdsman/pull/671/files) on the `pm25Measurement ` cluster, but it is not reported and I have so far been unable to add it under _device->Reporting_ tab (there's some error). Does anyone know what that attribute is about?

### What did you expect to happen?

If the PM2.5 value shown on the device display has changed, the `pm25` value should reflect it at the next reporting interval.

### Zigbee2MQTT version

 1.35.3 commit: fe0742a

### Adapter firmware version

20211115

### Adapter

CC2531 (coordinator fw)

### Setup

Docker container on RPi 3b

sjorge, 2024-03-15:
There is a conflict with the datatype on that cluster and the duplicate had to be removed, sadly we cannot (re)configure the reporting via the frontend. The VINDSTYRKA does not send a manufacturer ID so that's why there is a conflict.

Is how the cluster looks like now, with the DT as uint16

https://github.com/Koenkk/zigbee-herdsman/blob/2560a3e55c3a3f7065ac68ec9dc95ed59e10272b/src/zcl/definition/cluster.ts#L2678-L2688

The configure function works around this by specifying the correct datatype:
https://github.com/Koenkk/zigbee-herdsman-converters/blob/bc79fa48973fb78f2da70e8ab25a16040de08191/src/devices/ikea.ts#L1184-L1187

One could use MQTT via https://www.zigbee2mqtt.io/guide/usage/mqtt_topics_and_messages.html#zigbee2mqtt-bridge-request and also specify the correct DT. Otherwise the VINDSTYRKA will reject the command. To update the interval.

cbrunnkvist, 2024-03-20:
Did someone fix this already? Intentionally or unintentionally? Because, yesterday I pulled new Docker images - now at [1.36.0](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.36.0) commit: [86ed71c](https://github.com/Koenkk/zigbee2mqtt/commit/86ed71c) I noticed this morning that the pm2.5 readings now have a _smaller_ granularity - together with all the other measurements.

[image]

However it does _not_ seem to follow the minimum reporting interval of 60 s - rather it seems to be pretty much precisely 100 s!? 😂 

Something fishy is still going on, but this is definitely an improvement! I am going to have to dump my RRD and look at the individual values to ensure that I'm not mistakenly plotting some average or something...

sjorge, 2024-03-20:
I think the change I did landed in 1.36.0, not exactly sure it's been in a release for about 1,5 month I think.

~100s would fall within the min of 60s max of 120s, it will only report if the change in value is >=2 and it has not reported in the last 60s. It will report regardless after 120s. So your graph does seem to align roughly with what is within the expectation.

FranciscoSGPa, 2024-03-20:
I have similar issue with IKEA VINDSTYRKA with the addon for Homeassitant 1.36-1. In one of the previous versions I made it reporting every around 15 seconds. Unfortunately I cannot trace back which version was.  I do not remember if I got that by changing measuredValue or measuredValueIkea report configuration
With current version I cannot change reporting anymore and it reports to fixed 120 seconds. My application needs back better reporting. Any idea on how or what to change to a lower reporting period with the current version with Homeassitant as there was with a previous version? Or if newer version could revert back to the implementation in which better reporting was allowed.

cbrunnkvist, 2024-03-21:
Well, this is getting weirder. As I look deeper than my RRD:s, I dumping the updates as I receive them directly from MQTT:
```
Mar 21 21:44:41 (+12s) Received sensor update
Mar 21 21:44:41   pm25(20)
Mar 21 21:44:41   temperature(30)
Mar 21 21:44:41   humidity(43)
Mar 21 21:44:41   voc_index(180)
--
Mar 21 21:44:53 (+12s) Received sensor update
Mar 21 21:44:53   pm25(20)
Mar 21 21:44:53   temperature(30)
Mar 21 21:44:53   humidity(43)
Mar 21 21:44:53   voc_index(179)
--
Mar 21 21:45:05 (+12s) Received sensor update
Mar 21 21:45:05   pm25(20)
Mar 21 21:45:05   temperature(30)
Mar 21 21:45:05   humidity(43)
Mar 21 21:45:05   voc_index(179)
--
Mar 21 21:46:32 (+87s) Received sensor update
Mar 21 21:46:32   pm25(20)
Mar 21 21:46:32   temperature(30)
Mar 21 21:46:32   humidity(43)
Mar 21 21:46:32   voc_index(179)
--
Mar 21 21:46:45 (+13s) Received sensor update
Mar 21 21:46:45   pm25(20)
Mar 21 21:46:45   temperature(30)
Mar 21 21:46:45   humidity(43)
Mar 21 21:46:45   voc_index(179)
--
Mar 21 21:46:57 (+12s) Received sensor update
Mar 21 21:46:57   pm25(20)
Mar 21 21:46:57   temperature(30)
Mar 21 21:46:57   humidity(43)
Mar 21 21:46:57   voc_index(178)
--
Mar 21 21:47:09 (+12s) Received sensor update
Mar 21 21:47:09   pm25(20)
Mar 21 21:47:09   temperature(30)
Mar 21 21:47:09   humidity(43)
Mar 21 21:47:09   voc_index(178)
--
Mar 21 21:48:04 (+55s) Received sensor update
Mar 21 21:48:04   pm25(20)
Mar 21 21:48:04   temperature(30)
Mar 21 21:48:04   humidity(44)
Mar 21 21:48:04   voc_index(178)
```

There's seemingly sometimes no 0 change across all values. And sometimes 12 seconds, sometimes longer. 

Could there be some underlying time unit bug, where the UI / Config says 120 seconds but we end up with minumum 12 seconds?

sjorge, 2024-03-21:
Debug log on z2m side would tell you which value triggered the update. 

It could also be the device ignores the values and just does it's own thing too.

sjorge, 2024-03-21:
(Untested command as I don't have mine paired) You can probably mess around using mosquitto_pub  ...

```
mosquitto_pub -t zigbee2mqtt/bridge/request/device/configure_reporting -m "{'attribute':{ID: 0x0000, type: 57},'cluster':'pm25Measurement','id':'FRIENDLY_NAME','maximum_report_interval':300,'minimum_report_interval':60,'reportable_change':1}"
```

FranciscoSGPa, 2024-03-21:
Thanks for the suggestion. However, this comand answer is "Request 'zigbee2mqtt/bridge/request/device/configure_reporting' failed with error: 'Invalid payload'"
It only works with a payload: {"attribute":"measuredValue" ,"cluster":"pm25Measurement","id":"FRIENDLY_NAME","maximum_report_interval":120,"minimum_report_interval":60,"reportable_change":1}
But then the error is: 
Request 'zigbee2mqtt/bridge/request/device/configure_reporting' failed with error: 'ConfigureReporting FRIENDLY_NAME pm25Measurement([{"attribute":"measuredValue","minimumReportInterval":60,"maximumReportInterval":120,"reportableChange":1}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'INVALID_DATA_TYPE')

sjorge, 2024-03-21:
You can't use the text version of the attribute as the DataType is different. I'm sure I was able to provide it before by providing both as numbers. I'll see if i can find mine this weekend, but mine never worked well so I stop using it so it's currently not paired.

sjorge, 2024-03-22:
```
mosquitto_pub -t zigbee2mqtt/bridge/request/device/configure_reporting -m "{'attribute':{'id': 0, 'type': 57},'cluster':'pm25Measurement','id':'FRIENDLY_NAME','maximum_report_interval':300,'minimum_report_interval':60,'reportable_change':1}"
```

Found the issue, I forgot the `'`, also `ID` is apparently case sensitive so should be lower case here.

FranciscoSGPa, 2024-03-22:
Thank you very much sjorge!. It works now!. 
I published using MQTT: Publish service in homeassitant developer tools with:
Topic: zigbee2mqtt/bridge/request/device/configure_reporting
Payload:  {"attribute":{"id": 0, "type": 57} ,"cluster":"pm25Measurement","id":"IKEA_FRIENDLY_NAME/1","maximum_report_interval":30,"minimum_report_interval":15,"reportable_change":1}

The configuration was successfull and now reports every 30 seconds. These are the messages from the log confirming that it is correctly configured:
Info 2024-03-22 18:59:54 Configured reporting for 'IKEA_FRIENDLY_NAME/1', 'pm25Measurement.[object Object]'
Info 2024-03-22 18:59:54 MQTT publish: topic 'zigbee2mqtt/bridge/response/device/configure_reporting', payload '{"data":{"attribute":{"id":0,"type":57},"cluster":"pm25Measurement","id":"IKEA_FRIENDLY_NAME/1","maximum_report_interval":30,"minimum_report_interval":15,"reportable_change":1},"status":"ok"}'

cbrunnkvist, 2024-03-30:
Okay... So, given that
1: for me, the major problem or very long update intervals were resolved after updating to Zigbee2MQTT:1.36.0 + mosquitto:2.0
2: @FranciscoSGPa & @sjorge were able to resolve and document how to send the custom configure_reporting request
I think we can close this ticket unless there's something I missed. @Koenkk ?

Koenkk, 2024-03-31:
We can close this for now, it will be fully fixed once custom clusters are supported as mentioned in https://github.com/Koenkk/zigbee2mqtt/issues/21807#issuecomment-1999195484

closed 2024-04-01 by cbrunnkvist

sjorge, 2024-04-28:
> We can close this for now, it will be fully fixed once custom clusters are supported as mentioned in [#21807 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/21807#issuecomment-1999195484)

It should now be fixed on the latest dev release / in the may release.