Koenkk/zigbee2mqtt #21808: Aqara WSDCGQ11LM update rate 5x per seconds

corgan2222, 2024-03-15:
### What happened?

Hello,

since a while, I notice an insane update rate from one of my [Aqara WSDCGQ11LM](https://www.zigbee2mqtt.io/devices/WSDCGQ11LM.html).
I get several hundred updates in a row. Sometimes 3-5 per second. The values itself don't change that much.
I already reduced the xx_precision settings to 1.
Only 1 of 3 sensors shows this behavior. 

The only thing what I could spot was the "power_outage_count" which jump at the end of one of these "outbreaks".

[image]

But I could not find "power_outage_count" in the [documentation](https://www.zigbee2mqtt.io/devices/WSDCGQ11LM.html) ?

"battery": 97,
"linkquality": 69,

[image]
[image]
[image]

### What did you expect to happen?

reduce the update rate.

### How to reproduce it (minimal and precise)

no idea

### Zigbee2MQTT version

1.36.0 

### Adapter firmware version

20230507

### Adapter

Sonoff_Zigbee_3.0_USB_Dongle

### Setup

Add-on on Home Assistant 2024.3.1, Raspi5 Supervisor

### Debug log

Zigbee2MQTT:info  2024-03-15 05:34:37: MQTT publish: topic 'zigbee2mqtt/humidity_xiaomi_bad_stefan', payload '{"battery":97,"humidity":58.4,"linkquality":76,"power_outage_count":11574,"pressure":1001.8,"temperature":22.4,"voltage":2995}'
Zigbee2MQTT:info  2024-03-15 05:34:38: MQTT publish: topic 'zigbee2mqtt/humidity_xiaomi_bad_stefan', payload '{"battery":97,"humidity":58.4,"linkquality":72,"power_outage_count":11574,"pressure":1001.8,"temperature":22.4,"voltage":2995}'
Zigbee2MQTT:info  2024-03-15 05:34:38: MQTT publish: topic 'zigbee2mqtt/humidity_xiaomi_bad_stefan', payload '{"battery":97,"humidity":58.4,"linkquality":72,"power_outage_count":11574,"pressure":1001.8,"temperature":22.4,"voltage":2995}'
Zigbee2MQTT:info  2024-03-15 05:34:39: MQTT publish: topic 'zigbee2mqtt/humidity_xiaomi_bad_stefan', payload '{"battery":97,"humidity":58.4,"linkquality":72,"power_outage_count":11574,"pressure":1001.8,"temperature":22.4,"voltage":2995}'
Zigbee2MQTT:info  2024-03-15 05:34:39: MQTT publish: topic 'zigbee2mqtt/humidity_xiaomi_bad_stefan', payload '{"battery":97,"humidity":58.3,"linkquality":72,"power_outage_count":11574,"pressure":1001.8,"temperature":22.4,"voltage":2995}'
Zigbee2MQTT:info  2024-03-15 05:34:39: MQTT publish: topic 'zigbee2mqtt/humidity_xiaomi_bad_stefan', payload '{"battery":97,"humidity":58.3,"linkquality":72,"power_outage_count":11574,"pressure":1001.8,"temperature":22.4,"voltage":2995}'
Zigbee2MQTT:info  2024-03-15 05:34:39: MQTT publish: topic 'zigbee2mqtt/humidity_xiaomi_bad_stefan', payload '{"battery":97,"humidity":58.4,"linkquality":76,"power_outage_count":11574,"pressure":1001.8,"temperature":22.4,"voltage":2995}'
Zigbee2MQTT:info  2024-03-15 05:34:39: MQTT publish: topic 'zigbee2mqtt/humidity_xiaomi_bad_stefan', payload '{"battery":97,"humidity":58.4,"linkquality":72,"power_outage_count":11574,"pressure":1001.8,"temperature":22.4,"voltage":2995}'
Zigbee2MQTT:info  2024-03-15 05:34:40: MQTT publish: topic 'zigbee2mqtt/humidity_xiaomi_bad_stefan', payload '{"battery":97,"humidity":58.4,"linkquality":72,"power_outage_count":11574,"pressure":1001.8,"temperature":22.4,"voltage":2995}'
Zigbee2MQTT:info  2024-03-15 05:34:40: MQTT publish: topic 'zigbee2mqtt/humidity_xiaomi_bad_stefan', payload '{"battery":97,"humidity":58.4,"linkquality":76,"power_outage_count":11574,"pressure":1001.8,"temperature":22.4,"voltage":2995}'
Zigbee2MQTT:info  2024-03-15 05:34:40: MQTT publish: topic 'zigbee2mqtt/humidity_xiaomi_bad_stefan', payload '{"battery":97,"humidity":58.4,"linkquality":72,"power_outage_count":11574,"pressure":1001.8,"temperature":22.4,"voltage":2995}'
Zigbee2MQTT:info  2024-03-15 05:34:41: MQTT publish: topic 'zigbee2mqtt/humidity_xiaomi_bad_stefan', payload '{"battery":97,"humidity":58.4,"linkquality":69,"power_outage_count":11574,"pressure":1001.8,"temperature":22.4,"voltage":2995}'
Zigbee2MQTT:info  2024-03-15 05:34:41: MQTT publish: topic 'zigbee2mqtt/humidity_xiaomi_bad_stefan', payload '{"battery":97,"humidity":58.4,"linkquality":40,"power_outage_count":11574,"pressure":1001.8,"temperature":22.4,"voltage":2995}'
Zigbee2MQTT:info  2024-03-15 05:34:41: MQTT publish: topic 'zigbee2mqtt/humidity_xiaomi_bad_stefan', payload '{"battery":97,"humidity":58.4,"linkquality":76,"power_outage_count":11574,"pressure":1001.8,"temperature":22.4,"voltage":2995}'
Zigbee2MQTT:info  2024-03-15 05:34:41: MQTT publish: topic 'zigbee2mqtt/humidity_xiaomi_bad_stefan', payload '{"battery":97,"humidity":58.3,"linkquality":72,"power_outage_count":11574,"pressure":1001.8,"temperature":22.4,"voltage":2995}'
Zigbee2MQTT:info  2024-03-15 05:34:41: MQTT publish: topic 'zigbee2mqtt/humidity_xiaomi_bad_stefan', payload '{"battery":97,"humidity":58.3,"linkquality":72,"power_outage_count":11574,"pressure":1001.8,"temperature":22.4,"voltage":2995}'
Zigbee2MQTT:info  2024-03-15 05:34:41: MQTT publish: topic 'zigbee2mqtt/humidity_xiaomi_bad_stefan', payload '{"battery":97,"humidity":58.4,"linkquality":72,"power_outage_count":11574,"pressure":1001.8,"temperature":22.4,"voltage":2995}'
Zigbee2MQTT:info  2024-03-15 05:34:42: MQTT publish: topic 'zigbee2mqtt/humidity_xiaomi_bad_stefan', payload '{"battery":97,"humidity":58.4,"linkquality":72,"power_outage_count":11574,"pressure":1001.8,"temperature":22.4,"voltage":2995}'
Zigbee2MQTT:info  2024-03-15 05:34:42: MQTT publish: topic 'zigbee2mqtt/humidity_xiaomi_bad_stefan', payload '{"battery":97,"humidity":58.3,"linkquality":72,"power_outage_count":11574,"pressure":1001.8,"temperature":22.4,"voltage":2995}'
Zigbee2MQTT:info  2024-03-15 05:34:42: MQTT publish: topic 'zigbee2mqtt/humidity_xiaomi_bad_stefan', payload '{"battery":97,"humidity":58.3,"linkquality":72,"power_outage_count":11574,"pressure":1001.8,"temperature":22.4,"voltage":2995}'
Zigbee2MQTT:info  2024-03-15 05:34:42: MQTT publish: topic 'zigbee2mqtt/humidity_xiaomi_bad_stefan', payload '{"battery":97,"humidity":58.3,"linkquality":72,"power_outage_count":11574,"pressure":1001.8,"temperature":22.4,"voltage":2995}'
Zigbee2MQTT:info  2024-03-15 05:34:43: MQTT publish: topic 'zigbee2mqtt/humidity_xiaomi_bad_stefan', payload '{"battery":97,"humidity":58.3,"linkquality":72,"power_outage_count":11574,"pressure":1001.8,"temperature":22.4,"voltage":2995}'
[... 32 lines cut ...]
Zigbee2MQTT:info  2024-03-15 05:34:51: MQTT publish: topic 'zigbee2mqtt/humidity_xiaomi_bad_stefan', payload '{"battery":97,"humidity":58,"linkquality":69,"power_outage_count":11574,"pressure":1001.8,"temperature":22.4,"voltage":2995}'
Zigbee2MQTT:info  2024-03-15 05:34:52: MQTT publish: topic 'zigbee2mqtt/humidity_xiaomi_bad_stefan', payload '{"battery":97,"humidity":58,"linkquality":69,"power_outage_count":11574,"pressure":1001.8,"temperature":22.4,"voltage":2995}'
Zigbee2MQTT:info  2024-03-15 05:34:52: MQTT publish: topic 'zigbee2mqtt/humidity_xiaomi_bad_stefan', payload '{"battery":97,"humidity":58,"linkquality":40,"power_outage_count":11574,"pressure":1001.8,"temperature":22.4,"voltage":2995}'
Zigbee2MQTT:info  2024-03-15 05:34:52: MQTT publish: topic 'zigbee2mqtt/humidity_xiaomi_bad_stefan', payload '{"battery":97,"humidity":58.2,"linkquality":72,"power_outage_count":11574,"pressure":1001.8,"temperature":22.4,"voltage":2995}'
Zigbee2MQTT:info  2024-03-15 05:34:52: MQTT publish: topic 'zigbee2mqtt/humidity_xiaomi_bad_stefan', payload '{"battery":97,"humidity":58.2,"linkquality":69,"power_outage_count":11574,"pressure":1001.8,"temperature":22.4,"voltage":2995}'
Zigbee2MQTT:info  2024-03-15 05:34:53: MQTT publish: topic 'zigbee2mqtt/humidity_xiaomi_bad_stefan', payload '{"battery":97,"humidity":58.2,"linkquality":69,"power_outage_count":11574,"pressure":1001.8,"temperature":22.4,"voltage":2995}'
Zigbee2MQTT:info  2024-03-15 05:34:53: MQTT publish: topic 'zigbee2mqtt/humidity_xiaomi_bad_stefan', payload '{"battery":97,"humidity":58.4,"linkquality":65,"power_outage_count":11574,"pressure":1001.8,"temperature":22.4,"voltage":2995}'
Zigbee2MQTT:info  2024-03-15 05:34:53: MQTT publish: topic 'zigbee2mqtt/humidity_xiaomi_bad_stefan', payload '{"battery":97,"humidity":58.4,"linkquality":72,"power_outage_count":11574,"pressure":1001.8,"temperature":22.4,"voltage":2995}'
Zigbee2MQTT:info  2024-03-15 05:34:53: MQTT publish: topic 'zigbee2mqtt/humidity_xiaomi_bad_stefan', payload '{"battery":97,"humidity":58.4,"linkquality":72,"power_outage_count":11574,"pressure":1001.8,"temperature":22.4,"voltage":2995}'
[... cut ...] e":22.4,"voltage":2995}'
Zigbee2MQTT:info  2024-03-15 05:34:57: MQTT publish: topic 'zigbee2mqtt/humidity_xiaomi_bad_stefan', payload '{"battery":97,"humidity":58.9,"linkquality":65,"power_outage_count":11574,"pressure":1001.7,"temperature":22.4,"voltage":2995}'
Zigbee2MQTT:info  2024-03-15 05:35:07: MQTT publish: topic 'zigbee2mqtt/humidity_xiaomi_bad_stefan', payload '{"battery":97,"humidity":58.9,"linkquality":69,"power_outage_count":12842,"pressure":1001.8,"temperature":22.4,"voltage":2995}'

mrmaximas, 2024-03-15:
i have same sensor works without any problems, have you tried to change the battery?

corgan2222, 2024-03-15:
Yes, multiple time. As I wrote, I have 3 of these sensors. 2 of them run completely fine.

mrmaximas, 2024-03-15:
I think it's a problem with a specific sensor

mrskycriper, 2024-03-15:
I have two of them and haven't seen any problems with spamming

corgan2222, 2024-03-15:
Really strange. After removing the battery for hours, disconnecting the sensor from the mesh and reconnect to the coordinator, it looks good now.

closed 2024-10-12 by github-actions[bot]