Koenkk/zigbee2mqtt #21810: [New device support]: EKF "TS0219_TYZB01_fbo1dpof", "Tuya" smart siren.

onmobs, 2024-03-15:
### Link

https://ekfgroup.com/ru/catalog/products/umnaya-sirena-zigbee

### Database entry

{"id":29,"type":"Router","ieeeAddr":"0x3c2ef5fffe95c964","nwkAddr":6940,"manufId":0,"manufName":"_TYZB01_fbo1dpof","powerSource":"Mains (3 phase)","modelId":"TS0219","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1027,"inClusterList":[0,1,3,4,5,1280,1282,61183],"outClusterList":[3,25],"clusters":{"genBasic":{"attributes":{"modelId":"TS0219","manufacturerName":"_TYZB01_fbo1dpof","powerSource":2}},"ssIasZone":{"attributes":{"iasCieAddr":"0xe0798dfffefec605","zoneState":1,"zoneStatus":16}},"genPowerCfg":{"attributes":{"batteryVoltage":42,"batteryPercentageRemaining":200}},"ssIasWd":{"attributes":{"1":50,"2":50,"3":0,"maxDuration":240}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":80,"stackVersion":2,"hwVersion":2,"dateCode":"20190430","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1710524813786}

### Comments

Hello. I had bought another rare device - Zigbee EKF Connect smart siren TS0219_TYZB01_fbo1dpof (in fact not so smart), it is TuYa TS0224 analog but with fewer functions, works with same converter from TS0224, but only two functions seems to work - "alarm on" MQTT publish: zigbee2mqtt/zalarm/set {"warning":{"mode":"burglar"}} and "alarm off" zigbee2mqtt/zalarm/set {"warning":{"mode":"stop"}}, guess that`s all.  Please add it.

### External definition

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const tzLocal = {
    TS0219: {
        key: ['light', 'alarm_duration'],
        convertSet: async (entity, key, value, meta) => {
            let payload = null;
            if (key === 'light') {
                await entity.command('genOnOff', value.toLowerCase() === 'on' ? 'on' : 'off', {}, utils.getOptions(meta.mapped, entity));
            } else if (key === 'duration') {
                await entity.write('ssIasWd', {'maxDuration': value}, utils.getOptions(meta.mapped, entity));
            } else if (key === 'volume') {
                const lookup = {'mute': 0, 'low': 10, 'medium': 30, 'high': 50};
                value = value.toLowerCase();
                utils.validateValue(value, Object.keys(lookup));
                await entity.write('ssIasWd', {0x0002: {value: lookup[value], type: 0x0a}}, utils.getOptions(meta.mapped, entity));
            } 
            return {state: {[key]: value}};
        },
    },
};

const alarmConverter = {
alarm: {
    key: ['alarm'],
    convertSet: async (entity, key, value, meta) => {
        const mode = {'stop': 0, 'burglar': 1, 'fire': 2, 'emergency': 3, 'police_panic': 4, 'fire_panic': 5, 'emergency_panic': 6};
        const level = {'low': 0, 'medium': 1, 'high': 2, 'very_high': 3};
        const strobeLevel = {'low': 0, 'medium': 1, 'high': 2, 'very_high': 3};

        const values = {
            mode: value.toLowerCase() === 'on' ? 'emergency' : 'stop',
            level: 'medium',
            strobe: true,
            duration: 3600,
            strobeDutyCycle: 10,
            strobeLevel: 3,
        };

        let info;
        // https://github.com/Koenkk/zigbee2mqtt/issues/8310 some devices require the info to be reversed.
        if (['SIRZB-110', 'SRAC-23B-ZBSR', 'AV2010/29A', 'AV2010/24A'].includes(meta.mapped.model)) {
            info = (mode[values.mode]) + ((values.strobe ? 1 : 0) << 4) + (level[values.level] << 6);
        } else {
            info = (mode[values.mode] << 4) + ((values.strobe ? 1 : 0) << 2) + (level[values.level]);
        }

        await entity.command(
            'ssIasWd',
            'startWarning',
            {startwarninginfo: info, warningduration: values.duration,
                strobedutycycle: values.strobeDutyCycle, strobelevel: values.strobeLevel},
            utils.getOptions(meta.mapped, entity));
    }
}
};

  // Merge the custom converters with the existing ones
Object.assign(tz, alarmConverter);

const definition = {
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0219',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TYZB01_fbo1dpof'
        },
    ],
    zigbeeModel: 'TS0219',
    model: 'TS0219',
    vendor: 'TuYa',
    description: 'Tuya smart light & sound siren',
    fromZigbee: [],
    toZigbee: [tz.warning, tzLocal.TS0219, alarmConverter.alarm],
    exposes: [e.warning(), 
        exposes.binary('light', ea.ALL, 'ON', 'OFF').withDescription('Turn the light ON/OFF'),
        exposes.binary('alarm', ea.ALL, 'ON', 'OFF').withDescription('Turn the alarm ON/OFF'),
        exposes.numeric('duration', ea.STATE_SET).withValueMin(60).withValueMax(3600).withValueStep(1).withUnit('s')
            .withDescription('Duration of the alarm'),
        exposes.enum('volume', ea.STATE_SET, ['mute', 'low', 'medium', 'high'])
            .withDescription('Volume of the alarm'),
    ],
};

module.exports = definition;
```

mrskycriper, 2024-04-06:
Can you please capture some debug logs of all exposes in action? Even if they don't seem to do anything usefull.

mrskycriper, 2024-04-06:
Also, you please test it with this external converter and see, what is working properly?

```js
const {battery, forcePowerSource, iasWarning, iasZoneAlarm} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    fingerprint: tuya.fingerprint('TS0219', ['_TYZB01_fbo1dpof']),
    model: 'is-sr-sb',
    vendor: 'EKF',
    description: 'Smart siren',
    extend: [
        forcePowerSource({powerSource: 'Mains (single phase)'}),
        iasWarning(),
        iasZoneAlarm({zoneType: 'generic', zoneAttributes: []}),
        battery({voltage: true}),
    ],
};

module.exports = definition;
```

onmobs, 2024-04-07:
I have a sudden problem with z2m stability right now, it crashes after I press permit join in z2m interface, need some time to figure it out.

onmobs, 2024-04-09:
My system is running again (had to reinstall z2m and rejoin all devices tho), here is debug:
debug 2024-04-09 22:37:18Received Zigbee message from 'zalarm', type 'attributeReport', cluster 'ssIasZone', data '{"zoneStatus":16}' from endpoint 1 with groupID 0
debug 2024-04-09 22:37:18No converter available for 'TS0219' with cluster 'ssIasZone' and type 'attributeReport' and data '{"zoneStatus":16}'
debug 2024-04-09 22:39:52Received Zigbee message from 'zalarm', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":17}' from endpoint 1 with groupID 0
debug 2024-04-09 22:39:52No converter available for 'TS0219' with cluster 'ssIasZone' and type 'commandStatusChangeNotification' and data '{"extendedstatus":0,"zonestatus":17}'
debug 2024-04-09 22:39:52Received Zigbee message from 'zalarm', type 'attributeReport', cluster 'ssIasZone', data '{"zoneStatus":17}' from endpoint 1 with groupID 0
debug 2024-04-09 22:39:52No converter available for 'TS0219' with cluster 'ssIasZone' and type 'attributeReport' and data '{"zoneStatus":17}'
error 2024-04-09 22:40:25Publish 'set' 'alarm' to 'zalarm' failed: 'Error: ZCL command 0x3c2ef5fffe95c964/1 ssIasWd.startWarning({"startwarninginfo":5,"warningduration":3600,"strobedutycycle":10,"strobelevel":3}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 50068 - 1 - 70 - 1282 - 11 after 10000ms)'
debug 2024-04-09 22:40:25Error: ZCL command 0x3c2ef5fffe95c964/1 ssIasWd.startWarning({"startwarninginfo":5,"warningduration":3600,"strobedutycycle":10,"strobelevel":3}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 50068 - 1 - 70 - 1282 - 11 after 10000ms) at Object.start (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:63:23) at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:499:47) at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:35:20) at Request.send (/app/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:79:20) at Endpoint.zclCommand (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:760:28) at Endpoint.command (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:591:24) at Object.convertSet (/app/data/extension/externally-loaded.js:57:9) at Publish.onMQTTMessage (/app/lib/extension/publish.ts:259:36) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17)
debug 2024-04-09 22:41:27Received MQTT message on 'zigbee2mqtt/zalarm/set' with data '{"duration":1768}'
error 2024-04-09 22:41:27No converter available for 'duration' (1768)
debug 2024-04-09 22:41:29Received MQTT message on 'zigbee2mqtt/zalarm/set' with data '{"volume":"high"}'
error 2024-04-09 22:41:29No converter available for 'volume' ("high")

onmobs, 2024-04-09:
Your external converter is not working - got "device is unsupported error" when joining.

mrskycriper, 2024-04-09:
Made a stupid mistake with import once again, sorry. How about this one?

```js
const {battery, forcePowerSource, iasWarning, iasZoneAlarm} = require('zigbee-herdsman-converters/lib/modernExtend');
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    fingerprint: tuya.fingerprint('TS0219', ['_TYZB01_fbo1dpof']),
    model: 'is-sr-sb',
    vendor: 'EKF',
    description: 'Smart siren',
    extend: [
        forcePowerSource({powerSource: 'Mains (single phase)'}),
        iasWarning(),
        iasZoneAlarm({zoneType: 'generic', zoneAttributes: []}),
        battery({voltage: true}),
    ],
};

module.exports = definition;
```

onmobs, 2024-04-09:
Now it is detected as supported but clicking all exposes still doing nothing.

error 2024-04-09 22:59:43Publish 'set' 'warning' to 'zalarm' failed: 'Error: ZCL command 0x3c2ef5fffe95c964/1 ssIasWd.startWarning({"startwarninginfo":5,"warningduration":10,"strobedutycycle":0,"strobelevel":1}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 30567 - 1 - 40 - 1282 - 11 after 10000ms)'
debug 2024-04-09 22:59:43Error: ZCL command 0x3c2ef5fffe95c964/1 ssIasWd.startWarning({"startwarninginfo":5,"warningduration":10,"strobedutycycle":0,"strobelevel":1}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 30567 - 1 - 40 - 1282 - 11 after 10000ms) at Object.start (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:63:23) at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:499:47) at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:35:20) at Request.send (/app/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:79:20) at Endpoint.zclCommand (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:760:28) at Endpoint.command (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:591:24) at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:1141:13) at Publish.onMQTTMessage (/app/lib/extension/publish.ts:259:36) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17)
debug 2024-04-09 23:00:33Received Zigbee message from 'zalarm', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":200,"batteryVoltage":42}' from endpoint 1 with groupID 0
info 2024-04-09 23:00:33MQTT publish: topic 'zigbee2mqtt/zalarm', payload '{"ac_status":false,"battery":100,"battery_defect":false,"battery_low":false,"linkquality":25,"restore_reports":false,"supervision_reports":true,"tamper":false,"test":false,"trouble":false,"voltage":4200}'
debug 2024-04-09 23:00:33Received Zigbee message from 'zalarm', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":200,"batteryVoltage":42}' from endpoint 1 with groupID 0
debug 2024-04-09 23:00:33Received Zigbee message from 'zalarm', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":200,"batteryVoltage":42}' from endpoint 1 with groupID 0
debug 2024-04-09 23:00:33Received Zigbee message from 'zalarm', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":200,"batteryVoltage":42}' from endpoint 1 with groupID 0
debug 2024-04-09 23:00:33Received Zigbee message from 'zalarm', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":200,"batteryVoltage":42}' from endpoint 1 with groupID 0
debug 2024-04-09 23:00:33Received Zigbee message from 'zalarm', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":200,"batteryVoltage":42}' from endpoint 1 with groupID 0
debug 2024-04-09 23:00:33Received Zigbee message from 'zalarm', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":200,"batteryVoltage":42}' from endpoint 1 with groupID 0
info 2024-04-09 23:00:33MQTT publish: topic 'zigbee2mqtt/zalarm', payload '{"ac_status":false,"battery":100,"battery_defect":false,"battery_low":false,"linkquality":25,"restore_reports":false,"supervision_reports":true,"tamper":false,"test":false,"trouble":false,"voltage":4200}'
info 2024-04-09 23:00:33MQTT publish: topic 'zigbee2mqtt/zalarm', payload '{"ac_status":false,"battery":100,"battery_defect":false,"battery_low":false,"linkquality":25,"restore_reports":false,"supervision_reports":true,"tamper":false,"test":false,"trouble":false,"voltage":4200}'
info 2024-04-09 23:00:33MQTT publish: topic 'zigbee2mqtt/zalarm', payload '{"ac_status":false,"battery":100,"battery_defect":false,"battery_low":false,"linkquality":25,"restore_reports":false,"supervision_reports":true,"tamper":false,"test":false,"trouble":false,"voltage":4200}'
info 2024-04-09 23:00:33MQTT publish: topic 'zigbee2mqtt/zalarm', payload '{"ac_status":false,"battery":100,"battery_defect":false,"battery_low":false,"linkquality":25,"restore_reports":false,"supervision_reports":true,"tamper":false,"test":false,"trouble":false,"voltage":4200}'
info 2024-04-09 23:00:33MQTT publish: topic 'zigbee2mqtt/zalarm', payload '{"ac_status":false,"battery":100,"battery_defect":false,"battery_low":false,"linkquality":25,"restore_reports":false,"supervision_reports":true,"tamper":false,"test":false,"trouble":false,"voltage":4200}'
info 2024-04-09 23:00:33MQTT publish: topic 'zigbee2mqtt/zalarm', payload '{"ac_status":false,"battery":100,"battery_defect":false,"battery_low":false,"linkquality":25,"restore_reports":false,"supervision_reports":true,"tamper":false,"test":false,"trouble":false,"voltage":4200}'
debug 2024-04-09 23:00:34Received Zigbee message from 'zalarm', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":200,"batteryVoltage":42}' from endpoint 1 with groupID 0
[... cut ...]
debug 2024-04-09 23:00:34Received Zigbee message from 'zalarm', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":16}' from endpoint 1 with groupID 0
debug 2024-04-09 23:00:34Received Zigbee message from 'zalarm', type 'attributeReport', cluster 'ssIasZone', data '{"zoneStatus":16}' from endpoint 1 with groupID 0
info 2024-04-09 23:00:34MQTT publish: topic 'zigbee2mqtt/zalarm', payload '{"ac_status":false,"battery":100,"battery_defect":false,"battery_low":false,"linkquality":21,"restore_reports":false,"supervision_reports":true,"tamper":false,"test":false,"trouble":false,"voltage":4200}'

closed 2024-11-06 by github-actions[bot]