Koenkk/zigbee2mqtt #21814: [New device support]: Frient SMRZB-153 Smart Cable

Einar-Vaagland, 2024-03-15:
### Link

https://frient.com/products/smart-cable/

### Database entry

{"id":2,"type":"Router","ieeeAddr":"0x0015bc003700345c","nwkAddr":4219,"manufId":4117,"manufName":"frient A/S","powerSource":"Mains (single phase)","modelId":"SMRZB-153","epList":[1,2,242],"endpoints":{"1":{"profId":49353,"epId":1,"devId":1,"inClusterList":[5,6],"outClusterList":[],"clusters":{},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b001ccc3e6f","endpointID":1}],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":81,"inClusterList":[0,2,3,4,5,6,9,1794,2820,2821],"outClusterList":[3,10,25,1030],"clusters":{"genBasic":{"attributes":{"modelId":"SMRZB-153","manufacturerName":"frient A/S","powerSource":1,"zclVersion":7,"dateCode":"2023-09-20 03:41"}},"genOnOff":{"attributes":{"33024":0,"33025":0,"33026":4294967295,"onOff":0}},"haElectricalMeasurement":{"attributes":{"activePower":0,"acFrequency":49966,"rmsVoltage":23753,"rmsCurrent":0}},"seMetering":{"attributes":{"currentSummDelivered":[0,1554],"instantaneousDemand":0,"multiplier":1,"divisor":1000}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b001ccc3e6f","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00124b001ccc3e6f","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00124b001ccc3e6f","endpointID":1},{"cluster":1026,"type":"endpoint","deviceIeeeAddress":"0x00124b001ccc3e6f","endpointID":1}],"configuredReportings":[{"cluster":1794,"attrId":1024,"minRepIntval":10,"maxRepIntval":600,"repChange":1,"manufacturerCode":null},{"cluster":2820,"attrId":1285,"minRepIntval":10,"maxRepIntval":3600,"repChange":100,"manufacturerCode":null},{"cluster":2820,"attrId":1288,"minRepIntval":10,"maxRepIntval":3600,"repChange":100,"manufacturerCode":null},{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0,"manufacturerCode":null},{"cluster":1794,"attrId":0,"minRepIntval":5,"maxRepIntval":3600,"repChange":[1,1],"manufacturerCode":null},{"cluster":2820,"attrId":1291,"minRepIntval":5,"maxRepIntval":3600,"repChange":1,"manufacturerCode":null}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"dateCode":"2023-09-20 03:41","zclVersion":7,"interviewCompleted":true,"meta":{"configured":1015318758},"lastSeen":1710542095981}

### Comments

I asked ChatGPT to make the definition file, and it works. I've also created an icon for it below. Please verify it and add it to the library.

**SMRZB-153.js:**
```
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const e = exposes.presets;
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const reporting = require('zigbee-herdsman-converters/lib/reporting');

const definition = {
    zigbeeModel: ['SMRZB-153'],
    model: 'SMRZB-153',
    vendor: 'Frient A/S',
    description: 'Smart Cable - power switch with power measurement',
    fromZigbee: [
        fz.on_off,
        fz.electrical_measurement,
        fz.metering,
        {
            cluster: 'haElectricalMeasurement',
            type: ['attributeReport', 'readResponse'],
            convert: (model, msg, publish, options, meta) => {
                const result = {};
                if (msg.data['rmsVoltage'] !== undefined) {
                    result.voltage = msg.data['rmsVoltage'] / 100;
                }
                if (msg.data['rmsCurrent'] !== undefined) {
                    result.current = msg.data['rmsCurrent'] / 2000;
                }
                if (msg.data['acFrequency'] !== undefined) {
                    result.ac_frequency = msg.data['acFrequency'] / 1000;
                }
                if (msg.data['activePower'] !== undefined) {
                    result.power = msg.data['activePower'];
                }
                // Let's not manually set the state here to avoid conflicts
                return result;
            },
        },
    ],
    toZigbee: [tz.on_off],
    exposes: [
        e.switch(),
        e.power(),
        e.energy(),
        e.voltage(),
        e.current(),
        e.ac_frequency(),
    ],
    endpoint: (device) => {
        // Correctly returning an object mapping here
        return {default: 2};
    },
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(2);

        await endpoint.bind('genOnOff', coordinatorEndpoint);
        await reporting.onOff(endpoint);

[... 14 lines cut ...]
    },
};

module.exports = definition;
```

**Icon:**
`data:image/png;base64,iVBORw0KGgoAAAANSUhEUgAAAgAAAAIACAYAAAD0eNT6AAAAAXNSR0IArs4c6QAAAARnQU1BAACxjwv8YQUAAAAJcEhZcwAAFiUAABYlAUlSJPAAAJqdSURBVHhe7b0HmGXbWR3oOE7jnMBhPAMeh3EeEwY8M8x4bA8f2MJmTLTGgE0ywcZYtkUwSWBhBAKMhmAkCxBgJAQYFBASQiIIhff0UudUXd1dVd1V1blf5+6686999jp3nf/uc++t7n71JFjr+9Z39v73v8M9d59//efcW7d+g2EYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhmEYhvFrAZPJ5Dfv7Oz8NjDKvzX4m4K/Efbgf3P27Nnfdf78+d9z6dKl33fx4sXfu729/btB1M+dO/dHw/bfXb58+cOCHx62/wF1MMr//ebm5p/e2tr6M+H/Z+P4P9b2P3XhwoU/GWP+8Wj/kOj3BzB3XY5hGIZhGBmnT5/+HRDQjY2Njwxh/sS1tbV/cubMmRcHv211dfUHVlZW3nD8+PF3HTt27PDRo0c3jxw5cu3QoUM3g7eifDuOd4J3o3z38OHDd6N8Txn23fJ+izE3yzvKsIP3Y+7eL9Z6P9Z878SJ43dXVo7fCd4+efLELfDEiWM3o34jytdXV1cunzq1snH69OqhU6dW3xv8hdOnT77xzJnVHzt9+tR/OnPm1MvW18985cbG2hfGufnUSE7+dhw/Oo5/CUkIkg8kHlH+0KtXr/6hOI+/d319/XdGkvNb6uk1DMMwjAdHCMpvv359+49duXLlz16+vP3XL17c+rgLF7Y+Me6CPzPueL/43LmNrwkB/48hVv8lBP3nV1dPPh0Ct7GycuLGsWNH7x09CmGEQB6K47R8+PDB+4cOHdg5fPjQHB7cOXToYIjt4RnGWHE8VNiVp23srzbatY5+mV1bN676HTsGQvCnfl25Gxfr7MbXfuOM8SZglAeM/pN43T3jHIUN9q6963e08Pjxo5NIKoLHo3ysEPU47kRbWS8Y7TsnT66UY7wvUQZXCk+dOgneD96LZOTO6bXT18+snTm3tr52MN7Xd509t/7ms+fOvv7s5tkfjvf6P8fxe+J9f/n29uZLgv829sEXnz9//h9HAvJJV69e/Lhr1679pevXr/9x7Ju6hQzDMIznE3hEfeHC5t/Y3l7/Rxvn1r4qgv33nlxd+ZmVlWPvPXHi6Ing9omVo1ejHnepx24fP37kbtx43wuGYIXI9seOEB8KmIoZBIw2+LCc/VrM4tcifEi1sQzBJGkjte8Yc5/cL9txzHOOzU+2xtM6+x88uH9y4MC+AWHTsdmvdS45Huo895l4j5RHj4f9RLy3weMrkUxU0lbs4dclHkOurByfRHIxWV1dIZFcILG4ffr0qatRXovjU2fOnP65tbXTPxRJ48s2NzdeFInFC7e3z/5NfKyCp0J1yxqGYRgZeNyLYBl3Yh8bd2mffObMmX8eQfbbTp1aeX3c4T0eAfpcBPcbcbwTxxDyIyHkweNH7h87fggMET+onJDRNjl+IgQjjqgfOXqAxx3y8JE41rvkEKOdEKUBYQvhKeUQrXLcv/+ZvkwfJe2VAxEkaaP4tUifFlUUaRvrn+3alttb9YOHYr2Hwzd4+EjYK6O+M+AhnIsGo+1wZYzXM8Yvx3qeBm0kfbSO9wrHRTwY84GHjsQYI+R7T0aCMXiygYQDRKJAws4ykwXhDomnE5E43CdPnly5d/Lkyburq6s3T51CEnFqK/b7SiQKT6yvr7899v9/OXv27MvPnTv3L+N6+H+2t7c/4tq1a3+kXiqGYRgfOIiA9YfxOW4Eqr+JgLW2tvY5Eci+tH72/ZIIbN8eQe77g6+LwPe2lZWVp4NbcYt+I3j72LFj+Mz7Hj5vls+mg92jawRxCfwDYTpSRAjCtK/w4KFn+nIm/OayjNcJKqjzxNxNUVQb+2Vbq5yZx0KZ46lfi601jPXT9hbVJ5dDuPtzhSQKHJy/1MZ2Pf+9va6Hc+CYz8F03vF1U5iV2efw0SmPHIv2YK4vGkNt2RdEMkAel6QA1HLH48EThXEdTOKaKEQ5bJE0rCjvC+9F0nAnrqNbcU1dDh6Ja+0tca29IhKGfx7X3sfjuxR+4mAYxgMhgsiHBP8qgkkElc+JYPLvIN5x1/LW48ePHzl+/Ni1qWB3XxILxp1SR3xWG+07CG5xjIB5JAI4gmc5hqB3XzZDGTx06FDc9eHOesqprbsrxF02We8Qe7FAAkChISk2FB4VoM4+8N05eKhjjMckoxxbd6O0hRAMEpNF1LFQzoQd42Fcjp3Hp58y+7TWlOdRW5SLqIHRN4vwYE1xrvsnJmCcv544j+Q8W9+3rpPjcy6dT9fJNvZhebq2Kdne+6W7fTISgMJIAPIY5XzonT9FXusg/FhW8Sfbbd13G3B94FoB49oqR1xDrJPqAzJJkKShEPVIEAYMG/x2oryzemr1flzPVyNpOBDX9psiaf//4jr/ivPnz39W8O9cuH [... cut ...]
const {deviceEndpoints, onOff, electricityMeter, identify} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['SMRZB-153'],
    model: 'SMRZB-153',
    vendor: 'frient A/S',
    description: 'Automatically generated definition',
    extend: [deviceEndpoints({"endpoints":{"1":1,"2":2,"242":242}}), onOff({"powerOnBehavior":false,"endpoints":{"1":1,"2":2}}), electricityMeter(), identify()],
    meta: {"multiEndpoint":true},
};

module.exports = definition;
```

Koenkk, 2024-03-16:
Does it also work with the generated external definition (bottom of your post)?

Einar-Vaagland, 2024-03-18:
No, that is just the initial export from Zigbee2Mqtt. I wondered if I should replace it, but thought you might need it for something.

Koenkk, 2024-03-18:
Can you check if it works?

Einar-Vaagland, 2024-03-19:
The initial exported definition does not work.

Koenkk, 2024-03-20:
And what if you add 

```
    endpoint: (device) => {
        return {default: 2};
    },
```

Einar-Vaagland, 2024-03-21:
No, still unsupported, and there are errors when I click the Exposed states:
No converter available for 'state_1' ("ON")
No converter available for 'state_2' ("ON")

Koenkk, 2024-03-24:
Can you try with: https://gist.github.com/Koenkk/d2beb5ea646eabc3e08f9bca06c5d075

Koenkk, 2024-04-07:
Added!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-04-07 by Koenkk