Koenkk/zigbee2mqtt #21820: Crazy behaviour for IKEA LED2201G8

LeoeLeoeL, 2024-03-16:
### What happened?

Device IKEA LED2201G8 (firmware 3.0.10) goes in pairing mode or in "power on behaviour" every few minutes.
After a while, device is no more present in the Devices List.

### What did you expect to happen?

I expect a normal behaviour as a bulb should do

### How to reproduce it (minimal and precise)

I have only to wait few minutes (no more than 2) and the lamp switches on (if power on behaviour is ON) or begins to "pulse high and low".

### Zigbee2MQTT version

 1.36.0 commit: 2a53b8e5

### Adapter firmware version

 20230922

### Adapter

 zStack3x0 (Sonoff P)

### Setup

Proxmox CT

mrmaximas, 2024-03-16:
may be this is the lamp issue?? 
i have some the same lamps, not any problem on latest dev + sonoff p w 20221209

LeoeLeoeL, 2024-03-16:
I bought yesterday 2 LED2201G8 and put in the kitchen. Same behaviour for both.
Now they are paired with Hue Bridge (waiting for a Z2M fix) and are working without any issue.

srett, 2024-03-16:
Do you have `permit-join: true` in your config, or joining permanently permitted e.g. via the toggle in HA? If yes try disabling it until you actually need to pair new devices.

LeoeLeoeL, 2024-03-16:
[image]
[image]

But this isn't the problem.
Now I paired them from Hue to Deconz and they work perfectly.
So that...

LeoeLeoeL, 2024-03-21:
Update:
These devices have a problem with Deconz too.
Testing again on Hue; It seems they are working OK.

LeoeLeoeL, 2024-03-29:
I confirm these devices don't have problems under HUE.
Hue bridge always manages lights better than any other "generic" coordinator.
Why?

LeoeLeoeL, 2024-04-03:
I tried again with the latest 1.36.1 but same (bad) behaviour.

LeoeLeoeL, 2024-04-04:
I found another strange bug in Zigbee2MQTT.
As I said, these devices, when paired, disappear from DEVICES list after a while.
I show what I'm saying.
The devices are called LUCE CUCINA W1 and LUCE CUCINA W2.
Sorting in alphabetical order, they should be between...
[image]
but they aren't.
But they continue to exist in LOGS

[image]

LeoeLeoeL, 2024-07-20:
Update to 3.0.22.
Now the issue of re-pairing is fixed but lamps switch on "alone" after a while if POWER ON BEHAVIOR is ON.
Anyway, I'm not really sure if the latest firmware fixed the first problem or some following release of Z2M (Now 1.39.0).

LeoeLeoeL, 2024-07-20:
Update; Nothing has changed. The first of two lamps is no more in the mesh now.

LeoeLeoeL, 2024-07-22:
I continue to write these "blind messages".
Even with the latest firmware 3.0.22, Z2M and deconz switch on the lamps after minutes.
The same lamps work perfectly if paired with a HUE Bridge,

LeoeLeoeL, 2024-07-22:
From https://www.ikea.com/at/en/customer-service/product-support/smart-lighting/how-to-use-smart-lighting-pub53d86412
COMPATIBILITY
Which devices are compatible with the IKEA smart lighting products?
For the time being, our smart lighting products are certified to work with Amazon Alexa, Apple HomeKit and Google Assistant, provided you have a TRÅDFRI gateway. Also, since the IKEA smart lighting products use the Zigbee Light Link protocol, they are compatible with other Zigbee certified products. In addition, other device manufacturers may have made proprietary solutions in order to control IKEA smart lighting products without IKEA support.

Do the IKEA smart lighting products work with the Philips Hue Bridge?
Yes, you can use the IKEA smart lighting products together with the Philips Hue Bridge.

closed 2025-01-07 by github-actions[bot]

LeoeLeoeL, 2025-01-30:
I went to Ikea and swapped my 2 LED2201G8 with new ones.
Previous bulbs had 2030 (or 2330) printed on the box; New boxes have 2439.
I found FW 3.08. I tested and the issue was present.
I updated to 3.0.23 and the issue is still present.
LAST SEEN feature shows these devices speak too much; values normally don't go beyond 40 seconds.
Another thing "strange" is BIND and REPORTING sections are empty. All other bulbs in the network have something inside.
For instance, this is the LED1836G9 (TRADFRI bulb E26/E27, warm white, globe, opal, 806 lm)

[image]

[image]

So that the question is: " are we sure LED2201G8 are integrated in a good way?"