Koenkk/zigbee2mqtt #21822: problem adding new Ikea Devices

WilkoKetelaar, 2024-03-16:
### What happened?

After update to 1.36.0 it is not possible to add new ikea devices. 

There is an error in the log file:

Failed to configure 'Remote Keuken 1', attempt 4 (TypeError: Cannot read properties of undefined (reading 'split') at Object.configure (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/devices/ikea.ts:784:52) at Configure.configure (/opt/zigbee2mqtt/lib/extension/configure.ts:122:37) at Configure.onMQTTMessage (/opt/zigbee2mqtt/lib/extension/configure.ts:55:32) at EventEmitter.wrappedCallback (/opt/zigbee2mqtt/lib/eventBus.ts:174:23) at EventEmitter.emit (node:events:529:35) at EventBus.emitMQTTMessage (/opt/zigbee2mqtt/lib/eventBus.ts:115:22) at MQTT.onMessage (/opt/zigbee2mqtt/lib/mqtt.ts:141:27) at WebSocket.<anonymous> (/opt/zigbee2mqtt/lib/extension/frontend.ts:130:27) at WebSocket.emit (node:events:517:28) at Receiver.receiverOnMessage (/opt/zigbee2mqtt/node_modules/ws/lib/websocket.js:1209:20))

### What did you expect to happen?

Add all ikea devices normaly

### How to reproduce it (minimal and precise)

Install 1.36.0 and add ikea device

### Zigbee2MQTT version

1.36.0

### Adapter firmware version

20230507

### Adapter

sonof zbdongle-p

### Setup

on vm in proxmox

mrskycriper, 2024-03-18:
It seems to fail [here](https://github.com/Koenkk/zigbee-herdsman-converters/blob/481de6dbca424c63f9ae7da6fb6da6e344e92f1a/src/devices/ikea.ts#L784C36-L784C51), trying to parse the firmware version. 

What is the reported firmware version for this device? And if there are multiple different devices (with different models), please check all of them.

sandos, 2024-03-19:
Fairly sure I added the new Vallhorn on 1.36.0 so yes it likely works.

WilkoKetelaar, 2024-03-21:
Sorry for the late reply 

After further testing it still seems impossible to add new ikea devices to 1.36.0, today i try to add ikea parasoll but the device keeps failing interview and seems not supported.

Last week i added a bathurst water leakige sensor but this seems to be added corect but never changes update on leak...

also the remote i added last week seems to wotk but never updates its state, no battery and no button push.

After all this i added a sonof ZB mini and this seems to add without any problems...

To me it seems there is someting very wrong with 1.36.0-dev commit: [a4f9c49a], zigbee_herdsman_converters_version
19.2.1, zigbee_herdsman_version 0.38.1

It would be nice to have a working version again

sandos, 2024-03-21:
Are you aware that the new Ikea devices seem not to like alkaline / 1.5V batteries? People claim that the rechargeable Ikea batteries work better, which I just happen to just normally, so no issue for me. Specifically what I read was for false detections in Vallhorn, though, and battery draw.

WilkoKetelaar, 2024-03-21:
I have multiple vallhorns working fine when they were added before the update, after the update no ikea device like remote N2, parasoll are able to be added succesfully. battery are tested with rechargeble 1.2 volt and alkaline 1.5 volt and no difference. 

This problem is driving me crazy....

mrskycriper, 2024-03-21:
> I have multiple vallhorns working fine when they were added before the update, after the update no ikea device like remote N2, parasoll are able to be added succesfully. battery are tested with rechargeble 1.2 volt and alkaline 1.5 volt and no difference.
> 
> This problem is driving me crazy....

I’ve recently added VALLHORN sensor and multiple TRADFRI dimmers without any issues on dev release based on 1.36.0. Can you please provide zigbee-herdsman debug logs for unsuccessful pairing attempts of each device model?

WilkoKetelaar, 2024-03-21:
i need to check how to create such a log filem, is there any discryption on how to make this log?

mrskycriper, 2024-03-21:
> i need to check how to create such a log filem, is there any discryption on how to make this log?

https://www.zigbee2mqtt.io/guide/usage/debug.html#enabling-logging

Bruhildo, 2024-04-25:
@WilkoKetelaar did you had any luck pairing Parasoll? I am having the same issue with Parasoll and Vallhorn.

WilkoKetelaar, 2024-04-26:
No i just did a complete reinstall of Zigbee2mqtt and started from the beginning adding all device's again...

closed 2024-11-23 by github-actions[bot]