Koenkk/zigbee2mqtt #21823: [Feature request]: Helm chart

jlpedrosa, 2024-03-16:
### Is your feature request related to a problem? Please describe

Not a problem. 

### Describe the solution you'd like

Provide a helm chart that enables deploying zigbee2mqtt in kubernetes clusters via helm releases.

### Describe alternatives you've considered

The non-official charts available around, seem not very well maintained and/or updated frequently.

### Additional context

I am happy to contribute to the project and provide the helm charts for it. Tipically maintaining the images updated is relatively simple as long as we can enable renovate/dependabot on the repository.

Koenkk, 2024-03-17:
> I am happy to contribute to the project and provide the helm charts for it. Tipically maintaining the images updated is relatively simple as long as we can enable renovate/dependabot on the repository.

I'm fine with this, please make a pull request

> The non-official charts available around, seem not very well maintained and/or updated frequently.

What maintenance is needed?

jlpedrosa, 2024-03-17:
> What maintenance is needed?

It's not a lot, but also it's not 0. Some examples of maintenance:
* A new parameter is added to the configuration.yaml, probably it should be added to the Helm chart too (depends how we encode them)
* A new release of zigbee2mqtt is created (so a new docker image tag is released), we should update the default image in the chart (this typically can be done with renovate/dependabot).
* Kubernetes deprecates an API (Service, Deployment, ConfigMap) we would have to update the chart, based on the objects required for this object, this changes will be extremely infrequent as the API is far away from alpha/beta.

Koenkk, 2024-03-17:
> A new release of zigbee2mqtt is created (so a new docker image tag is released), we should update the default image in the chart (this typically can be done with renovate/dependabot).

This can be done in https://github.com/Koenkk/zigbee2mqtt/blob/master/.github/workflows/ci.yml

For the other points, I don't see how to automate them (that's fine)

jlpedrosa, 2024-03-17:
I'm currently playing with the chart, I'm getting out of the configuration.yaml as much as I can to other files. Kubernetes does not allow to modify the ConfigMaps from the container/pod. This means that the features that are managed through the UI but are not in a separate file, will fail to get updated. 

I could potentially dump the config into a volume with an init container, but that means that the config could not be managed via helm. Tradeoffs.

jlpedrosa, 2024-05-06:
@Koenkk  let's keep this open until we have tested it e2e and the documentation is merged?

pmarques, 2024-05-18:
Maybe is worth to just cross reference the new helm chart repo https://github.com/Koenkk/zigbee2mqtt-chart/pull/1

vsigler, 2024-06-01:
@pmarques Thanks! I was just looking where the helm chart suddenly went. :)

closed 2024-12-30 by github-actions[bot]