Koenkk/zigbee2mqtt #21824: AssertionError [ERR_ASSERTION]: lcw004 has multiple ota - > zigbee2mqtt does not start anymore?

Sweti934, 2024-03-16:
### What happened?

I added a few more lights to my network (43 devices). Then suddenly the zigbee2mqtt froze, and now i get this startup error message?

### What did you expect to happen?

A normal Startup...

### How to reproduce it (minimal and precise)

Try to start zigbee2mqtt

### Zigbee2MQTT version

 1.36.0-1 

### Adapter firmware version

not findable due to not starting frontend

### Adapter

Sonoff STick 2.0

### Setup

Z2M on Home assistant as Addon...

### Debug log

[17:18:23] INFO: Preparing to start...
[17:18:23] INFO: Socat not enabled
[17:18:23] INFO: Starting Zigbee2MQTT...
Zigbee2MQTT:info  2024-03-16 17:18:24: Logging to console and directory: '/config/zigbee2mqtt/log/2024-03-16.17-18-24' filename: log.txt
Zigbee2MQTT:info  2024-03-16 17:18:24: Starting Zigbee2MQTT version 1.36.0 (commit #unknown)
Zigbee2MQTT:info  2024-03-16 17:18:24: Starting zigbee-herdsman (0.35.1)
Zigbee2MQTT:error 2024-03-16 17:18:25: Failed to start zigbee
Zigbee2MQTT:error 2024-03-16 17:18:25: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2024-03-16 17:18:25: Exiting...
Zigbee2MQTT:error 2024-03-16 17:18:25: AssertionError [ERR_ASSERTION]: 'LCW004' has multiple 'ota', this is not allowed
    at processExtensions (/app/node_modules/zigbee-herdsman-converters/src/index.ts:119:32)
    at prepareDefinition (/app/node_modules/zigbee-herdsman-converters/src/index.ts:182:18)
    at findDefinition (/app/node_modules/zigbee-herdsman-converters/src/index.ts:294:16)
    at Object.findByDevice (/app/node_modules/zigbee-herdsman-converters/src/index.ts:266:22)
    at Device.resolveDefinition (/app/lib/model/device.ts:36:31)
    at Zigbee.start (/app/lib/zigbee.ts:69:13)
    at Controller.start (/app/lib/controller.ts:109:27)
    at start (/app/index.js:107:5)

sjorge, 2024-03-16:
Are you using an external converter? I can't find any reference to LCW004 in the zhc repo.

pixolio, 2024-04-03:
Have you perhaps installed an PHILIPS Hue Dymera wall light (with identification LCW004)? 
I've the same problem in my installation (after installing this lamp)

Koenkk, 2024-04-03:
This should be fixed in z2m 1.36.1

pixolio, 2024-04-03:
thanks @Koenkk for reaction...

...but meanwhile I have updated to 1.36.1 and the error is still there:

```
[22:31:21] INFO: Preparing to start...
[22:31:22] INFO: Socat not enabled
[22:31:24] INFO: Starting Zigbee2MQTT...
Zigbee2MQTT:info  2024-04-03 22:31:30: Logging to console and directory: '/config/zigbee2mqtt/log/2024-04-03.22-31-30' filename: log.txt
Zigbee2MQTT:info  2024-04-03 22:31:30: Starting Zigbee2MQTT version 1.36.1 (commit #unknown)
Zigbee2MQTT:info  2024-04-03 22:31:30: Starting zigbee-herdsman (0.40.3)
Zigbee2MQTT:error 2024-04-03 22:32:36: Failed to start zigbee
Zigbee2MQTT:error 2024-04-03 22:32:36: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2024-04-03 22:32:36: Exiting...
Zigbee2MQTT:error 2024-04-03 22:32:36: AssertionError [ERR_ASSERTION]: 'LCW004' has multiple 'ota', this is not allowed
    at processExtensions (/app/node_modules/zigbee-herdsman-converters/src/index.ts:124:28)
    at prepareDefinition (/app/node_modules/zigbee-herdsman-converters/src/index.ts:173:18)
    at findDefinition (/app/node_modules/zigbee-herdsman-converters/src/index.ts:285:16)
    at runNextTicks (node:internal/process/task_queues:60:5)
    at processTimers (node:internal/timers:509:9)
    at Object.findByDevice (/app/node_modules/zigbee-herdsman-converters/src/index.ts:257:22)
    at Device.resolveDefinition (/app/lib/model/device.ts:36:31)
    at Zigbee.start (/app/lib/zigbee.ts:69:13)
    at Controller.start (/app/lib/controller.ts:109:27)
    at start (/app/index.js:107:5)
```

lfilho, 2024-04-09:
Same thing started happening with me today. But my error has LCW005 in it instead of LCW004: "LCW005' has multiple 'ota', this is not allowed"

I have just upgraded to q.31.1-1 but it didn't solve: same error. The I also upgraded home assistant to latest (2024.4.2), same thing. 

I did t add or remove any device today. My wife turned a light up with the companion app but then when she went to turn it off it wasn't working anymore. I don't have any sort auto updates enabled so not sure what happened in this meantime.

lfilho, 2024-04-10:
Ah i see, so there's hope 🙌 thanks!
How about we keep the issue open until it's merged to main?

lfilho, 2024-04-11:
Thanks for reopening! Just out of curiosity, no pressure at all, any rough ideas when we'll see it on main? Thanks again

Koenkk, 2024-04-11:
With the new release which is every 1st of the month

lfilho, 2024-04-11:
Gotcha, thank you!

closed 2024-11-10 by github-actions[bot]