Koenkk/zigbee2mqtt #21825: Error: Cluster 'liXeePrivate' has no attribute 'motDEtat'

quentinus95, 2024-03-16:
### What happened?

Everytime the Lixee device is polled, Zigbee2MQTT displays the following error log:

> Error: Cluster 'liXeePrivate' has no attribute 'motDEtat'

The device works properly.

### What did you expect to happen?

No error in logs, or a way to silent it.

### Zigbee2MQTT version

1.36.0

### Adapter firmware version

20210708

### Adapter

Sonos

### Setup

Containerized Zigbee2MQTT

Minims, 2024-05-01:
Same issue with the 1.37  version updated today

Sabjul33, 2024-05-04:
Hi, Same thing for me V 1.36 updated to 1.37
Failed to read zigbee attributes: Error: Cluster 'liXeePrivate' has no attribute 'motDEtat'

Koenkk, 2024-05-05:
@SilentT-FR it seems `motDEtat` is missing from the [cluster](https://github.com/Koenkk/zigbee-herdsman/blob/1b76db9a8afd599162340cf98e0bad0b6ad351c7/src/zcl/definition/cluster.ts#L4578), do you know the attribute ID and data type?

scientas, 2024-05-05:
> @SilentT-FR it seems `motDEtat` is missing from the [cluster](https://github.com/Koenkk/zigbee-herdsman/blob/1b76db9a8afd599162340cf98e0bad0b6ad351c7/src/zcl/definition/cluster.ts#L4578), do you know the attribute ID and data type?

Hello, same problem...
Found that on instruction manual :
Add MOTDETAT value (from historic mode) --> Cluster : 0xFF66 - Attribut :
0x0009
Add null value for all of default metering properties
Add watchdog reset process when hardfault error
Fix duplicate voltage acquisition
Fix operator error for comparison
Hope this help...

Koenkk, 2024-05-05:
Does the manual also mention the data type?

scientas, 2024-05-05:
It seems to be "string" with "000000" default value

Koenkk, 2024-05-06:
Added it!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-12-03 by github-actions[bot]