Koenkk/zigbee2mqtt #21828: Extension of Calibration Range for Thermostat  (AVATTO) ME168

DoWenzl94, 2024-03-16:
### What happened?

Extension of Calibration Range for Thermostat  (AVATTO) ME168 in Tuya.js

Currently, in the Tuya.js file, the calibration range for the thermostat (AVATTO) ME168 is limited to -3 to 3. However, there are scenarios where a wider calibration range is needed. I have successfully extended the calibration range from -3 to 3 to -9 to 9 by manually modifying the relevant section in the Tuya.js file.

I propose to officially extend the calibration range for the thermostat in Tuya.js to accommodate broader needs. This would provide users with greater flexibility and better customization options for their thermostats:

fingerprint: tuya.fingerprint('TS0601', [
            '_TZE200_bvu2wnxz', /* model: 'ME167', vendor: 'AVATTO' */
            '_TZE200_6rdj8dzm', /* model: 'ME167', vendor: 'AVATTO' */
            '_TZE200_p3dbf6qs', /* model: 'ME168', vendor: 'AVATTO' */
            '_TZE200_rxntag7i', /* model: 'ME168', vendor: 'AVATTO' */
        ]),
        model: 'TS0601_thermostat_3',
        vendor: 'TuYa',
        description: 'Thermostatic radiator valve',
        fromZigbee: [tuya.fz.datapoints],
        toZigbee: [tuya.tz.datapoints],
        whiteLabel: [
            tuya.whitelabel('AVATTO', 'ME167', 'Thermostatic radiator valve', ['_T>
            tuya.whitelabel('AVATTO', 'ME168', 'Thermostatic radiator valve', ['_T>
        ],
        onEvent: tuya.onEventSetTime,
        configure: tuya.configureMagicPacket,
        exposes: [
            e.child_lock(), e.battery_low(),
            e.climate()
                .withSetpoint('current_heating_setpoint', 5, 35, 1, ea.STATE_SET)
                .withLocalTemperature(ea.STATE)
                .withSystemMode(['auto', 'heat', 'off'], ea.STATE_SET)
                .withRunningState(['idle', 'heat'], ea.STATE)
                .withLocalTemperatureCalibration(-9, 9, 1, ea.STATE_SET),   <--------------- HERE

### What did you expect to happen?

By extending the calibration range, users can fine-tune the behavior of their thermostats more precisely, especially in scenarios where wider temperature variations need to be accommodated. This enhancement would improve the overall usability and flexibility of Zigbee2mqtt for users with diverse requirements.

### Zigbee2MQTT version

1.36.0

### Adapter firmware version

20230507

### Adapter

CC2652P

### Setup

Pi4

Koenkk, 2024-03-17:
Updated.

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-03-17 by Koenkk

Koenkk, 2026-01-30:
Done!

> Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.