Koenkk/zigbee2mqtt #21829: Philips 929003017102 RDM004

robertteir, 2024-03-16:
### What happened?

LevelCtrl not working

Config with RDM004:

Source 1 Destination Coordinator: PowerCfg
Source 1 Destination Group: LevelCtrl and OnOff
Device mode: Single_push_button

Result: Single press results in OnOff and hold also results in OnOff.

(there is also no sync between the button and light state, if you turn of the light in HA and then you have to push the button twice to turn it on).

Config with RDM001 (working):

Source 1 Destination Coordinator: PowerCfg
Source 1 Destination Group: LevelCtrl and OnOff
Device mode: Single_push_button

Result: Single press results in OnOff and hold results in dimming.

(sync between light and button works as expected)

### What did you expect to happen?

holding button would result in dimming.

### How to reproduce it (minimal and precise)

Source 1 Destination Coordinator: PowerCfg
Source 1 Destination Group: LevelCtrl and OnOff
Device mode: Single_push_button

hold button

### Zigbee2MQTT version

1.36.0-1

### Adapter firmware version

26720700

### Adapter

ConBee2/RaspBee2

### Setup

vm in proxmox

c5n, 2024-07-07:
Hi, same issue here.
Just installed some older RDM001 (Firmware 1.0.5 / 20210504) with long press level control working as expected.
But the newer RDM004 ( Firmware 2.60.16 / 20220510) only switches the group on/off. No level control possible.

During setup I discovered, that on RDM001, LevelCtrl was bound with coordinator after pairing.
But RDM004 had no check-mark on LevelCtrl...

saoc90, 2024-09-23:
So do you think thats a firmware issue then? Is it possible to downgrade newer models as well?

robertteir, 2024-10-05:
Did some testing and I can't see the brightness_move_up/brightness_move_down event from the RDM004

c5n, 2024-11-10:
I would just like to add that my RDM004 module has the model number 9290030171A on the housing.
Is there any news on this misconduct?

closed 2025-01-17 by github-actions[bot]