Koenkk/zigbee2mqtt #21830: TS0504B RGBW controller does not expose `color_power_on_behavior` or `do_not_disturb`

bwindsor, 2024-03-16:
### What happened?

I should be able to set the power on behavior for this device.

### What did you expect to happen?

I am unable to set the power on behavior for this device through the Z2M frontend.

I notice that the TS0503B, which is exactly the same device without the white channel, does support these.

[image]

The only difference in the code is that the TS0504B has an additional custom color converter to handle the white channel:

https://github.com/Koenkk/zigbee-herdsman-converters/blob/523c1e7147a2178382c9cb883e047bffdd0c2a18/src/devices/tuya.ts#L1104-L1122

Is this extra `toZigbee` converter somehow removing the `color_power_on_behavior` and `do_not_disturb` exposes?

(aside, my device also has white label `{vendor: 'BTF-Lighting', model: 'C04Z'}` which is not included in the converters.)

### How to reproduce it (minimal and precise)

Connect a TS0504B device to a Zigbee network. It appears in Z2M but only light attributes are exposed. After a power interruption the lights come on and I can't change this from Z2M.

### Zigbee2MQTT version

1.34.0

### Adapter firmware version

0x26580700

### Adapter

ConBee2

### Setup

HA add-on, raspberry pi 4, HAOS.

Koenkk, 2024-03-17:
All these options should be present in the latest [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

bwindsor, 2024-03-17:
Right, sorry for unnecessary noise, I hadn't understood that the master branch of herdsman converters isn't what's in production, and z2m just refers to a particular release of the converters. I'll close this and test the dev branch and will reopen if I find the issue still exists.

closed 2024-03-17 by bwindsor