Koenkk/zigbee2mqtt #21836: SmartSense Z Sensor not pairing

gavinkleckner, 2024-03-17:
### What happened?

The SmartSense by digi Z sensor fails to interview.

### How to reproduce it (minimal and precise)

Hold down the button on the sensor until the green light starts rapidly flashing, indicating attempting to transmit that should initiate pairing

### Zigbee2MQTT version

1.36.0

### Adapter firmware version

6.10.3.0 build 297

### Adapter

Aeotec zi-stick ZGA008

### Setup

Add-on on Home Assistant OS on an HP elite desk 800 G4

### Debug log

2024-03-16 22:29:11Received Zigbee message from 'Coordinator', type 'commandNotification', cluster 'greenPower', data '{"data":[25,19,2,11,254,0],"type":"Buffer"}' from endpoint 242 with groupID null, ignoring since it is from coordinator
2024-03-16 22:30:02Saving state to file /config/zigbee2mqtt/state.json
2024-03-16 22:30:05Received Zigbee message from 'Coordinator', type 'commandNotification', cluster 'greenPower', data '{"data":[25,20,2,10,0,0],"type":"Buffer"}' from endpoint 242 with groupID null, ignoring since it is from coordinator

closed 2024-10-14 by github-actions[bot]

tristanalbano, 2025-03-02:
Hi, do you have any update on this?  I am trying to connect these sensors too and also can’t get past “starting interview”

gavinkleckner, 2025-12-02:
NOPE NO UPDATE. If you have info for me I’ll take it.