Koenkk/zigbee2mqtt #21842: Xiaomi GZCGQ01LM fails to configure

kremik, 2024-03-17:
### What happened?

Device is trying to configure itself without a request. The configuration is failing. Getting the same error even when the configuration is started via a button in Z2M web. 

### What did you expect to happen?

That the device will be configured without error

### How to reproduce it (minimal and precise)

Click on reconfigure in Z2M web.

### Zigbee2MQTT version

1.36.0

### Adapter firmware version

20210211

### Adapter

CC2652P

### Setup

RPi 4B 64bit, plain NOT containerized

### Debug log

Debug 2024-03-17 09:06:50Received MQTT message on 'zigbee2mqtt/bridge/request/device/configure' with data '{"id":"lightsensor1","transaction":"qvocs-1"}'
Info 2024-03-17 09:06:50Configuring 'lightsensor1'
Error 2024-03-17 09:07:00Failed to configure 'lightsensor1', attempt 3 (Error: Bind 0x54ef44100030211e/1 msIlluminanceMeasurement from '0x00124b0022ab72f1/1' failed (AREQ - ZDO - bindRsp after 10000ms) at Object.start (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/waitress.ts:63:23) at /opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:692:28 at Queue.execute (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/queue.ts:35:20) at Endpoint.bind (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:551:13) at Object.bind (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:43:9) at configure (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/devices/lumi.ts:2229:13) at Object.configure (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/index.ts:141:25) at Configure.configure (/opt/zigbee2mqtt/lib/extension/configure.ts:122:13) at Configure.onMQTTMessage (/opt/zigbee2mqtt/lib/extension/configure.ts:55:21) at EventEmitter.wrappedCallback (/opt/zigbee2mqtt/lib/eventBus.ts:174:17))
Info 2024-03-17 09:07:00MQTT publish: topic 'zigbee2mqtt/bridge/response/device/configure', payload '{"data":{"id":"lightsensor1"},"error":"Failed to configure (Bind 0x54ef44100030211e/1 msIlluminanceMeasurement from '0x00124b0022ab72f1/1' failed (AREQ - ZDO - bindRsp after 10000ms))","status":"error","transaction":"qvocs-1"}'

elgerg, 2024-06-15:
I can second this, on version [1.36.1](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.36.1) commit: [ffc2ff1d](https://github.com/Koenkk/zigbee2mqtt/commit/ffc2ff1d)

Coordinator revision - 20230507

Failed to configure 'LandingLumin', attempt 1 (Error: ZCL command 0x04cf8cdf3c7f243b/1 genPowerCfg.read(["batteryVoltage"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 13418 - 1 - 115 - 1 - 1 after 10000ms) at Object.start (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/waitress.ts:63:23) at ZStackAdapter.sendZclFrameToEndpointInternal (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:499:47) at Queue.execute (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/queue.ts:35:20) at Request.send (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:79:20) at Endpoint.zclCommand (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:760:28) at Endpoint.read (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:404:24) at configure (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/devices/lumi.ts:2199:13) at Object.configure (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/index.ts:162:21) at Configure.configure (/opt/zigbee2mqtt/lib/extension/configure.ts:122:13) at EventEmitter.wrappedCallback (/opt/zigbee2mqtt/lib/eventBus.ts:174:17))

However the device does work as intended.

elgerg, 2024-06-15:
Looks fixed in 1.37.0 but that version breaks my brightness stuff..

RubenKelevra, 2024-09-02:
> Looks fixed in 1.37.0 but that version breaks my brightness stuff..

Is it still broken in 1.40.0? I haven't noticed any issues with this device in 1.39.1 and 1.40.0.

closed 2025-01-07 by github-actions[bot]