Koenkk/zigbee2mqtt #21844: [New device support]: GiEX Water Valve GX02

dgaust, 2024-03-17:
### Link

https://www.aliexpress.com/item/1005006547328147.html

### Database entry

nil

### Comments

This is an updated version of the QT06 devices that are currently supported by zigbee2mqtt, I have managed to create an external definitions file for the device which works. It's the same as the QT06_2 device, just a different fingerprint.

### External definition

```shell
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const legacy = require('zigbee-herdsman-converters/lib/legacy');
const MINUTES_IN_A_DAY = 1440;
const SECONDS_IN_12_HOURS = 43200;

module.exports = [
    {
        fingerprint: [
            {modelID: 'TS0601', manufacturerName: '_TZE204_7ytb3h8u'},
        ],
        model: 'GX02',
        vendor: 'GiEX',
        description: 'GX02 GiEX Water Valve',
        extend: [],
        meta: {},
        onEvent: tuya.onEventSetLocalTime,
        fromZigbee: [legacy.fromZigbee.giexWaterValve],
        toZigbee: [legacy.toZigbee.giexWaterValve],
        exposes: [
            e.battery(),
            e.binary(legacy.giexWaterValve.state, ea.STATE_SET, 'ON', 'OFF')
                .withDescription('State'),
            e.enum(legacy.giexWaterValve.mode, ea.STATE_SET, ['duration', 'capacity'])
                .withDescription('Irrigation mode'),
            e.numeric(legacy.giexWaterValve.cycleIrrigationNumTimes, ea.STATE_SET)
                .withValueMin(0)
                .withValueMax(100)
                .withDescription('Number of cycle irrigation times, set to 0 for single cycle'),
            e.numeric(legacy.giexWaterValve.irrigationStartTime, ea.STATE)
                .withDescription('Last irrigation start time'),
            e.numeric(legacy.giexWaterValve.irrigationEndTime, ea.STATE)
                .withDescription('Last irrigation end time'),
            e.numeric(legacy.giexWaterValve.lastIrrigationDuration, ea.STATE)
                .withDescription('Last irrigation duration'),
            e.numeric(legacy.giexWaterValve.waterConsumed, ea.STATE)
                .withUnit('L')
                .withDescription('Last irrigation water consumption'),
            e.numeric(legacy.giexWaterValve.irrigationTarget, ea.STATE_SET)
                .withValueMin(0)
                .withValueMax(SECONDS_IN_12_HOURS)
                .withUnit('seconds or litres')
                .withDescription('Irrigation target, duration in seconds or capacity in litres (depending on mode), ' +
                    'set to 0 to leave the valve on indefinitely, ' +
                    'for safety reasons the target will be forced to a minimum of 10 seconds in duration mode'),
            e.numeric(legacy.giexWaterValve.cycleIrrigationInterval, ea.STATE_SET)
                .withValueMin(0)
                .withValueMax(SECONDS_IN_12_HOURS)
                .withUnit('sec')
                .withDescription('Cycle irrigation interval'),
        ],
    },
];
```

Koenkk, 2024-03-17:
Added!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-03-17 by Koenkk

robin0793, 2024-06-06:
Hi, i have the same device, but it reports as _TZE200_2ytb3h8u.
I added the external definition from the mentioned commit, seems to work fine.

[image]

dgaust, 2024-06-06:
Interesting, wonder what the differences are (if any)

image.png (view on web)<https://github.com/Koenkk/zigbee2mqtt/assets/21172172/e41660d6-5c63-4db5-b139-983601d4af61>

Federfox, 2024-06-07:
I have the exact same valve but it has a different Vendor ID, would it be possible to add manufacturerName _TZE284_7ytb3h8u ?
I have other identical tubes with a recognized vendor, I checked the board and it is identical.

Thank you!

hazmemoria, 2024-07-01:
I have the exact same valve but it has a different Vendor ID, would it be possible to add manufacturerName _TZE284_7ytb3h8u ?
Thank You
[image]

acuszka, 2024-07-01:
> I have the exact same valve but it has a different Vendor ID, would it be possible to add manufacturerName _TZE284_7ytb3h8u ? Thank You ![Screenshot 2024-07-01 10 29 17](https://private-user-images.githubusercontent.com/156531935/344600969-de70a78e-b6a0-4026-99ff-b1d1c140ba6e.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTk4MjMyMTAsIm5iZiI6MTcxOTgyMjkxMCwicGF0aCI6Ii8xNTY1MzE5MzUvMzQ0NjAwOTY5LWRlNzBhNzhlLWI2YTAtNDAyNi05OWZmLWIxZDFjMTQwYmE2ZS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNzAxJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDcwMVQwODM1MTBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1iMmMyZmM0M2VmZmJlZjQ2ZGU1NjE4NjNmYjYxZmJjNjc1NmFlY2VhYTgzZGNmNzIzNDI2MzRiMmI3MGIzY2RlJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.5ftRg2umkrM6oyCT0UlFMMsEGMZrw22cjUS5v-h9PT0)

I also have _TZE284_7ytb3h8u and it shows as unsupported.

oliverwoodings, 2024-07-01:
7ytb3h8u is already supported in the herdsman converters: https://github.com/Koenkk/zigbee-herdsman-converters/commit/dbef5e6b8c7b98ef230860880ac225eccfa2eb31

Likely you just need to wait for the next release of zigbee2mqtt so those changes get through. If you're impatient, switch to the dev branch and you'll get the latest version of herdsman (if you're using the HA addon, you need to switch to the 'Edge' version of z2m)

acuszka, 2024-07-02:
> 7ytb3h8u is already supported in the herdsman converters: [Koenkk/zigbee-herdsman-converters@dbef5e6](https://github.com/Koenkk/zigbee-herdsman-converters/commit/dbef5e6b8c7b98ef230860880ac225eccfa2eb31)
> 
> Likely you just need to wait for the next release of zigbee2mqtt so those changes get through. If you're impatient, switch to the dev branch and you'll get the latest version of herdsman (if you're using the HA addon, you need to switch to the 'Edge' version of z2m)

I confirm it's working with new zigbee2mqtt 1.39

[image]

feserdel, 2024-08-19:
Good morning, in my case, 

_TZE200_7ytb3h8u is detected wrongly as QT06_2 (but is GX02, the green one), however, another valve that I own _TZE204_7ytb3h8u is detected correctly as [GX02]

What is the solution?

[image]
[image]

NOTE: Im using Home Assistant add-on

Versión de Zigbee2MQTT [1.39.1]
zigbee-herdsman-converters version 19.72.0

hellomarb, 2024-08-23:
I have the same issue:

```
Zigbee-Model: TS0601
Zigbee-Manufacturer: _TZE200_7ytb3h8u
Model: QT06_2
``` 
But it should be the GX02

heidle78, 2024-08-29:
Here the same issues, correlation to the right valve GX02 not working - I have the green version
[image]

heidle78, 2024-08-29:
Deleted the device again and took now the DEV branch - no change:

[image]

jochenruben, 2024-08-31:
Hello,
I have the same problem three GX02 are recognised as QT06_2. Two of them are working. One can‘t be switched.
Will there be a solution?
Thanx a lot!

jochenruben, 2024-08-31:
> Hello, I have the same problem three GX02 are recognised as QT06_2. Two of them are working. One can‘t be switched. Will there be a solution? Thanx a lot!

Zigbee2MQTT-Version
[1.39.1](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.39.1) commit: [unknown](https://github.com/Koenkk/zigbee2mqtt/commit/unknown) With the latest Version of HA as adon

heidle78, 2024-09-01:
I updated the firmware and even there the same result:
[image]

heidle78, 2024-09-01:
It seems I can control, but the picture is definitely not correct - I have the green version

jochenruben, 2024-09-02:
Think I've identified the problem. I bought three GX02-ZG. Supported by Z2MQTT is GX02 (https://www.zigbee2mqtt.io/devices/GX02.html). Both are green and GX02-ZG is the lates version.
Sims like GX02-ZG are recognized as QT06_2. I got all three work as QT06_2, but they need to have a very good connectivity.

idogrf, 2024-09-02:
opened a PR to fix the GUI for `_TZE200_7ytb3h8u` - https://github.com/Koenkk/zigbee-herdsman-converters/pull/7935

idogrf, 2024-09-10:
as of version `1.40.1`, `_TZE200_7ytb3h8u` now appears as the correct version in the GUI

iLLNESS, 2025-10-04:
Is there a way to change the TZ this device reports for time?