Koenkk/zigbee2mqtt #21845: Activate and Desactivate Internet Access to All zigbee Device

KeepCool1, 2024-03-17:
### What happened?

I m runing Homeassistant on Dell Optiplex. Using Conbee 2 with Zigbee2mqtt.
Core 2024.3.1
Supervisor 2024.03.0
Operating System 12.1
Frontend 20240307.0

Really don't undestand how it is possible but the fact is my  network internet connection is always hacked through the zigbee. i have change network encryption key  disable joining but still have my internet connection hacked. When i remove the Conbee 2 stick then my internet connection is no more hacked. 
Is there any ways to disable the internet access for all the zigbee devices. without disconnecting the homeassistant host from the internet?

### What did you expect to happen?

Any feature or settings that can desactivate internet access to all zigbee devices.

### How to reproduce it (minimal and precise)

I don't know how they hack the network and get access to internet. But when i disconnect the conbee2 stick they do not succeed any more to get access. 

### Zigbee2MQTT version

1.36.0-1

### Adapter firmware version

Version 2.19.02/19.9.2022

### Adapter

Conbee 2

### Setup

X86

mrmaximas, 2024-03-17:
Sounds like science fiction - lan hacking through the zigbee protocol. Is it okay that zigbee isn't IP?

jgbreezer, 2024-03-17:
"hacked" can mean a lot of different things to different people. What exactly do you mean "network internet connection is always hacked"? Do you mean that it stops working? If you are claiming "hacked" in (IMO) the most popularly-understood way for technology (that someone malicious has affected your network negatively through means they shouldn't be able to/aren't intended for those purposes), how do you know that someone is intentionally doing this, as opposed to just network problems through configuration or device setup/bugs?

KeepCool1, 2024-03-18:
Hello, Thank you for your reply. I'm not professional. I'm living in a country where everybody don't have internet home. They only purpose off these people  for the moment is to get access to the internet. They are so many connected to my network and  they stay in front of my gate. My internet then become very slow and even opening home assistant take a while. First i decide to change WIFI setting. but the succeed to connecting to the network.  

My WIFI is secure: SSID not broadcasted, WiFi Channel  changed and passphrase changed many time, Security WPA2, a long passphrase with special characters. White list activated in the router with MAC address. Although all these setting the are still connected. I always check in the Router and see no new devices (that mean for me that they are not directly connected to the router!!!!)
The only way to make them get away is to switch off the router and no more internet for all of us!!!!

I never guest Zigbee maybe the problem after trying every think. i finally  unplug the zigbee  conbee 2 stick and now they come and try to connect to the network and never succeed.
I m  upset but i don't understand how can it be possible trough zigbee!!! that why my idea is to find a way to deactivate zigbee devices from getting access to internet.
Thanks!!!!

mrmaximas, 2024-03-18:
> Hello, Thank you for your reply. I'm not professional. I'm living in a country where everybody don't have internet home. The only purpose off these people for the moment is to get access to the internet. They are so many connected to my network and they stay in front of my gate. My internet then become very slow and even opening home assistant take a while. First i decided to change WIFI setting. but they succeed to connecting to the network.
> 
> My WIFI is secure: SSID not broadcasted, WiFi Channel changed and passphrase changed many time, Security WPA2, a long passphrase with special characters. White list activated in the router with MAC address. Although all these setting they are still connected. I always check in the Router and see no new devices (that mean for me that they are not directly connected to the router!!!!) The only way to make them get away is to switch off the router and no more internet for all of us!!!!
> 
> I never guest Zigbee may be the problem; after trying every think. i finally unplug the zigbee conbee 2 stick and now they come and try to connect to the network and never succeed. I m upset but i don't understand how can it be possible trough zigbee!!! that why my idea is to find a way to deactivate zigbee devices from getting access to internet. Thanks!!!!

It is not possible to access the internet directly via zigbee network, zigbee network is not a transport for tcp/ip. you need to research how to secure your wifi network, but this project page is not discussed here.

jgbreezer, 2024-03-19:
Btw, did you disable WPS? That is insecure, cos if its enabled people can brute-force hack in using the functionality in a short time (without being able to press the WPS button).

KeepCool1, 2024-03-20:
> It is not possible to access the internet directly via zigbee network, zigbee network is not a transport for tcp/ip. you need to research how to secure your wifi network, but this project page is [...]

> It is not possible to access the internet directly via zigbee network, zigbee network is not a transport for tcp/ip. you need to research how to secure your wifi network, but this project page is [...]

Hello,  thank you for your support it is helpful for my understanding.

KeepCool1, 2024-03-20:
> Btw, did you disable WPS? That is insecure, cos if its enabled people can brute-force hack in using the functionality in a short time (without being able to press the WPS button).

> Btw, did you disable WPS? That is insecure, cos if its enabled people can brute-force hack in using the functionality in a short time (without being able to press the WPS button).

Thank you  for your great support. Before WPS was enable. But later i disable it.

if the WPS is enabled and someone brute-force. and get access to the Wifi.  Will i find the new connected device in the list of connected devices of the router? in my case i always see no new device connected to my router but many people connected.

jgbreezer, 2024-04-24:
If they hack and connect to your network via WPS, yes you would see them in the connected-devices list. If there's noone in your connected devices list but your own devices, then the only way they could actually be connected through your router is if they've hacked one of those devices and turned it into a router for themselves to hide the traffic they use via a normal device on your network; or, but much much less likely, if they hacked your Wifi router in a way such that it doesn't show up in the list of devices (those methods, just like any other, would show up in your ISP if you have a way of seeing the bandwidth in use though thats not often provided by them on their admin pages for your account). Zigbee to internet bridges/routers might be hackable but ZHA/Zigbee2MQTT with a dongle (like your Conbee, or whatever other one) don't have major known security holes as far as I know, and it'd require some real experts to hack them if there aren't publicly published ways to hack them that they can follow a published recipe on a forum/hackers chat group to use. Changing your wifi ssid and password should also make it hard to reuse it - did you have to reprogram any home devices that use wifi too? (cheap IoT wifi stuff is likely to have insecurities and be hackable, and popular things might have well-known exploits).

Its far more likely in my mind, given you saying there's nothing showing up in the router, that the slowdown you experience is from either the network using a lot of bandwidth talking to cloud providers to update them about stuff on your zigbee network or from the frequencies getting crowded from all those kids out there with wifi (which is going to be competing with your own wifi and the zigbee too, and any bluetooth around, depends what channels they're all on). Without a serious investigation we can't tell for sure.

closed 2024-11-21 by github-actions[bot]