Koenkk/zigbee2mqtt #21847: z2m 1.36.0  + Home assistant 2024. + HassAgent: Entity 'homeassistant/button' is unknown

Blacksli, 2024-03-17:
### What happened?

I'm using Home assistant 2024.3.0 in conjunction with the "Hass Agent" program, which works with HA via mqtt and provides some button entities for certain actions
Today I updated zigbee2mqtt from version 1.35.1 to version 1.36.0
and errors appeared in the z2m logs:

> debug 2024-03-17 18:34:42: Received MQTT message on 'homeassistant/button/ALEKS10/setvolume_0/config' with data '{"availability_topic":"homeassistant/sensor/ALEKS10/availability","command_topic":"homeassistant/button/ALEKS10/setvolume_0/set","action_topic":"homeassistant/button/ALEKS10/setvolume_0/action","unique_id":"6fac7892-a0ae-43b8-b998-9ff6dd167784","object_id":"ALEKS10_setvolume_0","device":{"identifiers":"hass.agent-ALEKS10","manufacturer":"HASS.Agent Team","model":"Microsoft Windows NT 10.0.19045.0","name":"ALEKS10","sw_version":"2.0.1"},"name":"setvolume_0","state_topic":"homeassistant/button/ALEKS10/setvolume_0/state"}'
> info  2024-03-17 18:34:42: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"homeassistant/button"},"type":"entity_not_found"}'
> error 2024-03-17 18:34:42: Entity 'homeassistant/button' is unknown
> debug 2024-03-17 18:34:42: Received MQTT message on 'homeassistant/button/ALEKS10/setvolume_0/state' with data 'OFF'
> info  2024-03-17 18:34:42: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"homeassistant/button"},"type":"entity_not_found"}'
> error 2024-03-17 18:34:42: Entity 'homeassistant/button' is unknown
> debug 2024-03-17 18:34:42: Received MQTT message on 'homeassistant/button/ALEKS10/setvolume_80/config' with data '{"availability_topic":"homeassistant/sensor/ALEKS10/availability","command_topic":"homeassistant/button/ALEKS10/setvolume_80/set","action_topic":"homeassistant/button/ALEKS10/setvolume_80/action","unique_id":"3584b296-deea-4ef1-8653-190df45e70c6","object_id":"ALEKS10_setvolume_80","device":{"identifiers":"hass.agent-ALEKS10","manufacturer":"HASS.Agent Team","model":"Microsoft Windows NT 10.0.19045.0","name":"ALEKS10","sw_version":"2.0.1"},"name":"setvolume_80","state_topic":"homeassistant/button/ALEKS10/setvolume_80/state"}'
> info  2024-03-17 18:34:42: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"homeassistant/button"},"type":"entity_not_found"}'
> error 2024-03-17 18:34:42: Entity 'homeassistant/button' is unknown
> debug 2024-03-17 18:34:42: Received MQTT message on 'homeassistant/button/ALEKS10/setvolume_80/state' with data 'OFF'
> info  2024-03-17 18:34:42: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"homeassistant/button"},"type":"entity_not_found"}'
> error 2024-03-17 18:34:42: Entity 'homeassistant/button' is unknown
> 

and this repeats every 30 seconds (hass agent publishes these topics every 30 seconds)

But these topics have nothing to do with zigbee2mqtt

What exactly does zigbee2mqtt not like and how to fix it?

### How to reproduce it (minimal and precise)

use Home assistant + z2m 1.36.0 + hass agent with command like 'set volume'

### Zigbee2MQTT version

1.36.0

### Adapter firmware version

20211217

### Adapter

Sonoff P

### Setup

HA Addon

francisp2, 2024-03-18:
The homeassistant key is reserved for discovery messages. It seems your  "Hass Agent" configuration uses it for other purposes.

Blacksli, 2024-03-18:
> The homeassistant key is reserved for discovery messages. It seems your "Hass Agent" configuration uses it for other purposes.

That's right. This is a topic for auto-detecting buttons in Home assistant. But the publication of topics in it cannot affect the work of z2m in any way.

tammeryousef1006, 2024-03-25:
any solution ? im facing the same problem and dont know how to prevent it or minimize it

jensweimann, 2024-04-03:
So the issue is that the domain "button" is not supported in z2m yet?

Blacksli, 2024-04-03:
The problem is that z2m complains about topics that z2m has nothing to do with. And this error is constant and very intrusive. Why does z2m look at these topics? I noticed a peculiarity that all topics that contain swear words begin with “set”

Blacksli, 2024-04-03:
[image]

Blacksli, 2024-04-03:
The problem apparently lies in reading and processing messages from the broker. There is some kind of error in the z2m code due to which it is trying to process a message that begins with the word "set"

tammeryousef1006, 2024-04-03:
i got similiar issue kindly see the photo below 
[image]

Blacksli, 2024-04-03:
> i got similiar issue kindly see the photo below ![zigbee](https://private-user-images.githubusercontent.com/62266828/319163006-e5e619d9-f663-44dd-a7b6-8f1363a3c35a.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTIxNTAxNzgsIm5iZiI6MTcxMjE0OTg3OCwicGF0aCI6Ii82MjI2NjgyOC8zMTkxNjMwMDYtZTVlNjE5ZDktZjY2My00NGRkLWE3YjYtOGYxMzYzYTNjMzVhLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MDMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDAzVDEzMTExOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTUwNmUwMzQ4N2MwZmI3MDk1OTBiMmU2Mjg3NjA5YmE0YWIyNTNhMGRhY2Q5ZmUzMWRjYjYxZjY5MGU5OWFlYmEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.aNkniMuUZwQoetBcvzElXSpePjqjJNtKP686MfPNBs8)

Please enable the debug level log, and attach part of the log before the error. Thank you.

closed 2024-11-01 by github-actions[bot]