Koenkk/zigbee2mqtt #21850: Zemismart roller blinds only moves in steps of 10%

rschuiling, 2024-03-17:
### What happened?

I have just installed a [Zemismart roller blind motor](https://www.zigbee2mqtt.io/devices/ZM16EL-03_33.html) through Zigbee2MQTT. When I use the Home Assistant service cover.set_cover_position, the blinds will only move to a 10%-position (so 0, 10, 20, 30 etc). Any other position gets rounded, eg. 12 gets rounded to 10. But because it’s a ‘zebra’ blind, I need more granular control.

I can’t find a configuration setting in Zigbee2MQTT. Google, Github issues and the Home Assistant forum all come up empty.

Is there a way to move the cover in steps of 1% instead of 10%?

### What did you expect to happen?

I expected that the cover would move in steps of 1% instead of 10%.

### How to reproduce it (minimal and precise)

In Home Assistant, use service cover.set_cover_position. Set to e.g. 93%. When the cover reaches its position, see how it reports its position as 90%.

`service: cover.set_cover_position
data:
  position: 93
target:
  entity_id: cover.cover_zebra`

### Zigbee2MQTT version

1.35.3

### Adapter firmware version

20211217

### Adapter

Sonoff Zigbee 3.0 stick 2652P

### Setup

Add-on on HA on Intel NUC

rschuiling, 2024-03-19:
Thanks @julianfs, I changed the title and description to include my specific type of Zemismart roller blinds motor.

rschuiling, 2024-04-05:
I can´t find a solution to his. Also tried to get something from the logs but I can´t find anything relevant myself.

This is from the standard Z2M log, it just shows I try to set it to 38 and it ends up at 40:
```
info 2024-04-05 20:01:53MQTT publish: topic 'zigbee2mqtt/cover_zebra', payload '{"battery":70,"border":"up","click_control":"down","linkquality":138,"motor_fault":false,"position":38,"reverse_direction":"back","state":"STOP"}'
info 2024-04-05 20:01:53MQTT publish: topic 'zigbee2mqtt/cover_zebra', payload '{"battery":70,"border":"up","click_control":"down","linkquality":131,"motor_fault":false,"position":38,"reverse_direction":"back","state":"STOP"}'
info 2024-04-05 20:01:58MQTT publish: topic 'zigbee2mqtt/cover_zebra', payload '{"battery":70,"border":"up","click_control":"down","linkquality":138,"motor_fault":false,"position":40,"reverse_direction":"back","state":"STOP"}'
info 2024-04-05 20:01:58MQTT publish: topic 'zigbee2mqtt/cover_zebra', payload '{"battery":75,"border":"up","click_control":"down","linkquality":138,"motor_fault":false,"position":40,"reverse_direction":"back","state":"STOP"}'
```

I enabled the herdsman debug logging, but that only shows an error about a Zigbee bulb that is unreachable.

rschuiling, 2024-04-29:
I found the Zemismart cover motor in tuya.ts. There seem to be two mentions of the 'position' datapoint:
```
                [2, 'position', tuya.valueConverter.coverPosition],
                [3, 'position', tuya.valueConverter.raw],
```
What is the difference here? And which one would actually get used by Zigbee2mqtt?

Here is the full device info from tuya.ts:
```
    {
        fingerprint: tuya.fingerprint('TS0601', ['_TZE200_68nvbio9', '_TZE200_pw7mji0l', '_TZE200_cf1sl3tj', '_TZE200_nw1r9hp6', '_TZE200_9p5xmj5r',
            '_TZE200_eevqq1uv']),
        model: 'TS0601_cover_3',
        vendor: 'TuYa',
        description: 'Cover motor',
        fromZigbee: [tuya.fz.datapoints],
        toZigbee: [tuya.tz.datapoints],
        onEvent: tuya.onEventSetTime,
        options: [exposes.options.invert_cover()],
        configure: tuya.configureMagicPacket,
        exposes: [
            e.battery(), e.cover_position(),
            e.enum('reverse_direction', ea.STATE_SET, ['forward', 'back']).withDescription('Reverse the motor direction'),
            e.enum('border', ea.STATE_SET, ['up', 'down', 'up_delete', 'down_delete', 'remove_top_bottom']),
            e.enum('click_control', ea.STATE_SET, ['up', 'down']).withDescription('Single motor steps'),
            e.binary('motor_fault', ea.STATE, true, false),
        ],
        whiteLabel: [
            tuya.whitelabel('Zemismart', 'ZM16EL-03/33', 'Cover motor', ['_TZE200_68nvbio9']),
            tuya.whitelabel('Zemismart', 'ZM25EL', 'Cover motor', ['_TZE200_pw7mji0l']),
            tuya.whitelabel('Zemismart', 'ZM85EL-2Z', 'Roman Rod I type U curtains track', ['_TZE200_cf1sl3tj', '_TZE200_nw1r9hp6']),
            tuya.whitelabel('Hiladuo', 'B09M3R35GC', 'Motorized roller shade', ['_TZE200_9p5xmj5r']),
        ],
        meta: {
            // All datapoints go in here
            tuyaDatapoints: [
                [1, 'state', tuya.valueConverterBasic.lookup({'OPEN': tuya.enum(0), 'STOP': tuya.enum(1), 'CLOSE': tuya.enum(2)})],
                [2, 'position', tuya.valueConverter.coverPosition],
                [3, 'position', tuya.valueConverter.raw],
                [5, 'reverse_direction', tuya.valueConverterBasic.lookup({'forward': tuya.enum(0), 'back': tuya.enum(1)})],
                [12, 'motor_fault', tuya.valueConverter.trueFalse1],
                [13, 'battery', tuya.valueConverter.raw],
                [16, 'border', tuya.valueConverterBasic.lookup({
                    'up': tuya.enum(0), 'down': tuya.enum(1), 'up_delete': tuya.enum(2), 'down_delete': tuya.enum(3),
                    'remove_top_bottom': tuya.enum(4)})],
                [20, 'click_control', tuya.valueConverterBasic.lookup({'up': tuya.enum(0), 'down': tuya.enum(1)})],
            ],
        },
    }
```

rschuiling, 2024-11-04:
Well the issue is still present. Would be great if this issue could be kept alive.

rschuiling, 2025-01-05:
Still present...

rschuiling, 2025-03-11:
Still present

louim, 2025-04-18:
I have the same problem. I have 4 ZM16EL-03/33. 1 of them work as expected, and can be set to any position. The other 3 will only move by increment of 10 percent.

It seems to be something related to how the blind report their position, since when I use the remote control, the blind will go exactly where I want. If anyone have an idea how to investigate, I'd be happy to compare and test different commands.

rschuiling, 2025-04-18:
When you put it like that, it sounds like a firmware issue with the Zemismart. I sent an email to Zemismart after sales support to ask if a more recent firmware is available.

xjimy, 2025-05-18:
@rschuiling Have you received any response, and did you get a different firmware version? I also asked about a newer firmware due to another issue, and how to update the device.

rschuiling, 2025-05-18:
@xjimy Unfortunately there is no different firmware available, says Zemismart. A dead end it seems.

louim, 2025-07-26:
This is still happening. 🥲

closed 2025-10-04 by github-actions[bot]

rschuiling, 2025-10-19:
For anyone finding this: I just posted a [workaround](https://community.home-assistant.io/t/cover-only-moves-in-steps-of-10/703518/11?u=emphyrio) on the Home Assistant forum.