Koenkk/zigbee2mqtt #21854: Bticino K4027 Cover

gpapi319, 2024-03-18:
### What happened?

Cannot update Zigbe2mqtt after Version: 1.33.1-1.
If I update to newer versions, Btcino Covers are inverted and there is no way to make them work correctly (invert cover = true is not working any more).
Bticino Firmware is 1851907, i cannot update it from Zigbee2Mqtt and don't have Bticino Gateway.

### What did you expect to happen?

I expected that covers worked like previous versions (i.e. Enabling invert cover = true), with newer versions are inverted.

### How to reproduce it (minimal and precise)

Just update to newer versions and then covers are inverted

### Zigbee2MQTT version

All versions after  1.33.1-1

### Adapter firmware version

Don't know but i bought the adapter 3 years ago and never update since thend firmware

### Adapter

Slaesh's CC 2625RB

### Setup

Home assistant OS on Intel Nuc

### Debug log

1.36.0-1 (Invert Cover = true is not working, cover movement is not correct)

{"dateCode":" \u000020210407\u0000\u0000\u0000\u0000\u0000","description":"Shutter SW with level control","friendly_name":"Living Room Cover","hardwareVersion":9,"ieeeAddr":"0x0004740000b539f3","lastSeen":1710072375231,"manufacturerID":4129,"manufacturerName":" Legrand\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","model":"K4027C/L4027C/N4027C/NT4027C","modelID":" Shutter SW with level control\u0000"

Zigbee2MQTT:info  2024-03-10 13:08:15: Configuring 'Living Room Cover'
Zigbee2MQTT:debug 2024-03-10 13:08:15: Received Zigbee message from 'Living Room Cover', type 'attributeReport', cluster 'closuresWindowCovering', data '{"currentPositionLiftPercentage":100}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2024-03-10 13:08:15: MQTT publish: topic 'zigbee2mqtt/Living Room Cover', payload '{"action":null,"calibration_mode":"specific_nllv","last_seen":"2024-03-10T13:08:15+01:00","led_if_on":null,"led_in_dark":null,"linkquality":60,"position":0,"state":"CLOSE","tilt":100,"update":{"installed_version":1851907,"latest_version":2245123,"state":"available"},"update_available":true}'

Zigbee2MQTT:info  2024-03-10 13:08:16: MQTT publish: topic 'zigbee2mqtt/Living Room Cover', payload '{"action":null,"calibration_mode":"specific_nllv","last_seen":"2024-03-10T13:08:16+01:00","led_if_on":null,"led_in_dark":null,"linkquality":57,"position":0,"state":"CLOSE","tilt":100,"update":{"installed_version":1851907,"latest_version":2245123,"state":"available"},"update_available":true}'
Zigbee2MQTT:error 2024-03-10 13:08:16: Failed to configure 'Living Room Cover', attempt 1 (Error: ConfigureReporting 0x0004740000b539f3/1 closuresWindowCovering([{"attribute":"currentPositionTiltPercentage","minimumReportInterval":1,"maximumReportInterval":120,"reportableChange":1}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4129,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
    at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:319:28)
    at Endpoint.configureReporting (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:672:22)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/devices/legrand.ts:216:13)
    at Configure.configure (/app/lib/extension/configure.ts:122:13)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17))

Version: 1.33.1-1 (Invert Cover = true is working)

Zigbee2MQTT:info  2024-03-10 13:12:59: Living Room Cover (0x0004740000b539f3): K4027C/L4027C/N4027C/NT4027C - BTicino Shutter SW with level control (Router)

{"dateCode":" \u000020210407\u0000\u0000\u0000\u0000\u0000","description":"Shutter SW with level control","friendly_name":"Living Room Cover","hardwareVersion":9,"ieeeAddr":"0x0004740000b539f3","lastSeen":1710072911482,"manufacturerID":4129,"manufacturerName":" Legrand\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","model":"K4027C/L4027C/N4027C/NT4027C","modelID":" Shutter SW with level control\u0000","networkAddress":32727,"powerSource":"Mains (single phase)","softwareBuildID":"001c","type":"Router","vendor":"BTicino"}

gpapi319, 2024-04-05:
Hello, i'm stuck with version 1.33.1-1...I cannot upgrade because my cover would be reverted...Any hint? Thank you

FabianMangold, 2024-04-20:
Hi @gpapi319,

Have you tried upgrading the firmware using the latest Z2M DEV version ?
I have no BTicino devices, but similar Legrand devices, that work just fine (Cover inversion works) with newer Firmware.

gpapi319, 2024-04-20:
Hi, actually I found yesterday that all the buttons are phisically
inverted, so I was thinking about inverting them...

If i upgrade the firmware should I pair again all the devices?

Thank for the answer!

Thank you

FabianMangold, 2024-04-20:
Hi,
There is no need to repair the devices after the update, you may want to give this a try.
Good luck

gpapi319, 2024-04-20:
Thank you very much!
I will try asap and let you know!

Gianluca

FabianMangold, 2024-07-17:
Hi @gpapi319,

Any news on that, has it worked ?
Cheers,
Fabian

gpapi319, 2024-07-21:
Sorry for answering you so late.
I still have not tried because i am very busy at work and afraid to mess
all my zigbee devices up...

I'll study the firmware upgrade process and try as soon as i have some free
time to spend. Hope summer holidays will help!

I'll inform you after the upgrade.

Thank you

Gianluca

Rendeik, 2025-05-10:
Hi, I have a bticino K4027 device that was seen as bticino and now seen as legrang 067776A, when I invert it something works but after all the movement it always sees it closed. Even when open I press up it goes to 50% and then displays closed again.