Koenkk/zigbee2mqtt #21855: ERS-10TZBVK-AA / TS004F action does not work anymore in dev version

Minims, 2024-03-18:
### What happened?

ERS-10TZBVK-AA / TS004F action does not work anymore in dev version

Zigbee2MQTT version = 1.36.0-dev commit: [1a9e05e](https://github.com/Koenkk/zigbee2mqtt/commit/1a9e05e)

```
Info 2024-03-18 12:14:47MQTT publish: topic 'zigbee2mqtt/Bouton Bureau', payload '{"action":null,"action_rate":null,"action_step_size":null,"action_transition_time":null,"battery":100,"brightness":182,"last_seen":"2024-03-18T12:14:47+01:00","linkquality":87,"operation_mode":"event","voltage":3000}'
```

Zigbee2MQTT version = [1.36.0](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.36.0)
```
Info 2024-03-18 12:16:48MQTT publish: topic 'zigbee2mqtt/Bouton Bureau', payload '{"action":"single","action_rate":null,"action_step_size":null,"action_transition_time":null,"battery":100,"brightness":182,"last_seen":"2024-03-18T12:16:48+01:00","linkquality":109,"operation_mode":"event","voltage":3000}'
Info 2024-03-18 12:16:48MQTT publish: topic 'zigbee2mqtt/Bouton Bureau', payload '{"action":"","action_rate":null,"action_step_size":null,"action_transition_time":null,"battery":100,"brightness":182,"last_seen":"2024-03-18T12:16:48+01:00","linkquality":109,"operation_mode":"event","voltage":3000}'
Info 2024-03-18 12:16:48MQTT publish: topic 'zigbee2mqtt/Bouton Bureau/action', payload 'single'
```

### What did you expect to happen?

Stable behavior expected:

```
Info 2024-03-18 12:16:48MQTT publish: topic 'zigbee2mqtt/Bouton Bureau', payload '{"action":"single","action_rate":null,"action_step_size":null,"action_transition_time":null,"battery":100,"brightness":182,"last_seen":"2024-03-18T12:16:48+01:00","linkquality":109,"operation_mode":"event","voltage":3000}'
Info 2024-03-18 12:16:48MQTT publish: topic 'zigbee2mqtt/Bouton Bureau', payload '{"action":"","action_rate":null,"action_step_size":null,"action_transition_time":null,"battery":100,"brightness":182,"last_seen":"2024-03-18T12:16:48+01:00","linkquality":109,"operation_mode":"event","voltage":3000}'
Info 2024-03-18 12:16:48MQTT publish: topic 'zigbee2mqtt/Bouton Bureau/action', payload 'single'
```

### How to reproduce it (minimal and precise)

Install latest dev version and click on device button.

### Zigbee2MQTT version

1.36.0-dev commit: [1a9e05e](https://github.com/Koenkk/zigbee2mqtt/commit/1a9e05e)

### Adapter firmware version

20230507

### Adapter

Sonoff Dongle Plus 

### Setup

Z2M PI5 HA Addon

Koenkk, 2024-03-19:
Could you provide the debug log of this?

See [this](https://www.zigbee2mqtt.io/guide/usage/debug.html) on how to enable debug logging.

Minims, 2024-03-19:
Non Working Dev version:
```
Debug 2024-03-19 20:59:16Received Zigbee message from 'Bouton Bureau', type 'commandTuyaAction', cluster 'genOnOff', data '{"value":0}' from endpoint 1 with groupID 0
Info 2024-03-19 20:59:16MQTT publish: topic 'zigbee2mqtt/Bouton Bureau', payload '{"action":null,"action_rate":null,"action_step_size":null,"action_transition_time":null,"battery":100,"brightness":182,"last_seen":"2024-03-19T20:59:16+01:00","linkquality":76,"operation_mode":"event","voltage":3000}'
```

Working Stable version :
```
Debug 2024-03-19 20:56:50Received Zigbee message from 'Bouton Bureau', type 'raw', cluster 'genOnOff', data '{"data":[1,52,253,0],"type":"Buffer"}' from endpoint 1 with groupID 0
Info 2024-03-19 20:56:50MQTT publish: topic 'zigbee2mqtt/Bouton Bureau', payload '{"action":"single","action_rate":null,"action_step_size":null,"action_transition_time":null,"battery":100,"brightness":182,"last_seen":"2024-03-19T20:56:50+01:00","linkquality":76,"operation_mode":"event","voltage":3000}'
Info 2024-03-19 20:56:50MQTT publish: topic 'zigbee2mqtt/Bouton Bureau', payload '{"action":"","action_rate":null,"action_step_size":null,"action_transition_time":null,"battery":100,"brightness":182,"last_seen":"2024-03-19T20:56:50+01:00","linkquality":76,"operation_mode":"event","voltage":3000}'
Info 2024-03-19 20:56:50MQTT publish: topic 'zigbee2mqtt/Bouton Bureau/action', payload 'single'
```

Koenkk, 2024-03-20:
Pushed a fix, can you check if it works?

See [this](https://www.zigbee2mqtt.io/guide/usage/debug.html) on how to enable debug logging.

Minims, 2024-03-20:
It works ! Many thanks 🙏

closed 2024-03-20 by Minims