Koenkk/zigbee2mqtt #21875: EAYCBM-Z-2P Breaker Identified as wrong device

andreas-glaser, 2024-03-20:
### What happened?

I've paired my new breaker which I bought on aliexpress. https://www.aliexpress.com/i/1005004894626384.html
After pairing I'm getting a success message that the device is supported but then a completely different device is shown.
Is says it's a Smart Plug TS011F / _TZ3000_zrm3oxsh

Most datapoints are not exposed.

### What did you expect to happen?

I expected the right identification.

### How to reproduce it (minimal and precise)

Power on breaker. Pair.

### Zigbee2MQTT version

 1.36.0

### Adapter firmware version

20230507

### Adapter

Sonoff Dongle P

### Setup

docker

chrismarine74, 2024-03-25:
The same problem,

levonti, 2024-04-06:
i have same problem

pagaale, 2025-01-07:
Same problem

pagaale, 2025-03-09:
I still have the problem

Zoldex, 2025-03-30:
It wrongly detects as EAKCB-T-M-Z.
Using Z2M v 2.1.3

closed 2025-06-07 by github-actions[bot]