Koenkk/zigbee2mqtt #21876: [New device support]:

serot23, 2024-03-20:
### Link

https://es.aliexpress.com/item/1005006224025259.html?gatewayAdapt=glo2esp

### Database entry

{"id":34,"type":"EndDevice","ieeeAddr":"0xa49e69fffe48111b","nwkAddr":62597,"manufId":4098,"manufName":"_TZE200_cpbo62rn","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"�x�-\u0011�x�-g","65506":31,"65508":0,"modelId":"TS0601","manufacturerName":"_TZE200_cpbo62rn","stackVersion":0,"dateCode":"","appVersion":65}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1710924578955}

### Comments

I tried several times to delete and reconnect and it connects, it works. But is detected as the wrong device and is missing functions. Like manual hand pull/start for the curtains and lux meter is missing. I can't find external converter handles for this device in the list of herdmans converters. Sorry if i am filling this form wrong in, but i am a noob in this and don't always understand how github works.

### External definition

```shell
const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE200_cpbo62rn',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;
```

etiennedub, 2024-03-25:
I have the same device and I was able to make it work even if it was identify as the wrong device. On the device it's label as "LY-1668".

Someone in this [thread](https://github.com/Koenkk/zigbee2mqtt/issues/17436#issuecomment-1535869929) shared some datapoints to expose the luminosity sensor and the temperature sensor (I'm not sure for the hand mode). The main issue is that both devices are identified as `_TZE200_cpbo62rn` so I don't know how to add those changes without impacting the "LY-108" device. But I tested the datapoints 103 and 104 and they correctly report the luminosity and temperature.

Also, in my case, the calibration was inverted. I need to press "STOP" to start the calibration and ended it with "START". Same for the "work_state", "SUCCESS" and "LEARNING" are flipped.

serot23, 2024-05-02:
how do we get this to the developer, im not so good with github

serot23, 2024-07-08:
> I have the same device and I was able to make it work even if it was identify as the wrong device. On the device it's label as "LY-1668".
>
> Someone in this [...]

Yeah i have the same issue with the calibration. It is a pain in the as s, how did you get calibration to work. Becauese my Home assistant wasn't working probably so i needed to reset some stuff and the calibration is off now on the curtains. It is reversed like you said, but how do you start and save the calibration points? Do you start with open of closed curtains and which steps do you do after thst. Is it "stopt" than let it open than "start" and than wait to the open position and press "stop" again when it has get to the most open position? I can't get it to work correctly now.

serot23, 2024-07-09:
Ok i fixed it. What i did was i placed the motors so that the curtains blocked the window, than i pressed reset in z2m, than i press stop, than they opened the curtain, so when they where on the most open position i pressed start and than it stopped and was calibrated

Ru4rc, 2024-07-12:
I also just received the same LY-1668 device which reports as a LY-108 curtain motor. If there’s any help or info needed in creating a correct converter I‘d be happy to provide these informations. I’m not good with coding myself, but I would love to see the device working correctly :D

Koenkk, 2024-07-22:
Does this device work when stopping z2m, changing all occurrences of `_TZE200_cpbo62rn` to `_TZE200_bjzrowv2` in the `data/database.db` and starting z2m?

serot23, 2024-07-22:
> Does this device work when stopping z2m, changing all occurrences of `_TZE200_cpbo62rn` to `_TZE200_bjzrowv2` in the `data/database.db` and starting z2m?

İ will try in maybe an hour or 2 i will let you know

serot23, 2024-07-22:
> Does this device work when stopping z2m, changing all occurrences of `_TZE200_cpbo62rn` to `_TZE200_bjzrowv2` in the `data/database.db` and starting z2m?

Ok they do respond but the whole calibration is gone. And there is no calibration button now in z2m, see image.
[image]

Koenkk, 2024-07-23:
And when changing to `_TZE200_cpbo62rn`?

serot23, 2024-07-23:
> And when changing to `_TZE200_cpbo62rn`?

Than it works again like it worked before the change. The calibration is still there than

Koenkk, 2024-07-24:
Doesn't it allow calibration now via `set_upper_limit`?

serot23, 2024-07-24:
> Doesn't it allow calibration now via `set_upper_limit`?

İ don't see the option for set upper limit eith both codes. But when i changed to _TZE200_bjzrowv2 it was not only the calibration which was of, also one went the wrong way. So when i pressed open, one opened the curtain and the other closed the curtain. So i think _TZE200_cpbo62rn works better, but still isn't how it should be i guess

Koenkk, 2024-07-26:
@serot23 when using `_TZE200_cpbo62rn`, it should be detected as `TS0601_cover_6`, under the exposes tab you should find `set_upper_limit` which allows to calibrate, does this work?

serot23, 2024-07-26:
> @serot23 when using `_TZE200_cpbo62rn`, it should be detected as `TS0601_cover_6`, under the exposes tab you should find `set_upper_limit` which allows to calibrate, does this work?

Sorey my mistake, i was looking wrong last time. İ had upper limit from the beginning already. İt never changed, only when i switched to _TZE200_bjzrowv2 it disappears. But the device is still shown wrongly and missing the lux meter, pull and start and the calibration options work weird, it start when you press stop and not the other way around

Ru4rc, 2024-07-27:
Lemme just add my findings, I tinkered around a bit with external converters and for me, this has been working fine so far, except for a few little caveats. I got it to calibrate correctly and even the illuminance and temperature readings are working.
I just don't know about the pull-to-start feature, or if it's even supported by the device. The work state doesn't show if it's actually working, so that would probably also have to be adjusted.
Also the buttons for calibrating, setting the limits etc. could probably be labeled a bit better to understand in what order you have to press them.
Plus the device still shows as the LY-108 curtain motor if you're not using this external converter, so the fingerprinting is still an issue.
```javascript
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    // Since a lot of Tuya devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_cpbo62rn'}],
    model: 'LY-1668',
    vendor: 'Tuya',
    description: '3in1 Curtain Robot',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    configure: tuya.configureMagicPacket,
    exposes: [
        e.text('work_state', ea.STATE),
        e.cover_position().setAccess('position', ea.STATE_SET),
        e.battery(),
        e.enum('motor_direction', ea.STATE_SET, ['left', 'right']).withDescription('Motor side'),
        e.enum('set_upper_limit', ea.STATE_SET, ['start', 'stop']).withDescription('Learning'),
        e.enum('factory_reset', ea.STATE_SET, ['SET']).withDescription('Remove limits'),
	e.temperature(),
	e.illuminance(),
    ],
    whiteLabel: [tuya.whitelabel('Tuya', 'LY-1668', '3in1 Curtain Robot', ['_TZE200_cpbo62rn'])],
    meta: {
        // All datapoints go in here
        tuyaDatapoints: [
	    [1, 'state', tuya.valueConverterBasic.lookup({CLOSE: tuya.enum(2), STOP: tuya.enum(1), OPEN: tuya.enum(0)})],
            [2, 'position', tuya.valueConverter.coverPositionInverted],
            [3, 'position', tuya.valueConverter.coverPositionInverted],
            [7, 'work_state', tuya.valueConverterBasic.lookup({standby: tuya.enum(0), success: tuya.enum(1), learning: tuya.enum(2)})],
            [13, 'battery', tuya.valueConverter.raw],
            [101, 'motor_direction', tuya.valueConverterBasic.lookup({left: tuya.enum(0), right: tuya.enum(1)})],
            [102, 'set_upper_limit', tuya.valueConverterBasic.lookup({start: tuya.enum(1), stop: tuya.enum(0)})],
	    [103, 'temperature', tuya.valueConverter.raw],
	    [104, 'illuminance', tuya.valueConverter.raw],
            [107, 'factory_reset', tuya.valueConverter.setLimit],
        ],
    },
};

module.exports = definition;
```

serot23, 2024-07-28:
> Lemme just add my findings, I tinkered around a bit with external converters and for me, this has been working fine so far, except for a few little caveats. I got it to calibrate correctly and even the illuminance and temperature readings are working. I just don't know about the pull-to-start feature, or if it's even supported by the device. The work state doesn't show if it's actually working, so that would probably also have to be adjusted. Also the buttons for calibrating, setting the limits etc. could probably be labeled a bit better to understand in what order you have to press them. Plus the device still shows as the LY-108 curtain motor if you're not using this external converter, so the fingerprinting is still an issue.
> 
> ```js
> const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
> const tz = require('zigbee-herdsman-converters/converters/toZigbee');
> const exposes = require('zigbee-herdsman-converters/lib/exposes');
> const reporting = require('zigbee-herdsman-converters/lib/reporting');
> const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
> const e = exposes.presets;
> const ea = exposes.access;
> const tuya = require('zigbee-herdsman-converters/lib/tuya');
> 
> const definition = {
>     // Since a lot of Tuya devices use the same modelID, but use different datapoints
>     // it's necessary to provide a fingerprint instead of a zigbeeModel
>     fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_cpbo62rn'}],
>     model: 'LY-1668',
>     vendor: 'Tuya',
>     description: '3in1 Curtain Robot',
>     fromZigbee: [tuya.fz.datapoints],
>     toZigbee: [tuya.tz.datapoints],
>     configure: tuya.configureMagicPacket,
>     exposes: [
>         e.text('work_state', ea.STATE),
>         e.cover_position().setAccess('position', ea.STATE_SET),
>         e.battery(),
>         e.enum('motor_direction', ea.STATE_SET, ['left', 'right']).withDescription('Motor side'),
>         e.enum('set_upper_limit', ea.STATE_SET, ['start', 'stop']).withDescription('Learning'),
>         e.enum('factory_reset', ea.STATE_SET, ['SET']).withDescription('Remove limits'),
> 	e.temperature(),
> 	e.illuminance(),
>     ],
>     whiteLabel: [tuya.whitelabel('Tuya', 'LY-1668', '3in1 Curtain Robot', ['_TZE200_cpbo62rn'])],
>     meta: {
>         // All datapoints go in here
>         tuyaDatapoints: [
> 	    [1, 'state', tuya.valueConverterBasic.lookup({CLOSE: tuya.enum(2), STOP: tuya.enum(1), OPEN: tuya.enum(0)})],
>             [2, 'position', tuya.valueConverter.coverPositionInverted],
>             [3, 'position', tuya.valueConverter.coverPositionInverted],
>             [7, 'work_state', tuya.valueConverterBasic.lookup({standby: tuya.enum(0), success: tuya.enum(1), learning: tuya.enum(2)})],
>             [13, 'battery', tuya.valueConverter.raw],
>             [101, 'motor_direction', tuya.valueConverterBasic.lookup({left: tuya.enum(0), right: tuya.enum(1)})],
>             [102, 'set_upper_limit', tuya.valueConverterBasic.lookup({start: tuya.enum(1), stop: tuya.enum(0)})],
> 	    [103, 'temperature', tuya.valueConverter.raw],
> 	    [104, 'illuminance', tuya.valueConverter.raw],
>             [107, 'factory_reset', tuya.valueConverter.setLimit],
>         ],
>     },
> };
> 
> module.exports = definition;
> ```

Nice thanks for adding your findings. I have got the pull to start working out of no where. But it doesn't work all the time and i need to pull really hard, to the point my curtains will shred. I didn't change anything to the coding also i didn't add any converters, i need to charge them for the first time. So i climbed up to the motors to plug in the charging cable and i pulled the motor a bit accidentally, and then to curtains automatically opened. Than i tried again it didn't work, so then i tried to put the cable in again so i again accidentally pulled the motor and it again reacted to it. So it works but it need a lot of force i think. From the top pulling it sometimes works, but when i pull from the curtain at the bottom, it doesn't work at all

gosoares, 2024-08-16:
> Lemme just add my findings, I tinkered around a bit with external converters and for me, this has been working fine so far, except for a few little caveats. I got it to calibrate correctly and even [...]

Thanks. I have the same model and this converter worked for me. Just the working state and the learning values was still flipped, but it was easily fixed by swapping the enum values.

closed 2025-01-07 by github-actions[bot]