Koenkk/zigbee2mqtt #21877: Trying to add another IKEA E1603/E1702/E1708 messes with already configured one

ogghi, 2024-03-20:
### What happened?

I wanted to add another IKEA E1603/E1702/E1708 smart socket. 
Just by pressing "permit join" and then factory resetting the new device makes my existing one "reconfigure"?
The new one does not appear at all, the existing one now has status: "Interview failed"

The existing device still works fine so far. Even after HA restart same behavior.

### What did you expect to happen?

New device to get recognized.

### Zigbee2MQTT version

1.36.0

### Adapter firmware version

20210708

### Adapter

zStack3x0

### Setup

Proxmox HA VM, USB passthrough to VM, Z2M add-on

mrmaximas, 2024-03-20:
have you tried a newest firmware for adapter?

ogghi, 2024-03-20:
Adding other devices in the past worked just fine. I am wondering, are those Ikea devices faulty and have the same ID or so?

How can I update the USB adapter firmware? Never though about that one actually!

chris-1243, 2024-03-20:
@ogghi Have a look there

https://www.zigbee2mqtt.io/guide/adapters/#notes

mrmaximas, 2024-03-21:
@ogghi best working fw for CC2652P here https://github.com/Koenkk/Z-Stack-firmware/discussions/496
P7 is coming later

closed 2024-10-18 by github-actions[bot]