Koenkk/zigbee2mqtt #21879: HA Service "light off" toggles one light group on without changing the state in HA

0la1, 2024-03-20:
### What happened?

I have a script that turns off all lights when I am going to bed. I have lately added three Ikea GU10 lights (three lights, grouped to one as this is one light with three bulbs), which always get turned on when the script fires. Strange this is, that HA thinks the lights are still off, but the lights are actually lid. So I think the is issue that they somehow get a misleading signal in the background that's why I am posting this here.

PS.: I don't know here to look for the adapter version so the number if obviously not right.

### What did you expect to happen?

Turn the lights off, or leave them off in case they are already off

### How to reproduce it (minimal and precise)

Lights are off
Fire the script
lights turn on (very dimmed, maybe 10%)

### Zigbee2MQTT version

1.36.0

### Adapter firmware version

12345

### Adapter

HA Skyconnect

### Setup

Add-on for HA OS running on IntelNuc

jkufner, 2024-06-03:
After firing the lights off service:

````
service: light.turn_off
target:
  entity_id:
    - light.ikea_light_bulb_1
data: {}
`````

The light bulb toggles between off and 0% on (which glows at the lowest level it can do).

If I assign the same light bulb to a scene in turned off state, it works as expected.

kineticscreen, 2024-06-14:
I might have found something similar here - it happens if I 'dim' down the lights down all the way to off. If they then receive a proper "light off" command, they switch on, but still appear as off to Home Assistant.

0la1, 2024-06-18:
I have recently switched to a Poe based zigbee controller and this issue happens only twice in the last two month. So it definitely got better. 

However, I am still not aware of the root cause. All transition times have been removed from the Skript which helped as well.

closed 2025-01-07 by github-actions[bot]