Koenkk/zigbee2mqtt #21886: [New device support]: OXT (tuya) T363s 3 Phase power meter

BartRuSec, 2024-03-20:
### Link

https://houseiq.pl/pl/p/OXT-licznik-3-fazy-2-kierunkowy-ZigBee-TUYA/1859

### Database entry

{"id":10,"type":"Router","ieeeAddr":"0x943469fffec65a4a","nwkAddr":43048,"manufId":4098,"manufName":"_TZE200_bkkmqmyo","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"���-i���-i","65506":31,"65508":0,"65534":0,"modelId":"TS0601","manufacturerName":"_TZE200_bkkmqmyo","stackVersion":0,"dateCode":"","zclVersion":3,"appVersion":65,"powerSource":1}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-708457359},"lastSeen":1710942569728}

### Comments

It is recognized as  [Hiking DDS238-2 ](https://www.zigbee2mqtt.io/devices/DDS238-2.html#hiking-dds238-2). Following external definition works fine. It would be nice to extend [homeassistant.ts](https://github.com/Koenkk/zigbee2mqtt/blob/master/lib/extension/) with proper device classes:
``` yaml
      power_a:
        device_class: power
      power_b:
        device_class: power
      power_c:
        device_class: power
      current_a:
        device_class: current
      current_b:
        device_class: current
      current_c:
        device_class: current
      voltage_a:
        device_class: voltage
      voltage_b:
        device_class: voltage
      voltage_c:
        device_class: voltage
      power_reactive:
        device_class: reactive_power
      power_reactive_a:
        device_class: reactive_power
      power_reactive_b:
        device_class: reactive_power
      power_reactive_c:
        device_class: reactive_power
      power_factor:
        device_class: power_factor
      power_factor_a:
        device_class: power_factor
      power_factor_b:
        device_class: power_factor
      power_factor_c:
        device_class: power_factor
```

### External definition

```shell
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const { Buffer } = require('node:buffer');

const powerReactiveWithPhase = (phase) => e.numeric(`power_reactive_${phase}`, ea.STATE).withUnit('VAR').withDescription(`Instantaneous measured reactive power (phase ${phase.toUpperCase()})`);

const phaseVariant3WithPhase = (phase) => {
    return {
        from: (v) => {
            const buf = Buffer.from(v, 'base64');
            return {
                [`power_${phase}`]: (buf[3] | buf[2] << 8),
                [`power_reactive_${phase}`]: (buf[7] | buf[6] << 8),
                [`power_factor_${phase}`]: (buf[9] | buf[8] << 8) / 10,
                [`current_${phase}`]: (buf[12] | buf[11] << 8) / 1000,
                [`voltage_${phase}`]: (buf[14] | buf[13] << 8) / 10
            };
        },
    };
}

const definition =   {
    fingerprint: tuya.fingerprint('TS0601',['_TZE200_bkkmqmyo']),
    model: 'T363S',
    vendor: 'OXT',
    description: 'Smart energy monitor for 3P+N system',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    configure: tuya.configureMagicPacket,
    exposes: [tuya.exposes.switch(),tuya.exposes.voltageWithPhase('a'), tuya.exposes.voltageWithPhase('b'), tuya.exposes.voltageWithPhase('c'),
        powerReactiveWithPhase("a"),powerReactiveWithPhase("b"),powerReactiveWithPhase("c"),
        tuya.exposes.powerFactorWithPhase("a"),tuya.exposes.powerFactorWithPhase("b"),tuya.exposes.powerFactorWithPhase("c"),
        tuya.exposes.powerWithPhase('a'), tuya.exposes.powerWithPhase('b'), tuya.exposes.powerWithPhase('c'),
        tuya.exposes.currentWithPhase('a'), tuya.exposes.currentWithPhase('b'), tuya.exposes.currentWithPhase('c'),
        e.energy().withDescription('Total forward active energy'),
        e.produced_energy().withDescription('Total reverse active energy'), e.power_reactive(),
        e.power_factor().withUnit('%'), e.power(),
        e.ac_frequency(),
    ],
    meta: {
        tuyaDatapoints: [
            [1, 'energy', tuya.valueConverter.divideBy100],
            [6, null, phaseVariant3WithPhase('a')],
            [7, null, phaseVariant3WithPhase('b')],
            [8, null, phaseVariant3WithPhase('c')], // voltage and current and power
            [16, 'state', tuya.valueConverter.onOff],
            [101, 'total_energy', tuya.valueConverter.divideBy100], // total energy produced + consumed
            [102, 'produced_energy', tuya.valueConverter.divideBy100],
            [103, 'power', tuya.valueConverter.raw],
            [105, 'ac_frequency', tuya.valueConverter.divideBy100],
            [109, 'energy_reactive', tuya.valueConverter.divideBy100], // reactive energy in VArh
            [110, 'power_reactive', tuya.valueConverter.raw], // reactive power
            [111, 'power_factor', tuya.valueConverter.divideBy10],
            // Ignored for now; we don't know what the values mean
            [9, null, null],  // I don't know.
            [17, null, null], // I don't know.
            [18, null, null], // I don't know.
            [107,null,null],  // I don't know.
            [106,null,null]   // I don't know.
        ],
    }
}

module.exports = definition;
```

mrmaximas, 2024-03-20:
@BartRuSec  it may help to find the unknown values https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html

BartRuSec, 2024-03-21:
Thank @mrmaximas  for the tip. Before I created my own external converter, I browsed through this and other documentation and looked through the source code for inspiration. I don't have a Tuya gateway and I don't use Tuya Cloud, so I mainly checked what was displayed on the device vs the values ​​from EP and I managed to achieve what was needed and valuable for me (I don't need more information).

mrmaximas, 2024-03-21:
@BartRuSec I bought a wired Tuya Gateway for $20 for experiments, including firmware interception, I think it's a small price to pay for the development of this project :)

closed 2024-10-18 by github-actions[bot]

BartRuSec, 2024-10-24:
@DariuszLuber I don't think so, but the above definition (+ additional device class configuration) has been working successfully for me over the past few months as an external converter in Z2M.

unCleanCode, 2025-07-01:
@BartRuSec huge thanks! Thanks to your external definition I can finally properly detect this device and see a lot more of the values from it (i.e. separate voltage per line and separate energy)

I'm surprised why this isn't pushed to core system yet

Quick question: any chance tuya is giving any interface for configuring min/max voltage and other stuff (this device does support those via Tuya app)

BartRuSec, 2026-04-15:
Hi @unCleanCode, many thanks for the more complete version of this converter! Just wanted to let you know that I was trying to use on recent Z2M it but it seems to be invalid.