Koenkk/zigbee2mqtt #21894: Exception while calling fromZigbee converter: Cannot read properties of undefined (reading 'displayType')}

grraab2000, 2024-03-21:
### What happened?

Just getting started with ZigBee (house is mostly Z-Wave, but ....).
So, Installing Inovelli's, VZM36/VZM31-SN under ZHA, but many issues so moving from ZHA to Z2M since I have less than 6 devices. Overall nice. HOWEVER, VZM36 started decently under it and then suddenly changed behavior. Most of the entities just stopped working with this Error:
Exception while calling fromZigbee converter: Cannot read properties of undefined (reading 'displayType')}

This is only on the VZM36 Canopy.  On exposes, with individual refresh:
Light state and brightness work, 
Followed by Fan state. 
Sometimes, mode will work, but generally not. 
Breeze does not have a refresh. 
And all of the rest throws exception. 

The 31-SN works perfectly. 

### How to reproduce it (minimal and precise)

The wiring goes through the 31-SN switch, but is set on smart bulb mode (want power and ability to air-drop connection)

Install and pair a Inovelli's VZM36.

### Zigbee2MQTT version

    1.36.0 commit: unknown

### Adapter firmware version

6.10.3.0 build 297

### Adapter

EZSP v8 from  HASS yellow with POE. 

### Setup

installed to HASS yellow.

Koenkk, 2024-03-22:
@InovelliUSA could you check this?

InovelliUSA, 2024-03-22:
This is fixed in the dev branch:

https://github.com/Koenkk/zigbee-herdsman-converters/pull/7164

@grraab2000 please upgrade to the edge version until the fix gets into production.

Please refer to yellow warning in our help document: https://help.inovelli.com/en/articles/9048636-blue-series-fan-light-module-setup-instructions-zigbee2mqtt

closed 2024-03-23 by Koenkk