Koenkk/zigbee2mqtt #21903: RTX ZDM2LN dimmer detected as TS110E_2gang_2 - incorrect reads of brightnes when set from switch

Mnietek, 2024-03-22:
### What happened?

When brightnes is changed from HA everything is OK - published messages contains correct value of brightness.
When brightnes is changed by switch published messages contains last value set by HA. This value is not updated when using switch.  Also slider in HA user interface is not updated when using switch 

### What did you expect to happen?

When brightnes is changed by using switch there should be correct value in published messages and slider in HA should be updated. 

### How to reproduce it (minimal and precise)

Just change the brightness using the switch - the values ​​in published messages do not change.

### Zigbee2MQTT version

1.36.0 

### Adapter firmware version

20220219

### Adapter

Sonoff Zigbee 3.0 USB Dongle Plus

### Setup

Add-on on HA OS, x86-64

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/14712966/log.txt)

Mnietek, 2024-03-23:
Please help...

I can see in logs that there are correct messages from switch when I change brightness using a switch (data ‘{“61440”:XYZ}’):
`Debug 2024-03-23 20:59:09Received Zigbee message from ‘living_room/wall_switch’, type ‘attributeReport’, cluster ‘genLevelCtrl’, data ‘{“61440”:282}’ from endpoint 1 with groupID 0`

But messages published in topic ‘zigbee2mqtt/living_room/wall_switch/l1' contains constant value of brightness:
`Info 2024-03-23 20:59:09MQTT publish: topic ‘zigbee2mqtt/living_room/wall_switch/l1’, payload ‘{“brightness”:44,”power_on_behavior”:”on”,”state”:”ON”}’`

This switch is recognized as TS110E_2gang_2 and  I can read in 
https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/tuya.ts
that  brightness is read from msg.data['61440']:

```
            if (msg.data.hasOwnProperty('61440')) {
                result['brightness'] = utils.mapNumberRange(msg.data['61440'], 0, 1000, 0, 255);
            }
```

So why brightness is not properly reported in zigbee2mqtt topic?
But everything is OK when brightness is changed from HA frontend?

Can anyone provide some suggestion? What should I try to do with custom converters?
I feel like a little change is needed...

Here is logs: brightness reported by switch changes from 282 to 984, brightness published in zigbee2mqtt is 44 all the time.
Brightness is changed by switch.
```
Debug 2024-03-23 20:59:09Received Zigbee message from ‘living_room/wall_switch’, type ‘attributeReport’, cluster ‘genLevelCtrl’, data ‘{“61440”:282}’ from endpoint 1 with groupID 0
Info 2024-03-23 20:59:09MQTT publish: topic ‘zigbee2mqtt/living_room/wall_switch/l1’, payload ‘{“brightness”:44,”power_on_behavior”:”on”,”state”:”ON”}’
Debug 2024-03-23 20:59:09Received Zigbee message from ‘living_room/wall_switch’, type ‘attributeReport’, cluster ‘genLevelCtrl’, data ‘{“61440”:345}’ from endpoint 1 with groupID 0
Info 2024-03-23 20:59:09MQTT publish: topic ‘zigbee2mqtt/living_room/wall_switch/l1’, payload ‘{“brightness”:44,”power_on_behavior”:”on”,”state”:”ON”}’
Debug 2024-03-23 20:59:10Received Zigbee message from ‘living_room/wall_switch’, type ‘attributeReport’, cluster ‘genLevelCtrl’, data ‘{“61440”:408}’ from endpoint 1 with groupID 0
Info 2024-03-23 20:59:10MQTT publish: topic ‘zigbee2mqtt/living_room/wall_switch/l1’, payload ‘{“brightness”:44,”power_on_behavior”:”on”,”state”:”ON”}’
Debug 2024-03-23 20:59:10Received Zigbee message from ‘living_room/wall_switch’, type ‘attributeReport’, cluster ‘genLevelCtrl’, data ‘{“61440”:471}’ from endpoint 1 with groupID 0
Info 2024-03-23 20:59:10MQTT publish: topic ‘zigbee2mqtt/living_room/wall_switch/l1’, payload ‘{“brightness”:44,”power_on_behavior”:”on”,”state”:”ON”}’
Debug 2024-03-23 20:59:10Received Zigbee message from ‘living_room/wall_switch’, type ‘attributeReport’, cluster ‘genLevelCtrl’, data ‘{“61440”:533}’ from endpoint 1 with groupID 0
Info 2024-03-23 20:59:10MQTT publish: topic ‘zigbee2mqtt/living_room/wall_switch/l1’, payload ‘{“brightness”:44,”power_on_behavior”:”on”,”state”:”ON”}’
Debug 2024-03-23 20:59:11Received Zigbee message from ‘living_room/wall_switch’, type ‘attributeReport’, cluster ‘genLevelCtrl’, data ‘{“61440”:596}’ from endpoint 1 with groupID 0
Info 2024-03-23 20:59:11MQTT publish: topic ‘zigbee2mqtt/living_room/wall_switch/l1’, payload ‘{“brightness”:44,”power_on_behavior”:”on”,”state”:”ON”}’
Debug 2024-03-23 20:59:11Received Zigbee message from ‘living_room/wall_switch’, type ‘attributeReport’, cluster ‘genLevelCtrl’, data ‘{“61440”:659}’ from endpoint 1 with groupID 0
Info 2024-03-23 20:59:11MQTT publish: topic ‘zigbee2mqtt/living_room/wall_switch/l1’, payload ‘{“brightness”:44,”power_on_behavior”:”on”,”state”:”ON”}’
Debug 2024-03-23 20:59:12Received Zigbee message from ‘living_room/wall_switch’, type ‘attributeReport’, cluster ‘genLevelCtrl’, data ‘{“61440”:722}’ from endpoint 1 with groupID 0
Info 2024-03-23 20:59:12MQTT publish: topic ‘zigbee2mqtt/living_room/wall_switch/l1’, payload ‘{“brightness”:44,”power_on_behavior”:”on”,”state”:”ON”}’
Debug 2024-03-23 20:59:12Received Zigbee message from ‘living_room/wall_switch’, type ‘attributeReport’, cluster ‘genLevelCtrl’, data ‘{“61440”:784}’ from endpoint 1 with groupID 0
Info 2024-03-23 20:59:12MQTT publish: topic ‘zigbee2mqtt/living_room/wall_switch/l1’, payload ‘{“brightness”:44,”power_on_behavior”:”on”,”state”:”ON”}’
Debug 2024-03-23 20:59:12Received Zigbee message from ‘living_room/wall_switch’, type ‘attributeReport’, cluster ‘genLevelCtrl’, data ‘{“61440”:847}’ from endpoint 1 with groupID 0
Info 2024-03-23 20:59:12MQTT publish: topic ‘zigbee2mqtt/living_room/wall_switch/l1’, payload ‘{“brightness”:44,”power_on_behavior”:”on”,”state”:”ON”}’
Debug 2024-03-23 20:59:13Received Zigbee message from ‘living_room/wall_switch’, type ‘attributeReport’, cluster ‘genLevelCtrl’, data ‘{“61440”:910}’ from endpoint 1 with groupID 0
Info 2024-03-23 20:59:13MQTT publish: topic ‘zigbee2mqtt/living_room/wall_switch/l1’, payload ‘{“brightness”:44,”power_on_behavior”:”on”,”state”:”ON”}’
Debug 2024-03-23 20:59:13Received Zigbee message from ‘living_room/wall_switch’, type ‘attributeReport’, cluster ‘genLevelCtrl’, data ‘{“61440”:973}’ from endpoint 1 with groupID 0
Info 2024-03-23 20:59:13MQTT publish: topic ‘zigbee2mqtt/living_room/wall_switch/l1’, payload ‘{“brightness”:44,”power_on_behavior”:”on”,”state”:”ON”}’
Debug 2024-03-23 20:59:13Received Zigbee message from ‘living_room/wall_switch’, type ‘attributeReport’, cluster ‘genLevelCtrl’, data ‘{“61440”:984}’ from endpoint 1 with groupID 0
Info 2024-03-23 20:59:13MQTT publish: topic ‘zigbee2mqtt/living_room/wall_switch/l1’, payload ‘{“brightness”:44,”power_on_behavior”:”on”,”state”:”ON”}’
```

Mnietek, 2024-03-24:
By the way RTX ZDM2LN (_TZ3210_pagajpog) is similar to RTX ZDM1L (_TZ3210_weaqkhab). One is 2 gang the other - 1 gang.
This 1 gang device works perfectly... Maybe tere is something wrong with reading some data from multiEndpoint devices?

Tribalwarrior, 2024-05-19:
any solution find?

Tribalwarrior, 2024-05-19:
Same problem.
I noticed that when changing the brightness through the switch button, the "brightness": state changes, while "brightness_l1": or "brightness_l2":, depending on the switch key, does not change

Koenkk, 2024-05-21:
Pushed a fix, could you check if it works?

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

Tribalwarrior, 2024-05-21:
> Pushed a fix, could you check if it works?

Yes, on  (commit #ef68cc3) it's work great! Thanks! Master release?

closed 2024-05-22 by Koenkk