Koenkk/zigbee2mqtt #21904: Crash, possibly during firmware update

shaohme, 2024-03-22:
### What happened?

Z2M crashed while running. This happened a few minutes into firmware updating a light bulb. I experienced Z2M crash before, but haven't caught the final error/reason. It has been configured to only use syslog for logging.

```
~ $ npm start

> zigbee2mqtt@1.36.0 start
> node index.js

Error: ReadResponse 0x7cc6b6fffe08694d/1 genTime({"time":1711089100,"localTime":1711089100}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms)
    at Object.start (/chroot/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/waitress.ts:63:23)
    at /chroot/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:312:45
    at Queue.execute (/chroot/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/queue.ts:35:26)
    at ZStackAdapter.dataRequest (/chroot/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:965:9)
    at ZStackAdapter.sendZclFrameToEndpointInternal (/chroot/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:387:35)
    at Queue.execute (/chroot/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/queue.ts:35:20)
    at Request.send (/chroot/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:79:20)
    at Endpoint.readResponse (/chroot/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:509:13)
1 ~ $ 
```

### What did you expect to happen?

To run without crashing

### How to reproduce it (minimal and precise)

Run "enough" time for bug to trigger

### Zigbee2MQTT version

1.36.0

### Adapter firmware version

20230507

### Adapter

ZigStart UZG-01

### Setup

From shell, in tmux, on X86-64 Gentoo system.

### Debug log

I'll attach later if I'll manage to trigger the bug again. I didn't manage to catch the bug while debug logging to file was enabled.

chris-1243, 2024-03-22:
Which version of uzg-01 are you using? If based on CC2652P7, it might be the firmware the issue. 20230507 is unfortunately know to have some issues. @Koenkk is doing his best to build a new one for this chip but it takes time and multiple tries.

If you have the first generation based on CC2652P, then you may consider to update to this [firmware](https://github.com/Koenkk/Z-Stack-firmware/discussions/496)

shaohme, 2024-03-22:
> Which version of uzg-01 are you using? If based on CC2652P7, it might be the firmware the issue. 20230507 is unfortunately know to have some issues. @Koenkk is doing his best to build a new one for [...]

The label on the bottom says 'UZG-POE-0.5'. I bought it specifically for the CC2652P7 chip, and I have flashed it firmware targeting CC2652P7, so it should be true. I haven't disassembled it yet to check.

chris-1243, 2024-03-22:
No need to disassemble. Accordind to Zigstar documentation version 0.2+ are based on CC2652P7.

olgru, 2024-03-24:
I've the same problem. Latest version everywhere. Interesting that it worked pretty stable with UZG-01 2.0 before. I should never have updated that UZG Software to 2.5 as it all started than... I also see NWK_TABLE_FULL errors...

HobboRobin, 2024-04-16:
I have the same Problem. Is there any step further ? @Koenkk 
Currently i have a Zigstar v0.5 with the P7 inside and it is running stable with around 50 devices but when i get moore of the devices connected the adapter crashes every few hours. I have also really high traffic with around 10 spm02v02 devices that send without stop always data.
But now comes the interesting part. I have also 4 P2 Zigstar gateway´s that are v0.2 and they handle up to or even moore then the P7 thats why i think the firmware is making the Trouble.

Koenkk, 2024-04-16:
A new P7 firmware for testing can be found here; https://github.com/Koenkk/Z-Stack-firmware/discussions/496

shaohme, 2024-04-17:
My reported problem has not occurred since I reported it and begun running zigbee2mqtt with debug logging with stderr redirected to a file. Even after upgrading to 1.36.1. Not to imply causality. I'm just annoyed I have not triggered the bug yet...

HobboRobin, 2024-04-17:
i get daily times the problem. I have time at the weekend to flash the gateway firmware and then i will report :)

shaohme, 2024-10-15:
I'll close this for now, and reopen later if it occurs again

closed 2024-10-15 by shaohme