Koenkk/zigbee2mqtt #21905: Is it normal that a dongle can only support 35 connected devices?

sesame1215, 2024-03-22:
### What happened?

I've recently purchased a zigbee smart energy sensor SPM01-D2TZ from Zemismart. I am really happy with the result I have got. Now I can easily see all the energy data from HA and the data I got is the most accurate compared to all the other smart energy sensors I've tried.

But it seems like the sensor is sending all energy data every 5 sec. It somehow overloads my zigbee network. I really think the SPM01-D2TZ is a great purchase but is there any way to solve this overloading issue? Any setting I can change? Also, is it normal that the Sonoff dongle can only support 35 devices? Is there any better dongle out there that supports more connected devices?

https://www.reddit.com/r/homeassistant/comments/1bktu0f/is_it_normal_that_a_dongle_can_only_support_35/
[image]
[image]
[image]

### Zigbee2MQTT version

1.36.0

### Adapter firmware version

20230507

### Adapter

Sonoff Dongle-E

### Setup

Add-on on Hassos

mrskycriper, 2024-03-22:
> is there any way to solve this overloading issue

Tuya (Zemismart is par of this family) devices are famous for their terrible quality control and inconsistency. Unfortunately it is caused by the buggy firmware and could not be changed without a firmware update. Which is most likely never going to happen. I'd suggest geting rid of this device.

As for the Sonoff Dongle-E (and all others, based on EFR32MG21 chip), they are still not as stable as those based on Ti CC2652P chips (such as Sonoff Dongle-P). But the problem here is most likely with the network spam, coming from Tuya devices.

mrmaximas, 2024-03-22:
@sesame1215 it's normal behaviour for this device type - send data so often,  just add debounce: 5 (or more) for this talkative device. I have that too, it doesn't down the network, but I'm using CC2652P2

mrmaximas, 2024-03-22:
> > @sesame1215 it's normal behaviour for this device type - send data so often, just add debounce: 5 (or more) for this talkative device. I have that too, it doesn't shutdown the network, but I'm using CC2652P2 Sonoff Dongle P
> 
> Unfortunately, `debounce` is a purely visual band aid. The spam in the network will not stop. Only the mqtt message frequency will be reduced. And no, messages every ~200 ms (judging by the logs) are not healthy for the Zigbee network and will definitely affect it's reliability and performance.

Nevertheless, it works for me, and it works pretty stable.

closed 2024-10-19 by github-actions[bot]