Koenkk/zigbee2mqtt #21910: [New device support]: TS0601 vendor _TZE204_7gclukjs as Tuya for smart_human_presence_sensor_1

victoroos, 2024-03-22:
### Link

https://www.aliexpress.com/item/1005004961301451.html?spm=a2g0o.order_list.order_list_main.5.50fb79d20Pb3yH&gatewayAdapt=glo2nld

### Database entry

{"id":2,"type":"Router","ieeeAddr":"0xa4c138d40bfe91db","nwkAddr":58552,"manufId":4417,"manufName":"_TZE204_7gclukjs","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"stackVersion":0,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1711131109572}

### Comments

I tried to add the TS0601 but the vendor seems off.
I also tried to change the manuf ID in the database, but I don't see the change in the frontend.

### External definition

```shell
const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE204_7gclukjs',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;
```

victoroos, 2024-03-22:
Okey, the database change worked but it seems to still not be working. I guess teh vendor did chaneg more stuff when looking at the exposures.

drags3000, 2024-03-28:
https://github.com/wzwenzhi/Wenzhi-ZigBee2mqtt

closed 2024-10-25 by github-actions[bot]