Koenkk/zigbee2mqtt #21915: Issue with binding between Moes MS-105BZ and styrbar ikea

mitch55500, 2024-03-23:
### What happened?

I can't bind  Moes MS-105BZ with styrbar ikea.

I think it's because the switch Moes hasn't clusterOnOff. 
[image]

I'd like to control the MOES switch with the Styrbar.
Is there a solution?

### Zigbee2MQTT version

1.36.0

### Adapter firmware version

20211217

### Adapter

CC2652R stick

### Setup

Add-on Home Assistant in a VM

mrskycriper, 2024-03-23:
No On/Off cluster means no binding. Sometimes Tuya devices are too proprietary. The only possible fix is writing a custom firmware for the switch with proper Zigbee implementation.

mitch55500, 2024-03-23:
Who can write this?

Can I do it?

mrskycriper, 2024-03-24:
> Can I do it?

Probably not, otherwise you would have probably already done it ;)

Writing Zigbee firmware is not an easy thing to do :(

mitch55500, 2024-03-27:
An other question : how to know which dimmer switch has On/Off cluster?

mrskycriper, 2024-03-27:
> Thanks
> 
> 
> 
> An other question : how to know which dimmer switch has On/Off cluster?

If you don't have in on hand, then only by asking/searching. 

If you have it, then simply open clusters for this device tab in web interface and look for `genOnOff` in Output clusters sections.

mitch55500, 2024-03-27:
I see.

I'm just looking. I can't find much.

mitch55500, 2024-03-27:
I would like, using a Styrbar remote control from Ikea, to be able to operate the switch to be able to turn the light on or off or vary the light intensity of my lamp.

Input cluster? I have never seen this cluster on my devices.

I am not an expert, I search on [this page.](https://www.zigbee2mqtt.io/supported-devices/). Where can I see their Zigbee converters?

closed 2024-03-31 by mitch55500