Koenkk/zigbee2mqtt #21917: Crash ends with 'MAC transaction expired'

shaohme, 2024-03-23:
### What happened?

Z2M crashed while running. I didn't actively do anything. Just crashed on its own with this error:

```
~ $ npm start                                                                                                                                                                                                                                                                                                         

> zigbee2mqtt@1.36.0 start
> node index.js

Error: Command 0x00124b002919015f/1 ssIasZone.enrollRsp({"enrollrspcode":0,"zoneid":23}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC transaction expired' (240))
    at ZStackAdapter.sendZclFrameToEndpointInternal (/chroot/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:415:23)
    at Queue.execute (/chroot/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/queue.ts:35:20)
    at Request.send (/chroot/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:79:20)
    at Endpoint.command (/chroot/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:746:28)
    at Device.onZclData (/chroot/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/device.ts:229:13)
    at Controller.onZclOrRawData (/chroot/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/controller.ts:753:13)
1 ~ 
```

### What did you expect to happen?

Not crash. I realize transactions might expire, but should it crash because of it?

### How to reproduce it (minimal and precise)

Just run for a while

### Zigbee2MQTT version

1.36.0

### Adapter firmware version

20230507

### Adapter

ZigStart UZG-01 (UZG-POE-0.5)

### Setup

From shell, in tmux, on X86-64 Gentoo system.

### Debug log

[zigbee2mqtt_2024-03-23.06-04-32.log](https://github.com/Koenkk/zigbee2mqtt/files/14732886/zigbee2mqtt_2024-03-23.06-04-32.log)

Koenkk, 2024-03-24:
I don't see the error in the log, could you provide the herdsman debug log of the crash?

See [this](https://www.zigbee2mqtt.io/guide/usage/debug.html) on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

shaohme, 2024-03-24:
> I don't see the error in the log, could you provide the herdsman debug log of the crash?
> 
> See [this](https://www.zigbee2mqtt.io/guide/usage/debug.html) on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

Sure. The instance is now running with herdsman logging to stdout, in a tmux, and debug logging to file. I hope to trigger the bug again soon ...

shaohme, 2024-04-17:
> I don't see the error in the log, could you provide the herdsman debug log of the crash?
> 
> See [this](https://www.zigbee2mqtt.io/guide/usage/debug.html) on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

I have been running with debug logging in a tmux for about 3 weeks trying to capture to the bug on v.1.36.1 but it hasn't triggered yet.

shaohme, 2024-08-06:
Haven't been able to reproduce for a long time now. Assuming the issue disappeared ...

closed 2024-08-06 by shaohme