Koenkk/zigbee2mqtt #21919: [New device support]: ZG-204ZM similar to ZG-204ZL

kravemir, 2024-03-23:
### Link

https://www.aliexpress.com/item/1005006492967626.html

### Database entry

{"id":37,"type":"EndDevice","ieeeAddr":"0xa4c13835d225271e","nwkAddr":21969,"manufId":4098,"manufName":"_TZE200_2aaelwxk","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,3,1280,1,1024],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"modelId":"TS0601","manufacturerName":"_TZE200_2aaelwxk"}},"ssIasZone":{"attributes":{"iasCieAddr":"0xe0798dfffed3b030","zoneState":0}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200,"batteryVoltage":26}},"msIlluminanceMeasurement":{"attributes":{"measuredValue":32546}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":144,"stackVersion":2,"hwVersion":1,"swBuildId":"0122052017","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1711218578185}

### Comments

I ordered both variants, the PIR Infrared variant and the 24GHz radar variant from the AliExpress link.

----

Common manual for both specifies ZG-204ZS/ZG-204ZM as model names. And, that only ZG-204ZM has 24GHz sensor.

I disassembled both, and both of them have written ZG-204ZS on the PCB, but the 24GHz variant has got extra circuitry attached to it, looks like the 24GHz radar.

----

The PIR Infrared variant (ZG-204ZS) is detected as [ZG-204ZL](https://www.zigbee2mqtt.io/devices/ZG-204ZL.html#tuya-zg-204zl) in Zigbee2MQTT.

The 24GHz radar variant (ZG-204ZM) is not yet supported by Zigbee2MQTT.

----

Now, the confusing part

At first, I tried to base definition on ZL-204ZL - https://github.com/Koenkk/zigbee-herdsman-converters/blob/14dffaa06511876d096aa72e669df4b032f4cf33/src/devices/tuya.ts#L5627-L5655, and nothing worked, not even illuminance report.

Then, I tried to base definition [ZG-205Z/A (_TZE200_2aaelwxk)](https://github.com/Koenkk/zigbee-herdsman-converters/blob/14dffaa06511876d096aa72e669df4b032f4cf33/src/devices/tuya.ts#L5656-L5714), and illuminance and motion state started to work.

It works somewhat okay-ish with two obvious caveats:

- the presence attribute is always Nulls, even when motion state is large or small,
- the motion state is always large (something detected) or small, but based on the definition file, there should be also none and medium values.

I am also not sure, if configuration options have any effect.

### External definition

```shell
const {} = require('zigbee-herdsman-converters/lib/modernExtend');
// Add the lines below
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const {} = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: 'TuYa',
    description: 'Luminance motion sensor',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    exposes: [
        e.presence(), e.illuminance().withUnit('lx'),
        e.numeric('large_motion_detection_sensitivity', ea.STATE_SET).withValueMin(0).withValueMax(10).withValueStep(1).withUnit('x')
            .withDescription('Motion detection sensitivity'),
        e.numeric('large_motion_detection_distance', ea.STATE_SET).withValueMin(0).withValueMax(10).withValueStep(0.01).withUnit('m')
            .withDescription('Motion detection distance'),
        e.enum('motion_state', ea.STATE, ['none', 'small', 'medium', 'large']).withDescription('State of the motion'),
        e.numeric('fading_time', ea.STATE_SET).withValueMin(0).withValueMax(28800).withValueStep(1).withUnit('s')
            .withDescription('For how much time presence should stay true after detecting it'),
        e.numeric('medium_motion_detection_distance', ea.STATE_SET).withValueMin(0).withValueMax(6).withValueStep(0.01).withUnit('m')
            .withDescription('Medium motion detection distance'),
        e.numeric('medium_motion_detection_sensitivity', ea.STATE_SET).withValueMin(0).withValueMax(10).withValueStep(1).withUnit('x')
            .withDescription('Medium motion detection sensitivity'),
        e.binary('indicator', ea.STATE_SET, 'ON', 'OFF').withDescription('LED Indicator'),
        e.numeric('small_detection_distance', ea.STATE_SET).withValueMin(0).withValueMax(6).withValueStep(0.01).withUnit('m')
            .withDescription('Small detection distance'),
        e.numeric('small_detection_sensitivity', ea.STATE_SET).withValueMin(0).withValueMax(10).withValueStep(1).withUnit('x')
            .withDescription('Small detection sensitivity'),
    ],
    meta: {
        tuyaDatapoints: [
            [1, 'presence', tuya.valueConverter.trueFalse1],
            [2, 'large_motion_detection_sensitivity', tuya.valueConverter.raw],
            [4, 'large_motion_detection_distance', tuya.valueConverter.divideBy100],
            [101, 'motion_state', tuya.valueConverterBasic.lookup(
                {'none': tuya.enum(0), 'large': tuya.enum(1), 'medium': tuya.enum(2), 'small': tuya.enum(3)})],
            [102, 'fading_time', tuya.valueConverter.raw],
            [104, 'medium_motion_detection_distance', tuya.valueConverter.divideBy100],
            [105, 'medium_motion_detection_sensitivity', tuya.valueConverter.raw],
            [106, 'illuminance', tuya.valueConverter.raw],
            [107, 'indicator', tuya.valueConverter.onOff],
            [108, 'small_detection_distance', tuya.valueConverter.divideBy100],
            [109, 'small_detection_sensitivity', tuya.valueConverter.raw],
            // Not exposed DPs/untested
            // [103, 'motion_false_detection', tuya.valueConverter.raw],
            // [113, 'breathe_false_detection', tuya.valueConverter.raw],
            // [3, 'mov_minimum_distance', tuya.valueConverter.raw],
            // [110, 'micro_minimum_distance', tuya.valueConverter.raw],
            // [111, 'motionless_minimum_distance', tuya.valueConverter.raw],
            // [112, 'reset_setting', tuya.valueConverter.raw],
            // [114, 'time', tuya.valueConverter.raw],
            // [115, 'alarm_time', tuya.valueConverter.raw],
            // [116, 'alarm_volume', tuya.valueConverterBasic.lookup(
            //  {'low': tuya.enum(0), 'medium': tuya.enum(1), 'high': tuya.enum(2), 'mute': tuya.enum(3)})],
            // [117, 'working_mode', tuya.valueConverterBasic.lookup(
            // {'arm': tuya.enum(0), 'off': tuya.enum(1), 'alarm': tuya.enum(2),  'doorbell': tuya.enum(3)})],
            // [118, 'auto1', tuya.valueConverter.raw],
            // [119, 'auto2', tuya.valueConverter.raw],
            // [120, 'auto3', tuya.valueConverter.raw],
        ],
    },
};

module.exports = definition;
```

gijsh070, 2024-03-25:
I also got this to somehow work, but i dont see any entitities about breathing, do you?

Koenkk, 2024-03-27:
Breathing is also not supported by the converters in https://[doc.szalarm.com/zg-205ZL/cntop_zigbee_sensor.js](https://doc.szalarm.com/zg-205ZL/cntop_zigbee_sensor.js)

gijsh070, 2024-03-27:
You are right, i can see its not exposed DPs/untested. Hope it will get support in future. Thanks for your reply.

Koenkk, 2024-03-28:
@gijsh070 could you share your data/database.db entry of this device?

gijsh070, 2024-03-28:
I got a .js file from the seller but it seems like its the same as i send before. I bought 2 of these devices.

{"id":59,"type":"EndDevice","ieeeAddr":"0xa4c138a3a5310ec8","nwkAddr":21396,"manufId":4742,"manufName":"_TZE200_2aaelwxk","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,3,1280,1,1024],"outClusterList":[],"clusters":{"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200,"batteryVoltage":30}},"msIlluminanceMeasurement":{"attributes":{"measuredValue":1}},"genBasic":{"attributes":{"modelId":"TS0601","manufacturerName":"_TZE200_2aaelwxk","powerSource":3,"zclVersion":3,"appVersion":144,"stackVersion":2,"hwVersion":1,"swBuildId":"0122052017"}},"ssIasZone":{"attributes":{"iasCieAddr":"0xe0798dfffeb3d1da","zoneState":0}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":144,"stackVersion":2,"hwVersion":1,"swBuildId":"0122052017","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1711648444843}

looked-github, 2024-04-05:
Zigbee2MQTT current version: 1.36.1-1
Currently "Motion state" only accepts "large" or "none" states. "small" and "static" are missing. “Fading time” does not seem to be adjustable. Illuminance requires approximately -99% adjustment.

gargamelonly, 2024-05-13:
I have this device reported as _TZE200_kb5noeto and ZG-204ZM with FW 0122052017
It is identified by Z2M but I could not find a working setting to make it useful.
So far a useless device.

heisenberg2980, 2024-05-13:
@gargamelonly I have this device with the exact same values and FW and it is working as expected (except the issues mentioned in this thread). I changed the sensitivity to 10 (not sure if that makes any difference), and it detects motion using the PIR, and then it keeps the presence variable as ON while the presence is detected by the mmwave sensor, aren´t you experiencing the same behaviour?

gargamelonly, 2024-05-13:
@heisenberg2980 
Set "Static detection sensitivity" to 10 (max), does not help.
What do you have for "Fading time" and "Static detection distance" ?

heisenberg2980, 2024-05-13:
@gargamelonly I am not sure if changing any of the settings does anything at the moment, but I didn´t change the other values so I have the default settings (fading time = 30s, static detection distance = 6m)

truongbber, 2024-05-15:
[image]
error always Presence True status, I tried setting the sensitivity to the lowest, even though I moved out of the room, and the room does not have any devices causing interference, does anyone have a way to fix it?

kevincaradant, 2024-05-20:
Hi

I have the same device, and it mostly works as expected. However, I noticed that if I don't move for 10 seconds with the fading time set to 10 seconds, the state changes from "large" to "static." But a few seconds later, it shifts to "none."

I thought the device would stay at "static" until I left the room, but that's not the case. So, for me, it behaves more like a regular detection sensor rather than a presence sensor. :(

Has anyone else experienced this issue with _TZE200_kb5noeto (FW: 0122052017)?

gargamelonly, 2024-05-20:
Exactly the same behavior here.
Something is wrong with it (_TZE200_kb5noeto (FW: 0122052017)).
Can't solve it.

truongbber, 2024-05-21:
Is there any way to replace _TZE200_kb5noeto to _TZE200_2aaelwxk? In tuya application, the code _TZE200_2aaelwxk has version 2.1 and _TZE200_kb5noeto has version 2.1.1

bruno-dogancic, 2024-05-22:
Hey there, I've been testing the device and it stays "Presence: True" and "Motion state: large" all the time no matter the parameters (currently 35s, 5.5m and 10x). I tried restarting the device and re-paring, no changes. Is the device broken and is someone else having the same behaviour?

truongbber, 2024-05-24:
> Hey there, I've been testing the device and it stays "Presence: True" and "Motion state: large" all the time no matter the parameters (currently 35s, 5.5m and 10x). I tried restarting the device and re-paring, no changes. Is the device broken and is someone else having the same behaviour?
> 
> Device description: Zigbee Model: TS0601 Zigbee Manufacturer: _TZE200_kb5noeto Firmware version: 0122052017 Model: [ZG-204ZM](https://www.zigbee2mqtt.io/devices/ZG-204ZM.html#tuya-zg-204zm)
> 
> Update: I have recieved another ZG-204ZM from another seller and this one seems to work. I haven't tested it yet to confirm if it really works or if it is just changing states unlike the first one.

I know the product through smarthomescene, it seems like smarthomescene received money to advertise a trash product

BambamNZ, 2024-05-26:
> Hey there, I've been testing the device and it stays "Presence: True" and "Motion state: large" all the time no matter the parameters (currently 35s, 5.5m and 10x). I tried restarting the device and [...]

Same issues - no matter which setting you tweak it detects presence after you install the batteries and no amount of tweaking of settings get it to clear the presence. 

Motion_State changes between "Large" & "Static" but never goes to "None" which is suspect is why occupancy value remains "1" and not changes to "0" 

I opened mine and 1 visual difference from pictures on SmartHomeScence.com is that mine shows V2.1 versus the a date code on the pictures

mvadu, 2024-05-29:
> I know the product through smarthomescene, it seems like smarthomescene received money to advertise a trash product

well, smarthomescene article mentions `_TZE200_2aaelwxk` and we are dealing with `_TZE200_kb5noeto`, not sure i will blame him for this!

BambamNZ, 2024-05-29:
I used the link in the article to purchase and received a different unit from the review, so it is more on the AliExpress seller. It is disappointing but can only put it down as a learning experience at this point.

heisenberg2980, 2024-05-29:
> Seems _TZE200_2aaelwxk has different set of issues compared to _TZE200_kb5noeto, even tho they both have same fw

Wait a minute, the same exact device comes from two different manufacturers, and they even have different issues? this is getting interesting, and much more complex to debug/fix!

Basically people will always has to check the manufacturer when they open or review an issue for this device, as it might not apply to the other manufacturer.

In my case, with a _TZE200_kb5noeto, the device is reporting presence 100% of the time, so it is pretty useless at the moment, hoping that we are able to fix it as some point, as the device itself looks really promissing

truongbber, 2024-05-30:
I bought using all the links on AliExpress, they were all defective and useless

> well, smarthomescene article mentions `_TZE200_2aaelwxk` and we are dealing with `_TZE200_kb5noeto`, not sure i will blame him for this!

> well, smarthomescene article mentions `_TZE200_2aaelwxk` and we are dealing with `_TZE200_kb5noeto`, not sure i will blame him for this!

I bought using all the links on AliExpress, they were all defective and useless

Koterak, 2024-06-01:
I have 9 of this sensors - 8 are _TZE200_kb5noeto and 1 is _TZE200_2aaelwxk. For about 2 weeks I had problem with 2 of _TZE200_kb5noeto. It keeps showing presence. But I found that for one it was enough to just took batteries out for minute and it started working. For second I had to repair it and point to place where there is no movement to got to state presence false. And it started to work now. But sometimes it works for one day, or for hour. I don't think that there is hardware issue here, but I think that there must be something with communication.

tipofthesowrd, 2024-06-01:
Good to know; Mine also shows presence true and false (kb5noeto) but seems
very hit and miss regarding distance / fade time / etc.
Only way to get something useful from this would be to compare with the
Tuya zigbee hub imo

rolfvreijdenberger, 2024-06-01:
I have 2 of the _TZE200_kb5noeto
The issue I have is that i get a lot of false negatives (I am there, but the sensor detects 'none' as motion state,  and goes to the 'away' state for presence) which makes it unreliable as a presence detector.
PIR works fine and then the initial large/static as well .
I have tried tweaking different settings, and release 1.38 introduced 2 new ones. the motion detection sensitivity _seems_ to make a bit of a difference, but not sure yet (release was an hour ago :)) and for the detection mode (also new) I have filed a bug report.

The device seems to hold a lot of promise (batteries/mmwave/good adjustability/cheap) but fails to deliver so far. I hope for better integration/firmware update, whatever makes it work :)

pruvi9019, 2024-06-02:
The device works perfectly fine with the Tuya app using a Tuya Zigbee gateway. However, I encountered the same issues mentioned by others. I noticed that setting 'Static detection sensitivity' to 0 allows 'Presence' detection to work normally. I hope someone figures out a solution soon because I bought six of them from AliExpress.

papperone, 2024-06-02:
With latest Z2M 1.38.0-1 this device has 2 new parameters but one seems to be wrongly setup (see below error)
[image]

Koterak, 2024-06-02:
Yees, I already opened second report - there is wrong letter - "Dadar"
instead of "Radar"

papperone, 2024-06-02:
> […](#)

the issue is also the fact that the text passed is not the one expected, e.g. "Only PIR" instead of "only_pir", on top of the typo you mentioned ;)

kevincaradant, 2024-06-02:
https://github.com/Koenkk/zigbee2mqtt/issues/21919#issuecomment-2143883855

Does setting 'Static detection sensitivity' to 0 mean that the 'Large', 'Static', and 'None' options work as expected? Or does it mean that the 'Static' option is no longer recognized, effectively turning it into a simple motion detector instead of a presence detector?

Your previous answer was unclear to me. 

Thank you for your answer.

speedycars, 2024-06-03:
> large

> Hey there, I've been testing the device and it stays "Presence: True" and "Motion state: large" all the time no matter the parameters (currently 35s, 5.5m and 10x). I tried restarting the device and [...]

Just experienced the same "stuck-to-large" situation for the last couple of days. No matter what I did with the sensor, it never changed back to large.
I was thinking about it - if it's stuck to "large" then the PIR module should be falsely reporting this. Opened the case and re-soldered the 3 points on its back. Right after I put in batteries, it started acting like it should. I will test it for few more days though.

Koterak, 2024-06-03:
Firstly I tried the same. But I found out that (for me) it started to work
when I pull batteries out for 1/2 minutes. 8 of sensors are working good so
far - when it got stuck I have just restart them (got problems with 2 so
far, others are working fine). But last obe sensor have problem every few
hours. But maybe send photo which solder joints you resoldered? Maybe I
resoldered others and will try same as you

speedycars, 2024-06-04:
@Koterak, this might be a different issue. My reply was mostly about the issue described by @bruno-dogancic.
Anyway, here is the photo of the re-soldered 3 points:
[image]

Koterak, 2024-06-05:
Interesting, tried to resolder this pins too - today it’s second day when
device is working allright. Before it was getting stuck after few hours.

speedycars, 2024-06-05:
Great to hear there is a significant improvement!
Mine stays stable too so far.

Just a quick tip: If you got a screenshot from HA (History) where the "motion_state" attribute stays at "large" for hours, the aliexpress would provide a refund (hope that will help in improving the soldering quality). It took me some time to convince them I cannot do a video to proof my testing process as this is just a sensor and there are no visible signs on its hardware if it's working fine or not and that behavior is totally random, but in the end they agreed.

heisenberg2980, 2024-06-05:
@speedycars that's a very good idea, if we request the refund from Aliexpress hopefully the seller will improve the design pretty quickly. Just a quick question, has anyone had the same issue with the Tuya app? I don´t think there will be any difference as it seems a hardware issue, but just wondering if anyone has tried if the same issue happens using the Tuya app

speedycars, 2024-06-05:
I don't have a Tuya gateway anymore, never had a chance to test there. But, yes, I don't believe there will be any difference.

bruno-dogancic, 2024-06-05:
Great to hear it! I will try to tinker with mine devices to see what will happen

shartdain, 2024-06-06:
I ran MQTTX to see if it communicates at all. It goes silent for a long period. I think it could be software issue. Do we know if there a firmware update? Sending the set command to the queue {"motion_detection_mode": "Only Dadar"} give the error: can not enumirate

kevincaradant, 2024-06-08:
Hi every one, 

Is it only my device which is draining the battery ?

One battery for 2 weeks is so few :/
[image]

The device is not responding anymore at 5% since today.

shartdain, 2024-06-08:
> Hi every one,
> 
> Is it only my device which is draining the battery ?
> 
> One battery for 2 weeks is so few :/ ![Screenshot_20240608-115658](https://private-user-images.githubusercontent.com/6093175/337866622-a48e73b8-ecfb-4f05-88c5-aa3946509519.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTc4NjcwNDEsIm5iZiI6MTcxNzg2Njc0MSwicGF0aCI6Ii82MDkzMTc1LzMzNzg2NjYyMi1hNDhlNzNiOC1lY2ZiLTRmMDUtODhjNS1hYTM5NDY1MDk1MTkucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDYwOCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA2MDhUMTcxMjIxWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MzBjODM4MzllZDE3NDFjM2MxYTEzNDEzMDZlZWI1ZTkzMTAxNTg4NGYzZDI1NTAwOTM4MWUzMGM4NWY5OGY4MiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.cCpfZ_ohEIAzTgpT2cTBY634BVZTGv3wuisu6LtL3MQ)
> 
> The device is not responding anymore at 5% since today.

My staying at 100% for a week now.

shartdain, 2024-06-08:
Thank you @speedycars for a tip for resoldering the 3 pins!!! Once I have done it, it has stayed stable for a couple of days now. Still can not change the motion_detection_mode. Will see if the firmware update (if it ever becomes available) solves the issue.

ArkadiuszNiemiec, 2024-06-25:
I have V2.1. It was getting stuck, so I gave up. Guess I will power it on again and resolder if it gets stuck again.

francisp2, 2024-07-04:
> Just a quick tip: If you got a screenshot from HA (History) where the "motion_state" attribute stays at "large" for hours, the aliexpress would provide a refund 

Mine is doing that when paired to Zigbee2MQTT, but not when paired with a Tuya hub.

garret, 2024-07-04:
> Mine is doing that when paired to Zigbee2MQTT, but not when paired with a Tuya hub.

So is it working correctly with a Tuya hub?

francisp2, 2024-07-05:
With Zigbee2MQTT, it gets stuck in 'occupancy', with the Tuya hub there is only 'motion detected', but it does not get stuck (and even if you sit still, motion keeps detected).  (oh, and with Tuya hub, there is no lux in Home Assistant)

I still could try the soldering fix

mrmaximas, 2024-07-25:
Can you tell me how often in z2m this radar ("product_id": "kb5noeto") sends battery status? I haven't seen battery reports in the classic TUYA logs for a week now.

kubonm, 2024-07-31:
Hello, I have ZG-204ZM (_TZE200_kb5noeto ) with firmware 0122052017 using Z2M 1.39.0-1.
Presence detection seems to be not working at all, only in PIR mode (which I already have).
In combined mode it will stuck at "static" state. If I switch detection mode to radar it will never show motion/presence...
Is this something that I can solve with mentioned resoldering of pins? Or it's simply not possible to get it working so I should look for another presence radar?
Thank you very much for response.

My current "PIR detection working" config from Z2M:
 {
    "battery": 100,
    "fading_time": 30,
    "illuminance_lux": 0,
    "indicator": "ON",
    "linkquality": 96,
    "motion_detection_mode": "pir_and_radar",
    "motion_detection_sensitivity": 0,
    "motion_state": "none",
    "presence": false,
    "static_detection_distance": 8,
    "static_detection_sensitivity": 0
}

Klavishnik, 2024-08-13:
> Здравствуйте, у меня ZG-204ZM (_TZE200_kb5noeto) с прошивкой 0122052017, использующей Z2M 1.39.0-1. Обнаружение присутствия, похоже, вообще не работает, только в режиме PIR (который у меня уже есть). В комбинированном режиме он зависает в "статическом" состоянии. Если я переключу режим обнаружения на радар, он никогда не покажет движение/присутствие... Можно ли это решить с помощью упомянутой перепайки контактов? Или его просто невозможно заставить работать, поэтому мне следует поискать другой радар присутствия? Большое спасибо за ответ.
> 
> Моя текущая конфигурация «Обнаружение PIR работает» от Z2M: { "battery": 100, "fading_time": 30, "illuminance_lux": 0, "indicator": "ON", "linkquality": 96, "motion_detection_mode": "pir_and_radar", "motion_detection_sensitivity": 0, "motion_state": "none", "presence": false, "static_detection_distance": 8, "static_detection_sensitivity": 0 }

Same problem.
Did you solve it?

kubonm, 2024-08-13:
> Same problem. Did you solve it?

Hello, it magically started working, there are few problems - sometimes presence detection not working, only motion, but overal it 70% works.. :D

JerryLutor, 2024-08-13:
Hello. 
I have ZG-204ZS v2.1,  how add support to zigbee2mqtt 1.39.1 ?

vision2003, 2024-09-03:
I’ve got the kb5noeto version one. Detection works quite well actually, but it takes forever (sometimes up to 5 minutes) before it clears “Presence” to false" and “Motion state” to none.

gargamelonly, 2024-09-14:
Replaced the batteries today. 
After 5 months it dropped to 10% so I guess 6 month battery life, as stated by the specification, is ok.

kevincaradant, 2024-09-14:
https://github.com/Koenkk/zigbee2mqtt/issues/21919#issuecomment-2351060870

I've needed to replace them almost every month until now. I used duracell and energizer with the same problem. Bad luck perhaps?
There I switched to rechargeable batteries. They are almost 1 month old for now, it still works. I'll see if I make it to 2 months 🤣

Stephan-4711, 2024-09-24:
I have some of these Sensors too. My problem is, that they detect motion, when there is no motion. Do anybody else have the same problem?
I'm combining pir and radar. there is no animal or something which could be detected.

mrmaximas, 2024-09-24:
> I have some of these Sensors too. My problem is, that they detect motion, when there is no motion. Do anybody else have the same problem? I'm combining pir and radar. there is no animal or something which could be detected.

i have the same behaviour in the classic tuya system once-twice a week with the 1x pir/radar sensivity settings

alexsotoaguilera, 2024-10-25:
TS0601
_TZE200_kb5noeto

I have doubts about the states that the sensor should send. Motion state only shows the states "large", "small" or "none". "static" are missing. I have read that it also sends a "static" status. 

is "small" equivalent to "static"? or my sensor is not working?

on the other hand the  option to enable/disable led indicator does not work

kevincaradant, 2024-10-25:
@alexsotoaguilera, mine give me 'large' attribute when I move. And when I stay frozen ( sit on a chair for example), it give me the 'static' atrribute.

I never got 'small' atrribute. But I supposed it was for pets like cat. I don't have pets to check that.

alexsotoaguilera, 2024-10-25:
I also have it set to "pir_and_radar".  In my case, your state "static" behaves like my state "small". The important thing is that we both received only 3 statuses.

Large (when detected)
Small or Static (when I don't move )
None (when leave)

is anyone getting the 4 different states?

mrmaximas, 2024-10-25:
small means, for example, a person breathing in a static position. not a cat or small animal moving.

alexsotoaguilera, 2024-10-27:
finalmente he pedido devolución, después de unas horas de funcionamiento nunca detecta si no hay presencia solo sigue detectando el cambio entre "large" y "small" pero no cambia a 'none'.

[50 more replies]

closed 2025-05-15 by github-actions[bot]