Koenkk/zigbee2mqtt #21924: Yookee D10110 Smart blind controller identifying as TS0301

ss1225dev, 2024-03-24:
### What happened?

I have several motorized roller shades from Graywind blinds. They all have the same zigbee Yooksmart motor in them. Almost all of them identify and work correctly as Yookee D10110. 3 of them identify as unsupported manufacturer "_TZE200_9caxna4s" and model "TS0301". These appear to report battery correctly but nothing else. Any advice is appreciated! 

Info 2024-03-23 21:02:36Device '0x0ceff6fffe4d394b' joined
Info 2024-03-23 21:02:37MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0ceff6fffe4d394b","ieee_address":"0x0ceff6fffe4d394b"},"type":"device_joined"}'
Info 2024-03-23 21:02:37Starting interview of '0x0ceff6fffe4d394b'
Info 2024-03-23 21:02:39MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0ceff6fffe4d394b","ieee_address":"0x0ceff6fffe4d394b","status":"started"},"type":"device_interview"}'
Info 2024-03-23 21:02:52MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0ceff6fffe4d394b","ieee_address":"0x0ceff6fffe4d394b"},"type":"device_announce"}'
Info 2024-03-23 21:03:04Successfully interviewed '0x0ceff6fffe4d394b', device has successfully been paired
Warning 2024-03-23 21:03:04Device '0x0ceff6fffe4d394b' with Zigbee model 'TS0301' and manufacturer name '_TZE200_9caxna4s' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
Info 2024-03-23 21:03:05MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Automatically generated definition","exposes":[{"access":5,"description":"Remaining battery in %","label":"Battery","name":"battery","property":"battery","type":"numeric","unit":"%"},{"access":1,"category":"diagnostic","description":"Link quality (signal strength)","label":"Linkquality","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"TS0301","options":[],"supports_ota":false,"vendor":"_TZE200_9caxna4s"},"friendly_name":"0x0ceff6fffe4d394b","ieee_address":"0x0ceff6fffe4d394b","status":"successful","supported":false},"type":"device_interview"}'
Info 2024-03-23 21:03:05Configuring '0x0ceff6fffe4d394b'
Info 2024-03-23 21:03:05MQTT publish: topic 'homeassistant/sensor/0x0ceff6fffe4d394b/battery/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x0ceff6fffe4d394b"],"manufacturer":"_TZE200_9caxna4s","model":"Automatically generated definition (TS0301)","name":"0x0ceff6fffe4d394b","via_device":"zigbee2mqtt_bridge_0x00124b002a7b4e77"},"device_class":"battery","enabled_by_default":true,"entity_category":"diagnostic","object_id":"0x0ceff6fffe4d394b_battery","origin":{"name":"Zigbee2MQTT","sw":"1.36.0","url":"https://www.zigbee2mqtt.io"},"state_class":"measurement","state_topic":"zigbee2mqtt/0x0ceff6fffe4d394b","unique_id":"0x0ceff6fffe4d394b_battery_zigbee2mqtt","unit_of_measurement":"%","value_template":"{{ value_json.battery }}"}'
Info 2024-03-23 21:03:05MQTT publish: topic 'homeassistant/sensor/0x0ceff6fffe4d394b/linkquality/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x0ceff6fffe4d394b"],"manufacturer":"_TZE200_9caxna4s","model":"Automatically generated definition (TS0301)","name":"0x0ceff6fffe4d394b","via_device":"zigbee2mqtt_bridge_0x00124b002a7b4e77"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:signal","name":"Linkquality","object_id":"0x0ceff6fffe4d394b_linkquality","origin":{"name":"Zigbee2MQTT","sw":"1.36.0","url":"https://www.zigbee2mqtt.io"},"state_class":"measurement","state_topic":"zigbee2mqtt/0x0ceff6fffe4d394b","unique_id":"0x0ceff6fffe4d394b_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
Info 2024-03-23 21:03:07MQTT publish: topic 'zigbee2mqtt/0x0ceff6fffe4d394b', payload '{"battery":67,"linkquality":109}'
Info 2024-03-23 21:03:07Successfully configured '0x0ceff6fffe4d394b'

### What did you expect to happen?

All devices would identify as Yookee D10110

### Zigbee2MQTT version

1.36.0

### Adapter firmware version

20230507

### Adapter

TubesZB CC2652 P7

### Setup

Home Assistant Add-On

zalterego36, 2024-03-25:
I'm having a similar issue. I purchased Yoolax blinds identical to other blinds I've bought in the past that identified as D10110 and are now identifying as _TZE200_9caxna4s. I reached out to the seller and they seemed to indicate that there was a change in motor. I'm going to try to add support using the instructions, but have never done so before.

ss1225dev, 2024-03-25:
> I'm having a similar issue. I purchased Yoolax blinds identical to other blinds I've bought in the past that identified as D10110 and are now identifying as _TZE200_9caxna4s. I reached out to the [...]

I was able to fix this by creating the following external converter and saving as zigbee2mqtt/D10110.js

```
const {} = require('zigbee-herdsman-converters/lib/modernExtend');
// Add the lines below
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const {} = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['TS0301'],
    model: 'D10110', // Update this with the real model of the device (written on the device itself or product page)
    vendor: 'Yookee', // Update this with the real vendor of the device (written on the device itself or product page)
    description: 'Smart blind controller', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    extend: [],
    fromZigbee: [fz.cover_position_tilt, fz.battery],
    toZigbee: [tz.cover_state, tz.cover_position_tilt],
    exposes: [e.cover_position(), e.battery()], // Defines what this device exposes, used for e.g. Home Assistant discovery and in the frontend
};

module.exports = definition;
```

then adding this to zigbee2mqtt/configuration.yaml
```
external_converters:
  - D10110.js
```

Koenkk, 2024-07-03:
Added!

Could you also submit a picture for the docs? This can be done by:
1. Create a fork by clicking [here](https://github.com/Koenkk/zigbee2mqtt.io/fork)
2. Go to the `public/images/devices` directory, *Add file* -> *Upload files*
3. Upload the files and press *Commit changes*
4. Press *Contribute* -> *Open pull request* -> update title/description -> *Create pull request*

Make sure that:
- The filename is `D10110_1.png` (update `model` accordingly) 
- The size is 512x512
- The background is transparent (use e.g. [Adobe remove background](https://new.express.adobe.com/tools/remove-background))

closed 2024-07-03 by Koenkk

mkothawala, 2024-07-04:
@ss1225dev & @Koenkk 

I am trying to use your fix but encountering the following error. 

> error: 	z2m: Failed to load external converter file 'D10110.js' (Cannot find module 'zigbee-herdsman-converters/lib/extend'
> Require stack:
> - /app/dist/util/utils.js
> - /app/dist/util/settings.js
> - /app/index.js)

I am running Z2M as an addon and have version 1.39.0-1. Am I doing something wrong?

Thank you in advance.

mkothawala, 2024-07-04:
I figured out how to stop the error. I have to use Zigbee2MQTT edge.

Now that the external converter works, it still doesn't report the correct position. It just report 50% as the position regardless of where the shades actually are.

mkothawala, 2024-07-04:
I keep answering my own questions here so that the next person that tries to figure this out, does it faster. To get an accurate position from the shades, you have to set the upper and lower limit with the included remote control. This is with @ss1225dev fix using Z2MQTT edge.
If I have the time I will try it out in stable Z2MQTT with the unsupported TS0301 from vendor _TZE200_9caxna4s and report back if it works as @rohankapoorcom says it works for him.

rohankapoorcom, 2024-07-04:
I haven't switched to edge yet to test this, but looking at the commit history, I believe if you are running the latest edge version, it should already show this motor as supported since https://github.com/Koenkk/zigbee-herdsman-converters/commit/029257dda6bf586eb3f3dd0340ba99a7df71a1b9 is included in it.

You should not need a custom converter there.

rohankapoorcom, 2025-01-30:
Sorry for bumping this old thread @Koenkk - is there a way to remove the `tilt`? I just purchased a bunch more of these and they don't support tilting so having the tilting controls exposed in Home Assistant is strange.

Koenkk, 2025-02-01:
It uses [`e.cover_position()`](https://github.com/Koenkk/zigbee-herdsman-converters/blob/b3b8375b92c002bda3e48b5c6af7533399f646bd/src/devices/yookee.ts#L23) so that should not expose tilt (at least not with the latest z2m version)

rohankapoorcom, 2025-02-01:
Am I looking at the wrong file? I'm seeing tilt here: https://github.com/Koenkk/zigbee-herdsman-converters/blob/b3b8375b92c002bda3e48b5c6af7533399f646bd/src/devices/tuya.ts#L2675

Koenkk, 2025-02-02:
Ah sorry, I searched for `D10110`, fixed it!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

rohankapoorcom, 2025-02-02:
Thanks! Hopefully the same motor is not used in devices that support tilt.