Koenkk/zigbee2mqtt #21925: Philips Hue looses connection after power outage or OTA

RikJ01, 2024-03-24:
### What happened?

After a power loss or OTA of my Philips Hue lights, the connection with Zigbee2MQTT is lost. I would need to turn on "allow new devices to join", power cycle the bulb and use my [Philips Hue remote ](https://www.zigbee2mqtt.io/devices/324131092621.html) to factory reset the bulb.

### What did you expect to happen?

I would expect the Philips Hue devices automatically rejoin the network. My Ikea Tradfri bulbs and INNR wall plugs do this without any issue.

### How to reproduce it (minimal and precise)

Turn off the power to a Philips Hue bulb or do an OTA through Zigbee2MQTT.

### Zigbee2MQTT version

1.36.0

### Adapter firmware version

20230507

### Adapter

Electrolama zig-a-zig-ah! (zzh!)

### Setup

Home Assistant OS on Intel NUC

### Debug log

[log - kopie.txt](https://github.com/Koenkk/zigbee2mqtt/files/14734881/log.-.kopie.txt)

Funkiemitch, 2024-05-08:
I have the same problem.

After a ota update connection is lost and will not rejoin. also a restart of device and zigbee2mqtt wont help.

If a hue device is turned off physically (power down) it will not rejoin after powercycle.

adapter skyconnect

firmware zigbee2mqtt 1.3.7.0

RikJ01, 2024-05-22:
I have "fixed" this issue by removing the Zigbee2MQTT add-on, restart the whole system and re-installed the add-on. I have completely started over by adding all devices one by one.