Koenkk/zigbee2mqtt #21930: How to pair IH-K665 with zigbee2mqtt

kidekoop2016, 2024-03-24:
### Link

https://es.aliexpress.com/item/1005005651861263.html?spm=a2g0o.order_list.order_list_main.60.2dde194dRVk5Hr&gatewayAdapt=glo2esp

### Database entry

I cannot pair it

### Comments

Hi
I have this device and I want to connect to HA supervised using zigbee 3.0 USB dongle.
How could I get it?
There is no information here: https://www.zigbee2mqtt.io/devices/IH-K665.html
Regards

### External definition

```shell
no data
```

FuturamaPhill, 2024-04-02:
press down on the water symbol and hold it should flash blue when pairing, would be useful if this info was added to documentation

kidekoop2016, 2024-04-08:
> press down on the water symbol and hold it should flash blue when pairing, would be useful if this info was added to documentation

Thank you [FuturamaPhill](https://github.com/FuturamaPhill)

- I have tried it several times but it (with 2 devices) doesnt work (battery is ok)
- In addition I have update the firmware of my coordinator Sonoff USB  Dongle version P using the official guide: [SONOFF Zigbee 3.0 USB Dongle Plus ZBDongle-P](https://www.zigbee2mqtt.io/guide/adapters/#recommended)
  
Any suggestion?
Thanks in advance

kidekoop2016, 2024-11-22:
I found the solution. And it was pretty much more simple.
Just I pressed most powerfully. Thats was all.

Thank you¡

closed 2025-01-30 by github-actions[bot]