Koenkk/zigbee2mqtt #21934: lights not in group turning on/off when controlling group

twhiston, 2024-03-25:
### What happened?

I have a group with all the lights in a room, running zigbee2mqtt on a rpi 3. After a restart of the rpi I've found that some actions on my groups (via apple home) control other lights in the house. I've checked the groups and the lights are not bound to it, and checking the logs I can see that it does not appear to be getting messages related to controlling the lamp which should not be turning on, so I'm not sure why it happens but it's happened to multiple groups in my house now.

### What did you expect to happen?

Only the lights in the group turn on/off with the command

### How to reproduce it (minimal and precise)

Create a group of lights, restart computer, try controlling lights, other lights turn on

### Zigbee2MQTT version

1.36.0

### Adapter firmware version

0x264d0900

### Adapter

ConBee3 

### Setup

z2m on rpi, plugin to homebridge, controlled by homekit

### Debug log

In the case of this example you can see that the kitchen lamp is turning on when the studio lights group is being set on
```
Debug 2024-03-25 11:10:31Received MQTT message on 'zigbee2mqtt/Studio Lights/set' with data '{"brightness":254,"color_temp":303,"state":"ON"}'
Debug 2024-03-25 11:10:31Publishing 'set' 'state' to 'Studio Lights'
Debug 2024-03-25 11:10:31Received Zigbee message from 'Kitchen Lamp', type 'attributeReport', cluster 'manuSpecificPhilips2', data '{"state":{"data":[7,0,1,254,47,1],"type":"Buffer"}}' from endpoint 11 with groupID null
Debug 2024-03-25 11:10:31No converter available for '9290024688' with cluster 'manuSpecificPhilips2' and type 'attributeReport' and data '{"state":{"data":[7,0,1,254,47,1],"type":"Buffer"}}'
Debug 2024-03-25 11:10:31Publishing 'set' 'color_temp' to 'Studio Lights'
Info 2024-03-25 11:10:33MQTT publish: topic 'zigbee2mqtt/Studio Door 1', payload '{"brightness":254,"color":{"hue":36,"saturation":68,"x":0.417,"y":0.3963},"color_mode":"color_temp","color_temp":303,"linkquality":181,"state":"ON","update":{"installed_version":16780814,"latest_version":16785162,"state":"available"},"update_available":true}'
Info 2024-03-25 11:10:33MQTT publish: topic 'zigbee2mqtt/Studio Lights', payload '{"brightness":254,"color":{"hue":36,"saturation":68,"x":0.417,"y":0.3963},"color_mode":"color_temp","color_temp":303,"state":"ON"}'
Info 2024-03-25 11:10:33MQTT publish: topic 'zigbee2mqtt/Studio Door 2', payload '{"brightness":254,"color":{"hue":36,"saturation":68,"x":0.417,"y":0.3963},"color_mode":"color_temp","color_temp":303,"linkquality":193,"state":"ON","update":{"installed_version":16780814,"latest_version":16785162,"state":"available"},"update_available":true}'
Info 2024-03-25 11:10:33MQTT publish: topic 'zigbee2mqtt/Studio Door 3', payload '{"brightness":254,"color":{"hue":36,"saturation":68,"x":0.417,"y":0.3963},"color_mode":"color_temp","color_temp":303,"linkquality":187,"state":"ON","update":{"installed_version":16780814,"latest_version":16785162,"state":"available"},"update_available":true}'
Info 2024-03-25 11:10:33MQTT publish: topic 'zigbee2mqtt/Studio Door 4', payload '{"brightness":254,"color":{"hue":36,"saturation":68,"x":0.417,"y":0.3963},"color_mode":"color_temp","color_temp":303,"linkquality":180,"state":"ON","update":{"installed_version":16780814,"latest_version":16785162,"state":"available"},"update_available":true}'
Info 2024-03-25 11:10:33MQTT publish: topic 'zigbee2mqtt/Studio Middle 1', payload '{"brightness":254,"color":{"hue":36,"saturation":68,"x":0.417,"y":0.3963},"color_mode":"color_temp","color_temp":303,"linkquality":223,"state":"ON","update":{"installed_version":16780814,"latest_version":16785162,"state":"available"},"update_available":true}'
Info 2024-03-25 11:10:33MQTT publish: topic 'zigbee2mqtt/Studio Middle 2', payload '{"brightness":254,"color":{"hue":36,"saturation":68,"x":0.417,"y":0.3963},"color_mode":"color_temp","color_temp":303,"linkquality":188,"state":"ON","update":{"installed_version":16780814,"latest_version":16785162,"state":"available"},"update_available":true}'
Info 2024-03-25 11:10:33MQTT publish: topic 'zigbee2mqtt/Studio Middle 3', payload '{"brightness":254,"color":{"hue":36,"saturation":68,"x":0.417,"y":0.3963},"color_mode":"color_temp","color_temp":303,"linkquality":187,"state":"ON","update":{"installed_version":16780814,"latest_version":16785162,"state":"available"},"update_available":true}'
Info 2024-03-25 11:10:33MQTT publish: topic 'zigbee2mqtt/Studio Middle 4', payload '{"brightness":254,"color":{"hue":36,"saturation":68,"x":0.417,"y":0.3963},"color_mode":"color_temp","color_temp":303,"linkquality":180,"state":"ON","update":{"installed_version":16780814,"latest_version":16785162,"state":"available"},"update_available":true}'
Info 2024-03-25 11:10:33MQTT publish: topic 'zigbee2mqtt/Studio Window 1', payload '{"brightness":254,"color":{"hue":36,"saturation":68,"x":0.417,"y":0.3963},"color_mode":"color_temp","color_temp":303,"linkquality":187,"state":"ON","update":{"installed_version":16780814,"latest_version":16785162,"state":"available"},"update_available":true}'
Info 2024-03-25 11:10:33MQTT publish: topic 'zigbee2mqtt/Studio Window 2', payload '{"brightness":254,"color":{"hue":36,"saturation":68,"x":0.417,"y":0.3963},"color_mode":"color_temp","color_temp":303,"linkquality":180,"state":"ON","update":{"installed_version":16780814,"latest_version":16785162,"state":"available"},"update_available":true}'
Info 2024-03-25 11:10:33MQTT publish: topic 'zigbee2mqtt/Studio Window 3', payload '{"brightness":254,"color":{"hue":36,"saturation":68,"x":0.417,"y":0.3963},"color_mode":"color_temp","color_temp":303,"linkquality":187,"state":"ON","update":{"installed_version":16780814,"latest_version":16785162,"state":"available"},"update_available":true}'
Info 2024-03-25 11:10:33MQTT publish: topic 'zigbee2mqtt/Studio Window 4', payload '{"brightness":254,"color":{"hue":36,"saturation":68,"x":0.417,"y":0.3963},"color_mode":"color_temp","color_temp":303,"linkquality":179,"state":"ON","update":{"installed_version":16780814,"latest_version":16785162,"state":"available"},"update_available":true}'
Info 2024-03-25 11:10:33MQTT publish: topic 'zigbee2mqtt/Studio Door', payload '{"brightness":254,"color":{"hue":36,"saturation":68,"x":0.417,"y":0.3963},"color_mode":"color_temp","color_temp":303,"state":"ON"}'
Info 2024-03-25 11:10:33MQTT publish: topic 'zigbee2mqtt/Studio Middle', payload '{"brightness":254,"color":{"hue":36,"saturation":68,"x":0.417,"y":0.3963},"color_mode":"color_temp","color_temp":303,"state":"ON"}'
Info 2024-03-25 11:10:33MQTT publish: topic 'zigbee2mqtt/Studio Window', payload '{"brightness":254,"color":{"hue":36,"saturation":68,"x":0.417,"y":0.3963},"color_mode":"color_temp","color_temp":303,"state":"ON"}'
```

jamesonuk, 2024-04-04:
devices store which group they are bound to on them.   Z2M shows the group members it knows about not which devices are linked to groups.

My guess is that the kitchen light thinks it is part of the group and as the group messages happen at the Zigbee level it gets turned on as it thinks it is a group member.
Easiest way to be sure if to create a new group and link the kitchen light to that,   I suspect it would then stop coming on with the studio lights.

There is probably a better way you can reset the groups on a device but this has worked for me before

VDRainer, 2024-04-13:
Same Problem here.
Group 1 with two lights, any state change on that group changes third light, not in that group.

> Easiest way to be sure if to create a new group and link the kitchen light to that, I suspect it would then stop coming on with the studio lights.

A new group 2 with the third light doesn't fix this.

twhiston, 2024-04-17:
The strange thing for me is that this didn't happen initially, only after a reboot of the rpi which is running zigbee2mqtt

@VDRainer what I found worked for me is to delete all my groups and make them all again using specific group id's which were not used in my previous setup. I think I started at id 20 and then worked up from there, now everything seems to work as expected, though I guess I need to see if anything acts strangely after a few reboots.

jamesonuk, 2024-10-15:
bumping again

closed 2025-01-06 by github-actions[bot]