Koenkk/zigbee2mqtt #21940: [New device support]: ZN2S-RS3E-DH

MiAutomations, 2024-03-25:
### Link

https://www.zemismart.com/products/zn2s-rs3e-dh

### Database entry

{"id":99,"type":"Router","ieeeAddr":"0x28dba7fffea87141","nwkAddr":38609,"manufId":4098,"manufName":"_TZE204_znvwzxkq","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0601","manufacturerName":"_TZE204_znvwzxkq","powerSource":1,"zclVersion":3,"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1711380284223}

### Comments

Hello I try to add this new device that is a dimmer with 3 gangs, with a possible to change the lamp/bulb type over the  APP on the smart life tuya Led, Incandescent or Halogen , status light indicator on/off  , backlight dimmer percentage, backlight color switch ) red, blue, green, white, yellow, magenta, cyan, warm white, warm yellow ) child lock and restart status ( Power on / Power off last state) 

Can you please try include this new device as a supported device in the Z2M? 

Please let me know if you need any further information 

Thank you 

### External definition

```shell
const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE204_znvwzxkq',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;
```

lambsfry, 2024-04-01:
Just got  the same, these are the DP values from Tuya cloud platform:

{
  "result": {
    "properties": [
      {
        "code": "switch_led_1",
        "custom_name": "",
        "dp_id": 1,
        "time": 1711947764500,
        "value": true
      },
      {
        "code": "bright_value_1",
        "custom_name": "",
        "dp_id": 2,
        "time": 1711948874393,
        "value": 140
      },
      {
        "code": "brightness_min_1",
        "custom_name": "",
        "dp_id": 3,
        "time": 1711947511746,
        "value": 10
      },
      {
        "code": "led_type_1",
        "custom_name": "",
        "dp_id": 4,
        "time": 1711947516761,
        "value": "led"
      },
      {
        "code": "brightness_max_1",
        "custom_name": "",
        "dp_id": 5,
        "time": 1711947511546,
        "value": 1000
      },
      {
        "code": "countdown_1",
        "custom_name": "",
        "dp_id": 6,
        "time": 1711947517199,
        "value": 0
      },
      {
        "code": "switch_led_2",
        "custom_name": "",
        "dp_id": 7,
        "time": 1711947761016,
        "value": false
      },
      {
        "code": "bright_value_2",
        "custom_name": "",
        "dp_id": 8,
        "time": 1711947698011,
        "value": 490
      },
      {
        "code": "brightness_min_2",
        "custom_name": "",
        "dp_id": 9,
        "time": 1711947511845,
        "value": 10
      },
      {
        "code": "led_type_2",
        "custom_name": "",
        "dp_id": 10,
        "time": 1711947516776,
        "value": "led"
      },
      {
        "code": "brightness_max_2",
        "custom_name": "",
        "dp_id": 11,
        "time": 1711947511641,
        "value": 1000
      },
      {
        "code": "countdown_2",
        "custom_name": "",
        "dp_id": 12,
        "time": 1711947517273,
        "value": 0
      },
      {
        "code": "relay_status",
        "custom_name": "",
        "dp_id": 14,
        "time": 1711947516998,
        "value": "off"
      },
      {
        "code": "switch_led_3",
        "custom_name": "",
        "dp_id": 15,
        "time": 1711947760578,
        "value": false
      },
      {
        "code": "bright_value_3",
        "custom_name": "",
        "dp_id": 16,
        "time": 1711947548018,
        "value": 1000
      },
      {
        "code": "brightness_min_3",
        "custom_name": "",
        "dp_id": 17,
        "time": 1711947511874,
        "value": 10
      },
      {
        "code": "led_type_3",
        "custom_name": "",
        "dp_id": 18,
        "time": 1711947507414,
        "value": "led"
      },
      {
        "code": "brightness_max_3",
        "custom_name": "",
        "dp_id": 19,
        "time": 1711947511743,
        "value": 1000
      },
      {
        "code": "countdown_3",
        "custom_name": "",
        "dp_id": 20,
        "time": 1711947507414,
        "value": 0
      },
      {
        "code": "light_mode",
        "custom_name": "",
        "dp_id": 21,
        "time": 1711947512152,
        "value": "relay"
      },
      {
        "code": "backlight_switch",
        "custom_name": "",
        "dp_id": 26,
        "time": 1711947511953,
        "value": true
      },
      {
        "code": "night_light_color",
        "custom_name": "",
        "dp_id": 101,
        "time": 1711947512051,
        "value": "2"
      },
      {
        "code": "night_light_mode",
        "custom_name": "",
        "dp_id": 102,
        "time": 1711947512054,
        "value": "4"
      },
      {
        "code": "backlight",
        "custom_name": "",
        "dp_id": 103,
        "time": 1711947511247,
        "value": 30
      },
      {
        "code": "child_lock",
        "custom_name": "",
        "dp_id": 104,
        "time": 1711947516664,
        "value": false
      },
      {
        "code": "angle_horizontal",
        "custom_name": "",
        "dp_id": 105,
        "time": 1711947512255,
        "value": 5
      }
    ]
  },
  "success": true,
  "t": ,
  "tid": ""
}

And these are the value descriptors:

switch_led_1	Boolean	
"{true,false}"
bright_value_1	Integer	
{
  "min": 10,
  "max": 1000,
  "scale": 0,
  "step": 1
}
brightness_min_1	Integer	
{
  "min": 10,
  "max": 1000,
  "scale": 0,
  "step": 1
}
led_type_1	Enum	
{
  "range": [
    "led",
    "incandescent",
    "halogen"
  ]
}
brightness_max_1	Integer	
{
  "min": 10,
  "max": 1000,
  "scale": 0,
  "step": 1
}
countdown_1	Integer	
{
  "unit": "s",
  "min": 0,
  "max": 86400,
  "scale": 0,
  "step": 1
}
switch_led_2	Boolean	
"{true,false}"
bright_value_2	Integer	
{
  "min": 10,
  "max": 1000,
  "scale": 0,
  "step": 1
}
brightness_min_2	Integer	
{
  "min": 10,
  "max": 1000,
  "scale": 0,
  "step": 1
}
led_type_2	Enum	
{
  "range": [
    "led",
    "incandescent",
    "halogen"
  ]
}
brightness_max_2	Integer	
{
  "min": 10,
  "max": 1000,
  "scale": 0,
  "step": 1
}
countdown_2	Integer	
{
  "unit": "s",
  "min": 0,
  "max": 86400,
  "scale": 0,
  "step": 1
}
relay_status	Enum	
{
  "range": [
    "off",
    "on",
    "memory"
  ]
}
switch_led_3	Boolean	
"{true,false}"
bright_value_3	Integer	
{
  "min": 10,
  "max": 1000,
  "scale": 0,
  "step": 1
}
brightness_min_3	Integer	
{
  "min": 10,
  "max": 1000,
  "scale": 0,
  "step": 1
}
led_type_3	Enum	
{
  "range": [
    "led",
    "incandescent",
    "halogen"
  ]
}
brightness_max_3	Integer	
{
  "min": 10,
  "max": 1000,
  "scale": 0,
  "step": 1
}
countdown_3	Integer	
{
  "unit": "s",
  "min": 0,
  "max": 86400,
  "scale": 0,
  "step": 1
}
light_mode	Enum	
{
  "range": [
    "none",
    "relay",
    "pos"
  ]
}
switch_backlight	Boolean	
"{true,false}"

Working on an external converter now

lambsfry, 2024-04-01:
Have got something mostly working 

I have not however worked out what to do with these:
- switch_backlight Boolean
- light_mode Enum
- relay_status Enum

lambsfry, 2024-04-01:
Here is my converter:
[ZN2SRS3EDH.js.txt](https://github.com/Koenkk/zigbee2mqtt/files/14821443/ZN2SRS3EDH.js.txt)

- couldn't get the code markdown working so uploaded as a txt

lambsfry, 2024-04-01:
`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    // Since a lot of TuYa devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE204_znvwzxkq',
        },
    ],
    model: 'TS0601',
    vendor: '_TZE204_znvwzxkq',
    description: 'Smart Dimming 3 Gang Switch (3 Gang)',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    //onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: tuya.configureMagicPacket,
    exposes: [
        tuya.exposes.lightBrightnessWithMinMax().withEndpoint('l1'),
        tuya.exposes.lightBrightnessWithMinMax().withEndpoint('l2'),
        tuya.exposes.lightBrightnessWithMinMax().withEndpoint('l3'),
        tuya.exposes.countdown().withEndpoint('l1'),
        tuya.exposes.countdown().withEndpoint('l2'),
        tuya.exposes.countdown().withEndpoint('l3'),
        tuya.exposes.lightType().withEndpoint('l1'),
        tuya.exposes.lightType().withEndpoint('l2'),
        tuya.exposes.lightType().withEndpoint('l3'),

    ],
    meta: {
        multiEndpoint: true,
        tuyaDatapoints: [
            [1, 'state_l1', tuya.valueConverter.onOff, {skip: tuya.skip.stateOnAndBrightnessPresent}],
            [2, 'brightness_l1', tuya.valueConverter.scale0_254to0_1000],
            [3, 'min_brightness_l1', tuya.valueConverter.scale0_254to0_1000],
            [4, 'light_type_l1', tuya.valueConverter.lightType],
            [5, 'max_brightness_l1', tuya.valueConverter.scale0_254to0_1000],
            [6, 'countdown_l1', tuya.valueConverter.countdown],
            //
            [7, 'state_l2', tuya.valueConverter.onOff, {skip: tuya.skip.stateOnAndBrightnessPresent}],
            [8, 'brightness_l2', tuya.valueConverter.scale0_254to0_1000],
            [9, 'min_brightness_l2', tuya.valueConverter.scale0_254to0_1000],
            [10, 'light_type_l2', tuya.valueConverter.lightType],        
            [11, 'max_brightness_l2', tuya.valueConverter.scale0_254to0_1000],
            [12, 'countdown_l2', tuya.valueConverter.countdown],        
            //
            [15, 'state_l3', tuya.valueConverter.onOff, {skip: tuya.skip.stateOnAndBrightnessPresent}],
            [16, 'brightness_l3', tuya.valueConverter.scale0_254to0_1000],
            [17, 'min_brightness_l3', tuya.valueConverter.scale0_254to0_1000],
            [18, 'light_type_l3', tuya.valueConverter.lightType],        
            [19, 'max_brightness_l3', tuya.valueConverter.scale0_254to0_1000],
            [12, 'countdown_l3', tuya.valueConverter.countdown],
        ],
    },
    endpoint: (device) => {
        return {'l1': 1, 'l2': 1, 'l3': 1};
    },
};

module.exports = definition;`

Koenkk, 2024-05-04:
Added!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-05-04 by Koenkk

Casadadiana, 2024-05-04:
Wowwwww... perfect... thank you very very much, i was using a tuya hub just for this switch, but it was at cloud, sucks,.. now i'm local... :)